mavlink/mavlink #1165: Common flight mode architecture

LorenzMeier, 2019-06-26:
The goal is to move the flight mode selection into a new structure. Here are a couple of thoughts: https://docs.google.com/document/d/1LIcfOL3JrX-EznvXArna1h-sZ7va7LRTteIUzISuD8c/edit

LorenzMeier, 2019-08-01:
We should start to document the existing baseline flight modes. We can use a similar structure (but using MAVLink, not PX4 terminology) from here: https://docs.px4.io/v1.9.0/en/flight_modes/

This is driven by this base ENUM:
https://mavlink.io/en/messages/common.html#MAV_MODE

Where:

  * Manual: Direct pass-through from pilot to outputs. This really only applies to planes. Multicopters should by convention default to manual == stabilized == attitude control
  * Stabilized: Pilot in the loop. Attitude control. Roll/pitch are controlled by stick position, yaw rate by stick position
  * Guided: Pilot in the loop. Position control. X/Y velocity are controlled by stick position, yaw rate by stick position
  * Auto: Any flight mode in which the pilot is not in the loop (== no sticks). Includes missions, go-to commands, RTL, etc.

These should be documented properly as baseline and should be available by standard-compliant systems.

hamishwillee, 2019-08-02:
1.  Trying to get my head around based modes:
   - [MAV_MODE_FLAG_GUIDED_ENABLED](https://mavlink.io/en/messages/common.html#MAV_MODE_FLAG_GUIDED_ENABLED) - appears to not match above "guided mode enabled, system flies waypoints / mission items. - but everything else does. So is this incorrect as it is really for manual setpoint.
   - MAV_MODE has variants for armed and disarmed (e.g [MAV_MODE_GUIDED_DISARMED](https://mavlink.io/en/messages/common.html#MAV_MODE_GUIDED_DISARMED)), but docs are the same in each case: *System is allowed to be active, under autonomous control, manual setpoint*. What does "System is allowed to be active" mean? If the thing is disarmed I'd expect it perhaps to be in the specified mode but not armed.

auturgy, 2019-08-21:
Simply put: not supported in this form/structure at all.
You’re moving further away from what might be able to be put on the table to be a standard

hamishwillee, 2019-08-22:
And just to be more precise about the problem. 
1. The [original spec](https://docs.google.com/document/d/1LIcfOL3JrX-EznvXArna1h-sZ7va7LRTteIUzISuD8c/edit) is ambiguous and incomplete. It does not clearly explain the difference between common, core, custom modes, how they are each set/get, and how to resolve case where modes are similar but not the same. What id does do though is imply that there is a simple set of modes in an enum that you can set/get.
2. The discussion above https://github.com/mavlink/mavlink/issues/1165#issuecomment-517200692 appears to contradict the apparent intent of the spec and reintroduce the concept of base and custom/extended modes. So not takeoff, but "auto" + "takeoff". 

I suspect that is not what was intended.

hamishwillee, 2022-04-20:
Closing. This was addressed by RFC and is in prototype implementation phase.

closed 2022-04-20 by hamishwillee