mavlink/mavlink #1168: `PARAM_EXT_SET` and `PARAM_EXT_VALUE` not correctly truncated

julianoes, 2019-07-01:
It was raised at the PX4 DevSummit by one of the participants that the messages [PARAM_EXT_SET/VALUE/ACK](https://github.com/mavlink/mavlink/blob/95031e31a5eec47dc951fc09e2939071b8cf3c2b/message_definitions/v1.0/common.xml#L5008-L5030) are not correctly truncated (trimmed because of trailing zeros).

Looking at the C generated code of the packed struct this seems to be true:

https://github.com/mavlink/c_library_v2/blob/beb789085dde84cfafc3d227fca1be68199ac402/common/mavlink_msg_param_ext_value.h#L6-L13

@LorenzMeier are we missing something here? Or is the generated code not correctly ordering the fields for the arrays? Could it be an issue of the parser of uint8_t vs. char?

julianoes, 2019-07-01:
I did some digging and it seems that the order of the packed struct members is according to type length and not type length * array length as one might assume.

| | type length | type length * array length |
| ---- | ---- | ---- |
| float | 4 | 4 |
| float[10] | 4 | 40 |

This could be "fixed" by this change in pymavlink mavparse.py:
```
diff --git a/generator/mavparse.py b/generator/mavparse.py
index 85ddaf5..841467a 100644
--- a/generator/mavparse.py
+++ b/generator/mavparse.py
@@ -380,7 +380,7 @@ class MAVXML(object):
                 # when we have extensions we only sort up to the first extended field
                 sort_end = m.base_fields()
                 m.ordered_fields = sorted(m.fields[:sort_end],
-                                                   key=operator.attrgetter('type_length'),
+                                                   key=operator.attrgetter('wire_length'),
                                                    reverse=True)
                 m.ordered_fields.extend(m.fields[sort_end:])
             else:
```

However, this would probably break many messages on the wire, not just `PARAM_EXT_`.

I assume this is correctly implemented then (given the spec is to order by type length and not by wire length) but the `PARAM_EXT_` messages were badly designed by not having the 128 byte value defined last.

Therefore I conclude that if we want to fix it we need to add new messages and deprecate the old ons.

hamishwillee, 2019-07-01:
> I did some digging and it seems that the order of the packed struct members is according to type length and not type length * array length as one might assume.

Ordering by type rather than array is correct according to the mavlink spec: https://mavlink.io/en/guide/serialization.html#field_reordering

The spec says that this is an array of **char** in [PARAM_EXT_VALUE](https://mavlink.io/en/messages/common.html#PARAM_EXT_VALUE) (not floats). As you say param_type and param_value should be switched to ensure the array can be truncated.

I believe these are currently only used for camera API, so you might be able to break this by talking to Gus.

julianoes, 2019-07-02:
> Ordering by type rather than array is correct according to the mavlink spec: https://mavlink.io/en/guide/serialization.html#field_reordering

Great, that's what I was looking for!

> I believe these are currently only used for camera API, so you might be able to break this by talking to Gus.

If we change it we break QGC, MAVSDK, and camera implementations. For instance we would completely break interfacing to Yuneec H520.

hamishwillee, 2019-07-03:
Good point. I'd also throw in that @tridge promises another version of the parameter protocol at some point. It would be good to know when that will deliver, because there is no reason that couldn't be the replacement for this API.

LorenzMeier, 2019-07-10:
Given this is mostly an issue for cameras and they tend to have decent bandwidth for the typical use cases I'll close this - we will not fix it until introducing a new revision to the overall parameter protocol.

PR #1430 (2020-07-25): common: fix PARAM_EXT trimming
It turns out that PARAM_EXT messages were not trimmed as we thought they would be. The fields of a message are ordered by the size of the type, however it does not matter if the type is an array or not.
This means that char[128] is not at the end of the message and therefore not automatically trimmed.

This patch introduces new messages called _TRIMMED which has the param_value as part of the extension meaning that it will be at the end and trimmed correctly.

I'm open to other suggestions on how to do this or better naming suggestions.

FYI @DanielePettenuzzo.
C [... cut]
closed 2019-07-10 by LorenzMeier