mavlink/mavlink #1169: Component version

julianoes, 2019-07-01:
It would be nice for a MAVLink component other than an autopilot to be able to announce its version. 

This is probably related to #1133.

hamishwillee, 2019-07-02:
So a generic version of [AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION)?

olliw42, 2019-07-09:
the general "problem" here is that components can be of various types, and that for "some components" this sort of message are already available, specifically for the camera there is already a CAMERA_INFORMATION message.

At some place there was a discussion if that should be extended to all components, or if there better should be messages for the various types of components. Given that CAMERA_INFORMATION is already existing and fixed part of the camera microservice, the answer probably would be to add a message for all non-camera and non-autopilot components.

I'd like to emphasize that as part of the proposal #1133, I had suggested a COMPONENT_INFORMATION message already, which might serve as an indication, see the related QGC issue https://github.com/mavlink/qgroundcontrol/issues/7503. 
The bottom line is that with progress with that proposal, also this issue here would be resolved.

julianoes, 2019-07-09:
I think `COMPONENT_INFORMATION` messages would make sense. @olliw42 if you could make a pull request with the messages that would be great.

olliw42, 2019-07-09:
I'm happy to try doing this

could you tell which ids I should use for the message and related request command (I used 700 in both...)

auturgy, 2019-11-29:
Suggest close (https://github.com/mavlink/mavlink/pull/1176)

hamishwillee, 2020-05-12:
@julianoes Is this alive, or something that will be part of component information?

julianoes, 2020-05-13:
@hamishwillee good point. It should be part of the component information.

hamishwillee, 2020-05-18:
Excellent. Now of course you get to make that actually happen :-)

hamishwillee, 2020-07-20:
This got implemented as WIP, but is now being modified so that this information will be sent in the the component information metadata.

closed 2020-07-20 by hamishwillee