mavlink/mavlink #1171: Mission checksum package

holmnikolaj, 2019-07-03:
Instead of downloading all the mission from the drone to check that they are in sync it would be much faster if the mission started with a checksum item. As far as I can see there is no such thing at the moment but this would be very helpful for setups with long missions and poor link quality.

dagar, 2019-07-03:
Very interested in this as well. 

@antiheavy @DonLakeFlyer

Antiheavy, 2019-07-03:
Yes!  And helpful with knowing if the flight plan on the vehicle is the same or different than what is onscreen (eg QGC).

hamishwillee, 2019-07-03:
PX4 and QGC do something similar for parameters. It would be nice if we could use the 0th index for this, but ArduPilot populate that with the home position.

holmnikolaj, 2019-07-04:
It could be something as simple as a mission item that is simply ignored by the autopilot. It is only the UI that should be in charge of synchronization anyway. 
Then the seven parameters (or just some of them) can contain the checksum. 

1. The UI will upload the checksum mission item  as part of the mission if it supports it.
2. The Autopilot completely ignores it.
3. The UI reads the checksum mission item from the autopilot and verifies that the mission is in sync. 

The checksum item could be placed first in the mission of second (just after takeoff) if this is more convenient.

JohannesFriis, 2019-07-04:
I guess it could be done similar to MAV_CMD_DO_LAND_START (#189 ). This acts as a marker in the mission waypoint list. I know that making it a command has a double purpose, that makes sense for land start cmd, but the idea of a marker wp in the mission list containing the mission checksum could be implemented with small changes in both mavlink and autopilot's. i agree that placing the marker early in the mission list is preferred.

dagar, 2019-07-04:
I think it needs to be handled like metadata across multiple parts of the mission protocol to solve most of the problems efficiently. I'd also like to consider the checksum itself being part of the spec.

For example, `MISSION_CURRENT` (https://mavlink.io/en/messages/common.html#MISSION_CURRENT) is what the vehicle broadcasts so that the GCS can update the active waypoint. It could be extended to also broadcast the ID (mission checksum).

We should be able to request a checksum for an entire mission, subset range, or individual mission_item. That could also enable incremental sync.

LorenzMeier, 2019-07-04:
@dagar Proposal makes total sense. @holmnikolaj do you want to extend your PR to include a full proposal?

hamishwillee, 2019-07-05:
Along with broadcast, might be nice to be able to request rebroadcast. So if, for example you missed the mission change message you could still request a new one. Maybe use [MAV_CMD_REQUEST_MESSAGE](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE) for this

holmnikolaj, 2019-07-05:
The points of implementing it as a mission item was that:

1. The drone does not change the mission itself, so the sync issue will only arise if something happens in the GCS end. As it is now, the solution is to read all waypoints. With this solution it only have to read until the checksum.
2. We don't have to agree on how the checksum is calculated across PX4, Ardupilot, QGC, MP etc. Any GCS can make each own checksum calculation, leave it in the mission item and then read it again when needed. 
3. It will only add and extra package sent when uploading a mission. No extra communication to clutter the link.

Not saying it has to be that way, just some reasons way it was how I proposed this solution.

holmnikolaj, 2019-07-05:
@holmnikolaj do you want to extend your PR to include a full proposal?
@LorenzMeier What do you mean by full proposal? It was my understanding that everything is autogenerated based on the .xml file. If you refer to support in Px4/QGC etc. I plan on doing the Px4 part when we have agreed on the mavlink part.

hamishwillee, 2019-07-05:
@holmnikolaj Lorenz is probably away today. 

What I think he is suggesting is a more fully developed proposal in line with the suggestion @dagar made above: https://github.com/mavlink/mavlink/issues/1171#issuecomment-508531849

Basically what you proposed is a mission command within the mission. So if you have uploaded a mission previously you get a checksum. If you want to check the mission again you have to kick off another upload. The first item tells you that if the mission has changed, and you can ignore the rest of the items. That is good as far as it goes - but firstly you have to "pull" the information rather than being told that the mission has changed, and secondly this results in a whole sequence of mission upload that you might not actually need (unnecessary messages).

Daniel has proposed that the system might emit the checksum when something changes (and I have suggested you might do so on request). He has also said you could then define different levels of hash - for entire mission, subset range, or individual mission_item (ie to enable incremental sync).

To my mind that is probably more than a PR, but you could start outlining how you would do it. 

Or push back if you this does not make sense to you.

holmnikolaj, 2019-07-05:
As I understand it you want a solution where:

- The checksum is sent as part of MISSION_CURRENT.
- Possibility to request checksum without reading (parts of) mission from the MAV, which is already there if it is part of MISSION_CURRENT.
- There is support for checksum for mission sections and individual waypoints. 

I think the possibility to have checksums for sections of a mission makes good sense, but if you are in doubt about a single waypoint why not upload it again instead of requesting a checksum for it?

I want a solution where:
- The checksum is only calculated at the CGS. 

That way we avoid miscalculations in the checksum due to float/double precision. 
We save calculation time one the drone where it may actually be an issue. 
We don't (necessarily) have to agree on how to calculate it. 

To get all this I want to propose the following solution:
As part of the mission upload checksum items (as in the original proposal) with this format: 

1. mission checksum 1
2. mission checksum 2
3. section start sequence
4. section end sequence
5. section checksum 1
6. section checksum 2
7 empty

These could be added to a mission in the beginning of the entire mission, and again in in the beginning of mapping areas etc. 

The CGS will calculate the checksums, and the MAV just copies the relevant checksums to the MISSION_CURRENT message that will be sent when waypoint reached or by request as usual. 

That way we can get an implementation that will work without opening a big can of worms. We have the best of both yours and my idea. This whole thing can be ignored for GCS/autopilots that does not want to use it. When we one day decide to change the structure of missions to support section division and make the MAV calculate the checksum itself, this will still work by just leaving the mission checksum items out of the mission. 

The downside is that the mission will contain (otherwise unnecessary) items, but it supports enhancements where the drone calculates it itself and thereby removing the extra mission items.

DonLakeFlyer, 2019-07-05:
The checksums on sections seems overly complicated. Although something needs to be done about partial uploads. How do you update the checksum for the entire mission is you are only updating a single item.

Using a new mission item to store a checksum in a mission upload is tricky since old firmwares which don't know about this item will fail the upload. The whole thing is likely to need a capability bit.

> * Possibility to request checksum without reading (parts of) mission from the MAV

Having this as a separate command as well instead of just streaming MISSION_CURRENT seems like a good idea. A GCS doesn't really know the telemetry rate of that stream which can be adjusted. Having a more deterministic way to get it may speed things up in some case.

prebenfihl, 2019-07-09:
I am working with @holmnikolaj on this issue. We have updated the pull request to include some of the comments.

We have reduced the scope of the change to address only a checksum/unique id for the whole mission (I'll comment on our considerations shortly), and we have added a bit to MAV_PROTOCOL_CAPABILITY.

We have introduced a MISSION_UID message rather than extending MISSION_CURRENT. Logically, changes in the current waypoint has nothing to do with the mission UID, so we propose to keep them separate. We considered extending the MISSION_CHANGED message, which is logically a better fit, but requesting MISSION_CHANGED to determine if a GCS mission is in sync doesn't match the intention/description of MISSION_CHANGED very well.

prebenfihl, 2019-07-09:
Regarding partial mission upload: we have not specified this since the protocol for partial uploads is still to be defined (as far as we know) but MISSION_WRITE_PARTIAL_LIST cloud potentially be extended to include the new UID for the whole mission (which the GCS can calculate).

Regarding checksums/ids for parts of the mission: we agree with @hamishwillee that such a multi-level ID requires more than just a PR but we think that the mission UID would be a reasonable feature in itself (hence this proposal). We have marked the four parameters of the MAV_CMD_MISSION_UID cmd that was used for the section checksums/ids in the previous suggestion as reserved. We still see it as a viable solution but we think that the details should be defined in the context of a full multi-level ID solution.

hamishwillee, 2019-07-10:
@prebenfihl @holmnikolaj We have a MAVLink devcall this evening at 5pm AEST - in about 7 hours: https://mavlink.io/en/about/support.html#dev_call
Would you like to attend and discuss this?

As I understand it you still intend that MAV_CMD_MISSION_UID be sent in the mission? As pointed out by @DonLakeFlyer this will break older firmware, which should reject the mission. 

Further to that, I'd like the explicit agreement of @dagar that he is OK that this proposal does not preclude partial mission support in future. 

> The drone does not change the mission itself, so the sync issue will only arise if something happens in the GCS end. As it is now, the solution is to read all waypoints. With this solution it only have to read until the checksum.

Note, that you do need to consider that there may be multiple GCS or GSC and developer APIs affecting the mission. Not just a single GCS.

With respect to partial missions I believe the requirement is well understood, I just can't get the documentation reviewed: https://github.com/mavlink/mavlink-devguide/pull/157

@DonLakeFlyer As an aside, the behaviour for unsupported mission types is not documented in https://mavlink.io/en/services/mission.html - though it makes sense to reject the whole mission if an item is not understood because you don't know how important a particular item is to the whole mission. Does any Firmware do anything different here?

hamishwillee, 2019-07-11:
Thanks for attending the [20190710-Dev-Meeting](https://github.com/mavlink/mavlink/wiki/20190710-Dev-Meeting) @holmnikolaj !

Just to summarise the results, we agree the value of the checksum, and that your general approach is good. However we'd like to use the yet-to-implement mavlink microservices protocol to check the existence of the message rather than a capability bit.

@julianoes will look into defining/implementing that, starting in 4 weeks, and expected to get something working in 3 months.

hamishwillee, 2019-09-19:
Hi @holmnikolaj, 
FYI, the work for microservices versioning has not yet been done. @julianoes is hoping to assess the work required in the coming weeks. 

The discussion here was long ago, so can you please check my summary/recollection of the thinking:
- GCS generates a mission GUID for the mission is has created and inserts this as a mission item in a `MAV_CMD_MISSION_UID` (if it can verify the recipient understands this `MAV_CMD`)
- GCS can _check_ for mission changes in two ways:
  - Send `MAV_CMD_REQUEST_MESSAGE` and monitor for the `MISSION_UID`.
  - Download mission and check `MAV_CMD_MISSION_UID`. Continue download if it has changed, cancel download if it has not changed and just stick with current message.
- GCS should also monitor for `MISSION_UID` messages; if mismatched this should minimally inform user that stored mission is out of date, and potentially auto-upload new version. This is to cater for case where there are multiple potential setters and observers of missions. 
- Drone should store the UUID as part of the mission, and emit as a `MISSION_UID`:
  - If requested using `MAV_CMD_REQUEST_MESSAGE`
  - Whenever the stored mission has changed - ie after an upload, partial upload, or the mission is cleared. 

Further thoughts/questions:
- If a GCS misses a `MISSION_UID` from another system doing an update it can get out of sync.
- What if the GCS does a partial mission upload? In theory it should calculate the UUID for the whole mission and include the `MAV_CMD_MISSION_UID`, but there may be problems with this:
  - If the GCS is out of sync, its calculated `MAV_CMD_MISSION_UID` would be incorrect.
  - The partial update is not well defined, but we know from the message format it takes a start index and an end index. How will we update the `MAV_CMD_MISSION_UID` if it is not in the range being updated? What if there are multiples of `MAV_CMD_MISSION_UID` in the mission after the update?

holmnikolaj, 2019-09-19:
I discussed your questions with @prebenfihl.
We agree on your summary.
Regarding the partial upload, we think that this is beyond the scope of the issue for now. Lets create the UID and then agree on a solution for partial upload when appropriate if that is a fully supported feature (we are not aware of the state of partial uploads). 

We did however talk about some ways for handling the partial uploads with multiple GCS connected. 

We could have a look at how it is handled in STANAG 4586 where GCSs have different levels of interoperability (LOI) such that only one GCS is in charge and the rest can only listen. It can then hand over the control if needed. 

Otherwise we could spend the time between development of UID and Partial upload to agree on how the UID is calculated. That way the drone can respond with a new UID after partial upload and the drone can check if it gets the same.

hamishwillee, 2019-09-20:
Thanks @holmnikolaj - I'm not sure we can treat it as out of scope, because some systems do support it (e.g. ArduPilot - albeit in a horrible and broken way). 

Cooperative GCS interoperability is a bigger problem; it is not just a matter of GCS - any mavlink app would have to buy into this in some way. I wouldn't want to go down that path to fix this problem. 

> Otherwise we could spend the time between development of UID and Partial upload to agree on how the UID is calculated. That way the drone can respond with a new UID after partial upload and the drone can check if it gets the same.

I'm not _certain_ the GCS needs to know how the UID is generated, it just needs to know that it has changed. Depends on whether the GCS uses an id for comparison that it calculates OR it uses an ID that is given by the drone. The first way means that you have have a common mechanism for generating the hash, the second means that you have to have a way of reliably associating the hash with the mission that someone just uploaded.

A possible solution assuming the second option
1. We add hash/mission number field to [MISSION_COUNT](https://mavlink.io/en/messages/common.html#MISSION_COUNT) and [MISSION_ACK](https://mavlink.io/en/messages/common.html#MISSION_ACK)
1. GSC [uploads mission](https://mavlink.io/en/services/mission.html#uploading_mission). 
   - We don't care what the number is in `MISSION_COUNT`. 
   - Final `MISSION_ACK` from drone includes the drone-calculated hash for mission. 
   - Drone also emits the `MISSION_UID` so everything else knows about this change.
1. For [mission download](https://mavlink.io/en/services/mission.html#download_mission) the first message from drone is `MISSION_COUNT` which now includes the hash.
   - At this point the GCS could cancel the download, because it knows it has a matching hash.
1. The `MISSION_UID` can still be requested using `MAV_CMD_REQUEST_MESSAGE`. 
1. `MAV_CMD_MISSION_UID` is no longer needed, which is what breaks partial upload
1. The same approach works for [Partial mission upload](https://hamishwillee.gitbooks.io/ham_mavdevguide/content/v/mission_ap/en/services/mission.html#upload_partial) (note, link is not approved doc) which completes with `MISSION_ACK`. It doesn't work for partial download, but that really is out of scope.

What I really like about this is that I **think** it breaks nothing and could be rolled out straight away without side effects. i.e. if working with a drone or gcs that has not been updated or does not support this feature the fields for the UID will always be zero. We can make that a magic number that means unknown or not supported. Ie if GCS gets zero from upload it knows that the drone doesn't support hash, and if it gets zero from a download, same thing.

holmnikolaj, 2019-09-20:
Your proposal does have some advantages, but the reason why we wanted this in the first place was to ensure that the GCS could compute an ID for comparison.

Consider the case that you made some changes to your mission on your GCS and you don't know if you are in sync (due to program crash etc). The UID is only calculated on the drone and we don’t know if the UID, the GCS has, corresponds to the mission it shows. 

To solve that the GCS has to be able to calculate the UID itself and then we are back to all agreeing on how this is done. However, your proposed way of exchanging the UID would still make sense.

It might be the best option but as mentioned earlier I think it makes the implementation a lot more complicated.

However I we consider the UID calculated on the drone to be the "deciding" one then the GCS will just have to know how the autopilot that is connected calculates it. That could be different from autopilot to autopilot, and in case the calculation is not implemented on the GCS it only loses the feature mentioned above.

hamishwillee, 2019-09-23:
@holmnikolaj Sure.

So perhaps something like:
1. Check if feature is supported:
   - Maybe assume if `MAV_CMD_REQUEST_MESSAGE` with `MISSION_UID` returns any ACK except [MAV_RESULT_UNSUPPORTED](https://mavlink.io/en/messages/common.html#MAV_RESULT_UNSUPPORTED) the feature is supported on the flight stack? 
1. Add field for mission id to [MISSION_COUNT](https://mavlink.io/en/messages/common.html#MISSION_COUNT)
1. GSC calculates hash and [uploads mission](https://mavlink.io/en/services/mission.html#uploading_mission) with id in `MISSION_COUNT.mission_id`.
   - GCS does not store this as id unless gets accepted ACK!
1. For [mission download](https://mavlink.io/en/services/mission.html#download_mission) the first message from drone is `MISSION_COUNT` which now includes the id
   - At this point the GCS could cancel the download if the id is matching, or continue if it is not
   - An id of zero could mean feature is not supported or some other GCS uploaded the mission. Either way the GCS should continue download (it could then decide to re-upload a mission with the ID if it chose).
1. [Partial mission upload](https://hamishwillee.gitbooks.io/ham_mavdevguide/content/v/mission_ap/en/services/mission.html#upload_partial) resets the mission id to zero on drone (ie no additional messages or fields required)
1. Any accepted update to the mission causes drone to emit the `MISSION_UID` so everything else knows about this change (including full and partial updates)

For the partial upload I don't see any way to make these work "end to end" unless the id is generated on the drone (or both end). Realistically there are bigger problems than this with getting these to work robustly, so for this case just making it clear to other systems they can no longer rely on the stored copy is the important thing.

Any obvious holes in this?

holmnikolaj, 2019-09-23:
@hamishwillee 
What you describe sound good. If there is consensus about that solution I think we should go with that.

hamishwillee, 2019-09-23:
Cool :-) @julianoes @julianoes @auturgy @DonLakeFlyer There is a way to do this mission checksum without mission commands or breaking GCS or autopilot systems that don't know about it.  Broad scope of proposal above https://github.com/mavlink/mavlink/issues/1171#issuecomment-533980887 - but it comes down to adding the checksum in `MISSION_COUNT` and using the `MAV_CMD_REQUEST_MESSAGE` as verification that the service is supported.

1. Does this make sense? 
2. @DonLakeFlyer Any thoughts on how the GCS might calculate the checksum for a mission plan? We need to specify this in a way that is GCS agnostic/can be implemented by anyone.

prebenfihl, 2019-09-23:
A thought on the calculation of the mission UID: 
Using a standard checksum or hashing function seems reasonable (CRC, SHA, ...) and when balancing computational complexity with the uniqueness of the UID a CRC32C checksum could be a suggestion.

An array of  [mission_item](https://mavlink.io/en/messages/common.html#MISSION_ITEM) fields could be the basis for the checksum but omitting `<target_system>` and `<target_component>` (since two drones could fly the same mission). Sequence number (`<seq>`) of the mission item could probably also be omitted (changing the sequence number is reflected in the changed order of waypoints).

LorenzMeier, 2019-09-23:
We should use the same approach as in the param checksum, given this is the same approach.

hamishwillee, 2019-09-24:
@LorenzMeier @prebenfihl The param hash is just a CRC of all active param names and 4 byte values (see [param_hash_check()](https://github.com/PX4/Firmware/blob/master/src/lib/parameters/parameters.cpp#L1353) ).

Note we're proposing to calculate the hash on the GCS rather than drone, but the same approach makes sense if you assume you can just hash mission items (or rally point or geofence) in index order by name, command id, and field value. 

But what about the case where a mission is defined as a pattern(s)? Can we just get the GCS to convert patterns to waypoints and then calculate a hash?

prebenfihl, 2019-09-24:
Using the standard CRC32 checksum calculation of the param checksum makes good sense of course, and following a similar approach by using the (relevant*) field values of mission items in index order for the checksum would also makes sense (as outlined by @hamishwillee ).

Regarding missions defined as patterns: are all missions not exchanged between GCS and drone as a list of mission_items? If that is the case then it seems safe to just let the GCS generate the list of mission items before calculating the mission UID.
 
*Relevant field values (a change in any of these values will result in a different mission / mission UID just as a change in order of mission items would):
frame, command, autocontinue, param1, param2, param3, param4, x, y, z, and mission_type (for MAVLink2)

DonLakeFlyer, 2019-09-24:
> Regarding missions defined as patterns: are all missions not exchanged between GCS and drone as a list of mission_items?

Correct.

hamishwillee, 2019-09-25:
> Regarding missions defined as patterns: are all missions not exchanged between GCS and drone as a list of mission_items? If that is the case then it seems safe to just let the GCS generate the list of mission items before calculating the mission UID.

@prebenfihl That was really my question - is it safe? Clearly it is if the same pattern always results in the same mission. Does it matter if another GCS with the same pattern generates a different set of mission items?

My gut feeling is no - the GCS knows the hash of whatever mission it has stored and can always match to the vehicle hash. The only "issue" is that if you had two GCS with the same mission (ie loaded from a file) they might calculate a different hash - so one will have to update.

hamishwillee, 2019-09-25:
@DonLakeFlyer So does this seem sane to you? Something you would think reasonable and useful to implement QGC side?

DonLakeFlyer, 2019-09-25:
> And helpful with knowing if the flight plan on the vehicle is the same or different than what is onscreen (eg QGC).

For this specific case this would help and not be super hard to make work. That said my opinion is that you kind of have to work hard to get yourself into this situation. @Antiheavy Wondering why you think this is important, and how the user gets themselves into this state?

> Instead of downloading all the mission from the drone to check that they are in sync it would be much faster if the mission started with a checksum item.

Making this work would be a ton of work since it requires some sort of database of known Plans keyed by CRC.

hamishwillee, 2019-09-26:
> Making this work would be a ton of work since it requires some sort of database of known Plans keyed by CRC.

The [current proposal](https://github.com/mavlink/mavlink/issues/1171#issuecomment-533980887) doesn't require that. The hash is associated with a vehicle - and would either be stored per-vehicle or generated when needed. There is no longer a concept that the hash is IN the plan.

> Wondering why you think this is important, and how the user gets themselves into this state?

Probably because multiple things can edit a mission and you want to know why the vehicle isn't following the path you see on screen.

DonLakeFlyer, 2019-09-26:
> Probably because multiple things can edit a mission and you want to know why the vehicle isn't following the path you see on screen.

I get that it could happen. I don't get how it will happen in most normal usage. Don't want to expend a lot of effort on a use case which just doesn't happen very often.

> The hash is associated with a vehicle - and would either be stored per-vehicle or generated when needed. There is no longer a concept that the hash is IN the plan.

Somebody explain exactly from an end user perspective what problem this is solving then. Please explain in a sequence of end user steps.

hamishwillee, 2019-10-01:
@holmnikolaj @prebenfihl You get first shot at answering ^^^^ https://github.com/mavlink/mavlink/issues/1171#issuecomment-535590681 . Don has the fun on implementing QGC, so a convincing argument as to why this is something that users will value (perhaps over the millions of other things he might do) is worth providing.

auturgy, 2019-10-01:
> @holmnikolaj @prebenfihl You get first shot at answering ^^^^ #1171 (comment) . Don has the fun on implementing QGC, so a convincing argument as to why this is something that users will value (perhaps over the millions of other things he might do) is worth providing.

I’d just add that in my view it’s important to separate validation of the mission from verification that it’s come from a trusted source.
The latter is a lower level task that should either be done via mavlink signing or encryption of the transport/carrier.

Regards,

James

hamishwillee, 2019-10-01:
@auturgy Never thought otherwise :-). Would have assumed mavlink signing, since that is what it is for.

My understanding is that this has the same purpose as the parameter protocol cache - it allows easy and lightweight checking that GCS version of the mission matches that on the vehicle. Not a big deal for small missions, but given the size of missions you can generate with patterns, could have significant impact both on link usage and the time to confirm to a user that the current mission displayed when they start the GCS is correct.

But let's see what @holmnikolaj has to say - not my PR.

fnoop, 2019-10-01:
Hi, would it be possible to add an overall mission checksum field to MISSION_COUNT, and also a field which contains an array of checksums, one per MISSION_ITEM?  MISSION_COUNT can be sent in answer to MISSION_REQUEST_LIST and also as a trigger to anything listening (MISSION_STATUS might be a better name, and MISSION_REQUEST_STATUS to retrieve it).

If the GCS receives an array of checksums per item, it's then up to the GCS to determine if it's missing a single item and request it, or a partial/full list, and it's then much easier for the GCS to determine if the order of items has changed (eg. if an item was added or deleted in the middle) and reorder internally rather than have to retrieve an entire list.

prebenfihl, 2019-10-01:
The short version: we see a real benefit in the easy and lightweight way of matching missions between GCS and drones as outlined by @hamishwillee .

_In terms of user steps:_
The user has a mission with a survey pattern and a fairly precise polygon fence (that is easily a few hundred waypoints) that she flies several times to inspect differences over time. She prepares the flight at the office and uploads the mission. When she connects the drone at the flying sight, the connection is established faster (missions are already in sync). (I would expect the workflow of preparing a flight at one location and then later (re)connecting the drone at a different location to be a common workflow.)

Another user has a similar sized mission. During the flight his GCS device crashes (a poorly written driver for the radio module). When the GCS is running again the mission can be loaded fast from disk and then just verified with the mission UID. (This does not happen very often but regaining full mission status in a fast way seems important in such a situation).

A user has two GCS running on different devices that initially have the same mission (e.g. one device acts as backup for the other or the user needs to be able to handover control to a second GCS). When switching to the backup device (or handing over control to the second device) the user needs a fast way to verify that the missions are still in sync.

@DonLakeFlyer, does such use cases make sense to you?

Sorry for the delayed input…

prebenfihl, 2019-10-01:
@fnoop, the "change detection" part of your comment may be handled with the [MISSION_CHANGED](https://mavlink.io/en/messages/common.html#MISSION_CHANGED) message assuming that you start with missions that are in sync. Achieving mission sync in the first place with that kind of "compressed" `mission_item`s list is possible, but is it needed?

fnoop, 2019-10-01:
@prebenfihl MISSION_CHANGED doesn't really help in the case of adding or deleting an item, or in the case of an interrupted upload/download.  Having an array of item checksums in a single meta-data message would allow the GCS to determine any of these situations and the best way to recover from them.

For example the link might have dropped two items before finishing downloading 500 mission items, so the GCS could easily determine it just needs the last two and re-request them.  Or one waypoint in the middle of 500 might have been deleted, the metadata would allow the GCS to work out how to reorganise the mission without having to re-download 499 items.  etc etc

DonLakeFlyer, 2019-10-01:
> My understanding is that this has the same purpose as the parameter protocol cache - it allows easy and lightweight checking that GCS version of the mission matches that on the vehicle. Not a big [...]

Exactly. You boot QGC, connect to the vehicle, get a UID for the Plan from the vehicle and then what? There is no way to connect that UID to a Plan which "maybe" is sitting in the file system somewhere. There is no way to connect the two together in a reasonable way without a lot of code on the QGC side as well as a likely different way to store Plans. Hence my comment about needing a database of Plans keyed by CRC.

> I would expect the workflow of preparing a flight at one location and then later (re)connecting the drone at a different location to be a common workflow.

I don't think that is the common case. Yes the Plan is prepared at one location, but I don't think it's that common to load it then. The more common case is to load at the field. And in many cases to tweak it at the field to account for things like current wind direction.

Another note keep in mind that QGC deals in the concept of a Plan, not a Mission. A Plan is a mission, geofence and rally points treated as a single unit. It is the entire thing which needs a UID to be able to relate back to the origin Plan.

I am not arguing that this wouldn't be a nice feature. I myself have wanted something like it for a while but for a different reason. As it stands now if you load the Plan back from the vehicle it degrades into a big pile of waypoints. The Complex Item structure for things like Survey are lost. I'm just saying that I don't think the usage model from an order of the way things are normally done is that common or critical. Which is why I haven't pushed for it myself. Just want to make sure everyone understands what they are really getting. Also that in order to make it work it's a minor amount of work on the firmware and a big pile of work in QGC to do it right. And given that it's not something that I would have time to work on for quite some time. So I'm all fine with working through how to make it work. But once it all gets spec'ed out and implemented on the firmware side don't expect anything quick to happen on the QGC side.

DonLakeFlyer, 2019-10-01:
> does such use cases make sense to you?

Yes they do. My only comment is that I see them as uncommon.

DonLakeFlyer, 2019-10-01:
> The hash is associated with a vehicle - and would either be stored per-vehicle or generated when needed. There is no longer a concept that the hash is IN the plan.

I don't quite get this part? Why isn't the hash associated with the plan and stored with the plan?

Also there needs to be a solution with respect to create a super-hash something or other which is the UID for the mission+geofence+rally points.

DonLakeFlyer, 2019-10-01:
> GSC calculates hash and [uploads mission](https://mavlink.io/en/services/mission.html#uploading_mission) with id in `MISSION_COUNT.mission_id`.
> * GCS does not store this as id unless gets accepted ACK!

Don't quite get this part either. Why does it need to only store UID after ack. The GCS is going to need to store a UID for a Plan at creation time, not at upload time. Those two steps may happen at different points in time.

DonLakeFlyer, 2019-10-01:
So the basic use case is:
* Vehicle already has a Plan on it
* Gcs wants to skips downloading that Plan if possible

In order to do that (at a high level):
* There needs to be a UID for the Plan. For now I think that can the triplet of mission/fence/rally UIDs.
* **Both** the vehicle and the gcs needs to be able to generate UIDs for a set of mission items in a repeatable manner which yields the same uid no matter where it is generated.
* The user creates a Plan, the gcs stores the Plan UID triplet with the Plan.
* The user uploads the Plan to the vehicle
* The user closes the gcs
* The user restarts the gcs
* The gcs asks the vehicle for the Plan triplet. The vehicle generates the triplet if it hasn't already, and sends it in response to the query.
* The gcs takes that Plan triplet and tries to match it against something it already has in storage
* If it finds it, it loads that Plan from storage and shows it in the ui. It does not download the Plan from the vehicle.
* If it doesn't find it, it dowloads the Plan from the vehicle.

Some things to note:
* In order to make this work, the UIDs do not need to be uploaded with the mission.
* Nor do they need to be downloaded with the mission.
* The UID query mechanism is a completely seperate thing. And that is an important point since the gcs is going to want to be able to match a triplet without having to start any sort of a download. Since it can only download a single type at a time (mission/fence/rally) it doesn't work to get the UID during the download process.

DonLakeFlyer, 2019-10-01:
For the UI query I'd propose a new COMMAND_LONG which takes a MAV_MISSION_TYPE as the parameter. The response to that should send another new mavlink message which includes MAV_MISSION_TYPE, and a UID amd an indicator that the vehicle does not have that type of mission on it. Which could possibly be UID=0.

With that firmwares which don't support it will just Nack the COMMAND_LONG thus requiring no additional capability bits.

prebenfihl, 2019-10-03:
> Another note keep in mind that QGC deals in the concept of a Plan, not a Mission. A Plan is a mission, geofence and rally points treated as a single unit. It is the entire thing which needs a UID to be able to relate back to the origin Plan.

This is an important distinction (which seems to me to be a little unclear so far). The concept of a Plan is natural for GCS software (probably with different implementations though) but all plans are exchanged with the drone as a list of mission_items in a MAVLink-world (that's the Mission). We deliberately propose to construct a Mission UID since that is a well-defined concept in MAVLink. You could (theoretically) have two plans that unfold to the same list of mission_items and therefore the same mission UID, so the mission UID doesn't uniquely match a mission on the drone with a GCS plan (even if you did the database of plans keyed by CRC). But on the other hand, the drone is not flying a plan, it's flying a mission, and if you have a plan that results in a mission UID equal to that on the drone then you are "in sync"  - there is no difference from the flight controller's point of view - and it is this fast "sync testing" capability that this proposal is targeting.

prebenfihl, 2019-10-03:
>   does such use cases make sense to you?

> Yes they do. My only comment is that I see them as uncommon.

It is a good point that most plans will be edited on site to account for the current wind conditions and such (but unless you clear the mission after your flight you will have a download of a large mission anytime you connect to a drone that has flown a large mission). But an uncommon use case that has a high level of risk (like the second use case) may be worth some effort.

prebenfihl, 2019-10-03:
> don't quite get this part? Why isn't the hash associated with the plan and stored with the plan?

The original proposal had an idea of putting the mission UID in a mission_item which would then be in the mission. Now the proposal just adds the calculated misssion UID to the mission_count and mission_ack messages, hence not in the mission (but the flight controller would still store the mission UID of the current mission).

> Also there needs to be a solution with respect to create a super-hash something or other which is the UID for the mission+geofence+rally points.

I find this part interesting. I would expect that mission, geofence, and rally points was considered three different lists of mission_items (and hence effectively just three different missions) when reading the description of MAV_MISSION_TYPE. Are you starting one mission upload as defined by the mission protocol with mixed mission_types in the mission_items or are you doing three separate mission uploads?

prebenfihl, 2019-10-03:
> Don't quite get this part either. Why does it need to only store UID after ack. The GCS is going to need to store a UID for a Plan at creation time, not at upload time. Those two steps may happen at different points in time.

The proposed mission UID is calculated directly from the list of mission_items that the plan is expanded into, so all mission_items needs to be completed before the mission UID can be calculated (it is not a plan UID which could be generated when the plan is created). Furthermore, only when the GCS receives the MISSION_ACK is it confirmed that the drone has accepted the mission and that GCS and drone are in sync. The GCS probably want to store the last known mission UID from the drone to allow for easy sync checks if the plan changes.

prebenfihl, 2019-10-03:
I have a few more comments for @DonLakeFlyer and @fnoop but I'll have to return to that...

DonLakeFlyer, 2019-10-03:
From a  QGC standpoint: 

When you connect to a vehicle the mission, fence and set of rally point are downloaded from the vehicle sequentially. They constitute the "Plan". They all affect the outcome of the flight. If you want to skip downloading **all** of these the following things need to happen:
* The combination of these three mission/fence/rally things needs to be uniqely identifed. I propose using a triplet of three UIDs.
* In order to skip downloading completely I propose a new command to query the current UID for each type.
* With that triplet I can reload the Plan from disk (if I can find it) instead of downloading from the Vehicle.
* You can't do this just from the Mission part. For example you could have two Plans on disk. The mission part of these is exactly the same. But the fences are different. If you only look for a match on mission and then reload that Plan then you are showing correct mission but incorrect fences. That is just as bad as the mission on the screen being out of sync with the mission on the vehicle.

Also from a user model standpoint it is way better to generate the UID on the gcs as well. Without that you would have to go back and modify an existing plan on disk after you do an upload. That's  a strange modification pattern in that it will futz file dates and so forth when as far as the user is concerned they didn't modify the file.

hamishwillee, 2019-10-03:
@DonLakeFlyer An original feature of the design was that only the GCS should generate the hash; the assumption being that it is hard to guarantee that both ends are generating it in the same way. The rest of my design was based on that assumption - if it isn't "true" then your proposal is obviously better. In particular it means that partial updates will work properly.

So my understanding of the above is that:
1. There are separate hashes for mission, fence, rally points.
1. The hashes can be generated on both GCS or Drone
1. There is a message like `PLAN_UID` that is broadcast by drone whenever any part of a plan changes. 
   - This either contains UID for all parts of the plan, or just one part (e.g. mission) and an identifier stating which part. I prefer the later because if a plan extends to some other type you just add a new value.
1. GCS can request the UID triplet (or parts of it) at any point using a command with `MAV_CMD_REQUEST_MESSAGE` with `PLAN_UID` and the specific id index/enum. This returns `MAV_RESULT_UNSUPPORTED` if the feature is not supported.

Not mentioned, but I might STILL extend MISSION_COUNT to include the triplet in the main sequence.

Is that where you were heading?

DonLakeFlyer, 2019-10-03:
> Is that where you were heading?

Pretty much. 

Except:
#3 I guess the broadcast thing would handle multiple writers to the vehicle. Makes sense, wasn't thinking about that possibility.
#4 The request for a UID is made using a COMMAND_LONG. That way firmwares which don't support it can nack the command. Otherwise you will need yet another capability bit. If the firmware Acks, it then also sends a PLAN_UID mavlink message with the values. Maybe more than once to work around noisy comms. If MAV_CMD_REQUEST_MESSAGE can do the same Nacking from firmwares which don't support PLAN_UID then even better your way.

> Not mentioned, but I might STILL extend MISSION_COUNT to include the triplet in the main sequence.

Certainly doesn't hurt.

> the assumption being that it is hard to guarantee that both ends are generating it in the same way.

I don't see why that's a problem. It's not any more complex then the parameter hash which is generated on both sides.

hamishwillee, 2019-10-03:
@DonLakeFlyer Sounds like we're heading towards consensus. I'd rather pay the cost of the drone calculating the hash too, as it allows us to deal with all edge cases.

> #4 The request for a UID is made using a COMMAND_LONG. That way firmwares which don't support it can nack the command. Otherwise you will need yet another capability bit. If the firmware Acks, it [...]

Yes, I am proposing exactly the model that you are - a command protocol request for the UID. This will either be acked (supported) and then the PLAN_UID emitted, or NACKED as not supported. The only remaining decision here is whether the emitted message always contains all triplets, or you individually request UIDs for each part of plan. 

Historically we'd use a dedicated command for requesting that `PLAN_UID ` message. My proposal is to instead use `MAV_CMD_REQUEST_MESSAGE` - a generic single-shot message requestor in which you can specify the message you want emitted. Note that this is implemented in ArduPilot but not yet in PX4. Should be pretty easy to add, and no harder than just adding support for a specific message.

The only point of clarification with the above  is that I never refer to using `COMMAND_LONG` - but to using the  command protocol. The nuance being that the command protocol allows you to send the request in _either_ a COMMAND_LONG or a COMMAND_INT, and generally I much prefer the COMMAND_INT (and wish the COMMAND_LONG had never been born :-) ).

DonLakeFlyer, 2019-10-03:
> The only remaining decision here is whether the emitted message always contains all triplets, or you individually request UIDs for each part of plan.

You individually request to future proof yourself against additional MAV_MISSION_TYPEs showing up in the future. If you get them all at once then PLAN_UID is locked down to current set. Or limited to as many as you can fill into the structure as it grows.

hamishwillee, 2019-10-04:
Yes, that is what I think too. I'm going to disappear for a few days on another task and to give people a chance to comment. Assuming we're all in broad agreement is everyone happy for me to create a PR for the message changes? (that is an open question to all of you). That would happen late next week. If you want it earlier, then anyone is free to do this work.

prebenfihl, 2019-10-04:
> Not mentioned, but I might STILL extend MISSION_COUNT to include the triplet in the main sequence.

If MISSION_COUNT is extended then it might be worth also extending MISSION_ACK to include the UID. It is not necessarily required because of the PLAN_UID broadcast but conceptually it fits very nicely with the mission protocol (in stead of listening for the broadcast in a separate mechanism/process).

DonLakeFlyer, 2019-10-04:
> Assuming we're all in broad agreement is everyone happy for me to create a PR for the message changes?

Fine by me

[3 more replies]