mavlink/mavlink #1175: Update docs related to AUTOPILOT_VERSION

olliw42, 2019-07-09:
so that #1156 isn't lost (I couldn't reopen it)

julianoes, 2019-07-09:
I have no opinion on that one, I'd need to read through it all.

hamishwillee, 2019-07-10:
Thanks for "reopening" @olliw42 

@julianoes The important bit is in https://github.com/mavlink/mavlink/issues/1156#issuecomment-506602488

When I started working on MAVLink the first thing I naively documented all the messages/ways of getting version info: https://mavlink.io/en/guide/mavlink_version.html#determining-protocolmessage-version

The problem is that not all the methods are implemented everywhere so you can't rely on them. 
- You can rely on HEARTRBEAT for the version number of the library (as it is auto-generated) and the  start marker byte for the protocol version of the current message (but not whether other versions are supported by the library). 
- You must be able to rely on PROTOCOL_VERSION.
- It is unclear from the docs whether you can rely on AUTOPILOT_VERSION

The action here are to:
1. Move PROTOCOL_VERSION to minimal.xml, and ideally as a follow on make common.xml split  off minimal.xml. 
2. Add note to AUTOPILOT_VERSION and to https://mavlink.io/en/guide/mavlink_version.html to make it clear that this cannot be relied on.

julianoes, 2019-08-26:
> ArduPilot has an AUTOPILOT_VERSION_REQUEST message in it’s dialect, which is used by MissionPlanner to query for AUTOPILOT_VERSION, and MissionPlanner kind of requires a component to respond to this request in order to work properly. This isn't so for QGroundControl.

I can't find AUTOPILOT_VERSION_REQUEST. I think you meant  [MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES) which is actually what we use in QGC and MAVSDK as well to connect to an autopilot.

MAVSDK currently does not implement [PROTOCOL_VERSION](https://mavlink.io/en/messages/common.html#PROTOCOL_VERSION) but probably should.
(The only problem there is that it's marked WIP and that you should not use it in production. If the message is actually required to correctly connect then a change to this message would break everything. Given it's already in QGC the WIP tag probably needs to be removed and we have to stabilize it.)

@olliw42 does MP connect to components without an autopilot (e.g. an antenna tracker)? And if so does it still require the `AUTOPILOT_VERSION` message for that?

@hamishwillee 
> 
1.    Move PROTOCOL_VERSION to minimal.xml, and ideally as a follow on make common.xml split off minimal.xml.
2.    Add note to AUTOPILOT_VERSION and to https://mavlink.io/en/guide/mavlink_version.html to make it clear that this cannot be relied on.

I think that makes sense and should not break MP as such.

olliw42, 2019-08-26:
> I can't find AUTOPILOT_VERSION_REQUEST.

it is in ArduPilot's dialect: https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L1335-L1339

> I think you meant MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES

no, I meant the AUTOPILOT_VERSION_REQUEST message used by ArduPilot/MP

> MAVSDK currently does not implement PROTOCOL_VERSION but probably should.

this would be, in my understanding, a necessary conclusion (= must)

>Given it's already in QGC the WIP tag probably needs to be removed and we have to stabilize it

this would be, in my understanding, a necessary conclusion (= must)

> does MP connect to components without an autopilot (e.g. an antenna tracker)? 

yes

and I find it actually very useful (it's one main reason I use MP a lot). It may not be that useful for the typical user, I guess, given GQC's design target it's probably OK QGC doesn't do it. MAVSDK I would however expect that it's capable of this. 

> And if so does it still require the AUTOPILOT_VERSION message for that?

not for bare connecting if I recall correctly, but it won't do anything useful for you without. I would need to retest what exactly the issue was but I think it e.g. wouldn't even start to download parameters. So, essentially, yes, MP does require it also for components

> Add note

I think one really should make also a fuller set of changes in the docs, like those suggested in https://github.com/mavlink/mavlink/issues/1156#issuecomment-506602488

> should not break MP as such

I can't see how one could come to a reasonable/satisfying spec without MP having to change it's behavior

julianoes, 2019-08-26:
> it is in ArduPilot's dialect:

Aha thanks, blame the GitHub search!

> MAVSDK I would however expect that it's capable of this.

It will work but it will complain for not getting an AUTOPILOT_VERSION quite a lot which can be confusing. If you know to ignore the error printfs you're ok.

> I can't see how one could come to a reasonable/satisfying spec without MP having to change it's behavior

1. MP could probe for both AUTOPILOT_VERSION and PROTOCOL_VERSION.
2. Existing implementations will answer with AUTOPILOT_VERSION and work as expected.
3. New compliant implementations will answer to either both or only PROTOCOL_VERSION.
4. MP can accept either or both.

The only case that does not work is a current (old) MP version with a new thing that only answers to PROTOCOL_VERSION.

olliw42, 2019-08-26:
> It will work but it will complain for not getting an AUTOPILOT_VERSION quite a lot which can be confusing. If you know to ignore the error printfs you're ok.

I think that's a very strange behavior, and I would have though that the clarification which this issue aims at should overcome this.

In my simple mind the situation is very simple:

The protocol version negotiation is (= must be) done by PROTOCOL_VERSION (which does not (!) mean that implementation can not also support alternative ways, but each must support this at minimum)

AUTOPILOT_VERSION is voluntarily, and therewith essentially obsolete for version negotiation, and therewith for the functioning of the basic protocol 

AUTOPILOT_VERSION by its design makes sense for autopilots but really not for other components (except of the cap field which was (mis-)used for version negotiation which however is now done via PROTOCOL_VERSION, so there is zero reason to make a component supporting AUTOPILOT_VERSION)

I in contrast would think that an autopilot should support AUTOPILOT_VERSION, a camera CAMERA_INFORMATION, and all other components COMPONENT_INFORMATION.

this simple minded view looks very logical to me ... and that's why I go with it, simple and some logic

>  MP could probe for both AUTOPILOT_VERSION and PROTOCOL_VERSION.

yeah ... so MP has to change it's behavior, all I was trying to say :)

julianoes, 2019-08-26:
Ok, agreed.

hamishwillee, 2020-06-22:
I'm not sure out of all of this what docs have to change :-(

is it still the original actions:

- Move PROTOCOL_VERSION to minimal.xml, and ideally as a follow on make common.xml split off minimal.xml.
- Add note to AUTOPILOT_VERSION and to https://mavlink.io/en/guide/mavlink_version.html to make it clear that this cannot be relied on.