mavlink/mavlink #1190: Support for radio pairing message set

LorenzMeier, 2019-08-01:
We need to facilitate it to pair radios through a standard implementation in the protocol.

MatejFranceskin, 2019-08-01:
I think that it should be enough to send the signal to switch into pairing mode. RC type parameter defines the method used, for example: NFC, Microhard, ...
This command could be send from AP to Pairing Manager running on companion computer or vice versa if the hardware NFC and secure element are located on flight controller.

hamishwillee, 2019-08-05:
So just add some more [RC_TYPE](https://mavlink.io/en/messages/common.html#RC_TYPE)?

My understanding of the current legacy method is that you can send this message (say, [via QGC](https://docs.qgroundcontrol.com/en/SetupView/Radio.html#additional-radio-setup)) to the Flight controller, which is attached to the receiver. For a spektrum the FC is able to put it in binding mode, so you can then manually do whatever is required on the transmitter to actually complete binding.
What I think you are proposing is exactly the same thing for other types - a message that can be sent by "some system" to tell another system that it should get ready to pair - with the remainder of the operation being manual. 

Is there any value in including additional pairing information in the message? For example, to pair a headset to my phone I tap with NFC - and I believe that this initiates Bluetooth pairing which includes the ID of the headset from the NFC communication. The user doesn't have to open bluetooth or enter any details in most cases (though in some they might have to enter a passphrase).

I don't know if you can do something similar with microhard (say) but it is obviously the less a user has to do, the better.

hamishwillee, 2020-06-22:
@MatejFranceskin @LorenzMeier Closing as stale. Reopen if it is needed and someone has been nominated as a stakeholder.

closed 2020-06-22 by hamishwillee