mavlink/mavlink #1191: Documentation rendering mixes messages with commands visually, confusing developers

LorenzMeier, 2019-08-01:
@hamishwillee Commands need a different visual style because developers do not understand that a command is not a message. They need to render into a different format.

hamishwillee, 2019-08-02:
@LorenzMeier Obvious "solution" is to add a visual identifier to every type indicating what it is - command, message, enum. This could be a prefix or suffix, and it could link to the section heading, which then explains what the contained types are. If we do this then the first one would be my preference.

[image]

Reasons not to do it are that lots of work has gone into making these as compact as possible to reduce download/load time. Adding links and new text counters that. Also, they are already pretty different, and you might argue that someone who is confused by this now will remain confused by whatever we do ... see the difference now:

[image]

hamishwillee, 2019-08-02:
Another alternative might simply be to change the font on everything in these sections. Probably less effective.

hamishwillee, 2019-08-02:
I tried this an it buggers up the menu. Might work with an icon, but not sure anything would meaningfully tell the difference between a command and message visually.

[image]

hamishwillee, 2019-08-02:
OK, so proposal below. Basically same thing, but link/text for the type is in the description. Does this work for you?

[image]

PR #1195 (2019-08-08): Add highlighting of item type
Fixes #1191

Adds links at start of body text which makes it easier to tell that something is a command or message.

[image]

If you click the link, you get taken to the top of the section. For commands you get some advisory text that explains that commands are not messages (the note below):

[image]
closed 2019-08-08 by LorenzMeier