mavlink/mavlink #1198: CAMERA_INFORMATION.cam_definition_version usage unclear

hamishwillee, 2019-08-07:
How is [CAMERA_INFORMATION.cam_definition_version](https://mavlink.io/en/messages/common.html#CAMERA_INFORMATION) expected to be used?

Our understanding is that:
- a component owner who iterates the camera definition would update both `cam_definition_version` and the same number in the definition file. 
- A GCS should check `cam_definition_version` and if the value is different to the stored XML it should re-download the cached version.
- Presumably the versioned xml files have to be available at individual URLs so that old firmware will still work ...or are we assuming that firmware will always be updated in-sync?

Correct?

The problem we see is that it is easy to update the XML file, but the version is in code, which is much harder to update and deploy -  see [discussion here](https://github.com/mavlink/mavlink/pull/1176/files#r311301451). 

Is the assumption that an update to the definition file will actually _require_ an update the component code? 

Are there other better ways of doing this? For example it might be slightly easier for a component to specify that next time message is sent it should request forced cache update (rather than having to store a specific version). 

Of course if the XML file URL has to be versioned this is "mostly" moot -
-  the component owner will have to go in code to make the URL change for any changes to the definition file.
- It does mean that the definition version is pointless - the camera can get the definition change from the changed URL

@dogmaphobic Can you please advise the intent. @olliw42 FYI

dogmaphobic, 2019-08-08:
QGC doesn't even look at the version within the XML file. That's for the camera side to handle it, if it wants to. What QGC will look for is the version defined in `cam_definition_version` within the [CAMERA_INFORMATION](https://mavlink.io/en/messages/common.html#CAMERA_INFORMATION) message. That's the version it uses to figure out if it has the file already cached or not. If it doesn't, it will fetch the file defined in `cam_definition_uri` within the same message. QGC caches the file using a file name off the camera make/model appended with the version number (`foo_bar_001.xml`).

The camera handler (at the camera side) needs to set the proper version within the message. It can be hard coded (compile time change) or if it can parse the XML file itself to find the version (run time change.)

The camera definition file itself, as far as QGC is concerned, is version agnostic. QGC will happily parse and build whatever UI is defined within the camera definition file. As long as the parameters and options (and rules, exclusions, etc.) defined within are in sync with the camera handler, it makes no difference to QGC.

In other words, the version number is important only for the camera handler. It must make sure the parameter and options defined within the camera definition file matches the parameter and options it can handle.

olliw42, 2019-08-08:
many thx for the detailed explanation
I think this is what was understood what the current behavior is
the question however is if this is the most useful behavior

I e.g. think that it should be possible to update the xml file without having the user to also update the firmware of the component, a xml update would then not require any action by the user

also, this essentially means that versioning must be done via the URI, since there isn't any way to provide different xmls for different component firmware versions (there must be different xml files located at different URIs)
if however the versioning is done via the URI anyway, then the `cam_definition_version` field and attached mechanism is totally obsolete since QGC would see that changed URI and update ...  
so, A = B <> A, and it's not clear what `cam_definition_version` and attached mechanism should be good for ...

dogmaphobic, 2019-08-08:
>I e.g. think that it should be possible to update the xml file without having the user to also update the firmware of the component, a xml update would then not require any action by the user

What _user_? QGC is not affected by a change in the XML file other than loading it and building the UI defined within it.

If a camera manufacturer needs to update an XML file, it implies a change in parameters and/or options, which must be reflected in its firmware. I can't see why a camera definition would change without a corresponding change in the firmware save for typos or other non functionality change.

>also, this essentially means that versioning must be done via the URI, since there isn't any way to provide different xmls for different component firmware versions (there must be different xml files located at different URIs)

I'm sure I follow this. You certainly have a specific case in mind but I'm not understanding it. The URI, as far as QGC is concerned is just a way to get the file. As I mentioned above, I don't use the file name. I build it based on the camera's make and model name along with the version number defined in `cam_definition_version`. Most of the time I get stuff like "definition.xml" with no other identifying information, which is totally fine given how it all works.

olliw42, 2019-08-08:
>  I can't see why a camera definition would change without a corresponding change in the firmware save for typos or other non functionality change.

bugs?
improvements such as adding `<update>`, `<exclude>`, `<parameterranges>`?
new features in the UI which only need changes in the xml?
added languages in `<localization>`?

really ??

> I'm not sure I follow this. You certainly have a specific case in mind but I'm not understanding it. 

a new firmware version may come with a new xml, which may be incompatible with an older xml. However, users still may use an old firmware version which uses this older xml. Thus, both xmls must be stored somewhere, and QGC somehow must be able to determine which of them to load. 

Currently, the only possibility for this is to give the two xmls different URIs, e.g. by adding a version to its name or storing them in different subfolders, or whatever, so different URIs.

So, versioning is done via the URI anyway, and the argument goes on as in the above.

It can't be true that most of the time you just get "definition.xml", since that's not a sufficiently complete URI to locate that file.

dogmaphobic, 2019-08-08:
>bugs?
improvements such as adding <update>, <exclude>, <parameterranges>?
new features in the xml defnition or UI which only need changes in the xml?
added languages in <localization>?

Right, as I said, _save for typos or other non functionality changes._ If you see that coming, just add code to parse the version within the XML file in order to fill in the `cam_definition_version` field. That way, you can change the XML file at any time, update its version number and the firmware will give the right version without having to have it hardcoded.

>a new firmware version may come with a new xml, which may be incompatible with an older xml. However, users still may use an old firmware version which uses this older xml.

The old XML file is giving off the right version of the file to load. That's why they are versioned. Firmware X sends X in `cam_definition_version` and new firmware Y sends a Y in `cam_definition_version`. Everybody is happy.

>Currently, the only possibility for this is to give the two xmls different URIs, e.g. by adding a version to its name or storing them in different subfolders, or whatever, so different URIs.

The firmware contains its own matching version of the XML file. Why would it have two XML files?

>It can't be true that most of the time you just get "definition.xml", since that's not a sufficiently complete URI to locate that file.

Of course not. I was just simplifying some http://192.168.42.1/somedir/definition.xml.

olliw42, 2019-08-08:
frankly, I don't think you were trying to think about what was being said. It doesn't make sense to continue, so I TO.

dogmaphobic, 2019-08-08:
All I read was your complaining about the way it is and no clear explanation for a reason it doesn't work. Can you provide a clear example of a case where the current implementation does not work? This is in use for quite a while and I have not heard from anyone about a case where it didn't work as defined above. I assume you have a good reason for the complaints. I just have not gotten them through your posts above.

julianoes, 2019-08-08:
> Presumably the versioned xml files have to be available at individual URLs so that old firmware will still work ...or are we assuming that firmware will always be updated in-sync?

So I think there are two cases here to think about:

1. The file is hosted on the camera itself. Therefore, the firmware and the xml file are updated in sync. Therefore the URL can stay the same because the GCS does not want to use an older xml file because it wouldn't match the param API.

2. The file is hosted on a server in the internet. In this case different firmware versions might be distributed and being used at the same time. Therefore, the filename needs to include some number and each firmware needs to point to the respective file online.

@olliw42 does that clarify it?

dogmaphobic, 2019-08-08:
Ha! I was only considering a setup where the GCS always has a (TCP/IP) link with the camera. Most of these settings only make sense if you have feedback, which implies a video stream, which implies a TCP/IP link. If you don't have a TCP/IP link, then it would resort to FTP over MAVLink. But yes, I assumed the definition file residing within the camera, hence, always in sync.

hamishwillee, 2020-06-22:
IMO the assumption "definition file residing within the camera, hence, always in sync." is going to be true for most cases. 

But the spec allows an URL, so this inconsistency makes me uncomfortable. As does leaving this issue open with no movement. 

@dogmaphobic Is this something you have any "ideas" for? 

If not, would it be reasonable to perhaps mandate that they must be in sync - perhaps by stating that it can only be on the vehicle and accessed via mavftp. Then we could resolve the inconsistency for now, and if someone comes along later who needs it on the wider web, then they will have to help think up a solution.

dogmaphobic, 2020-06-22:
If the file name is unique for the version, it doesn't matter if the file is within the vehicle or some server.

hamishwillee, 2020-06-24:
@dogmaphobic That's true. I guess if you're hosting on a server then it is kind of obvious that you will need unique versions, and if you're hosting on the device then things will be in sync. 

I'm going to close this. Can be reopened if someone runs into a problem while trying to use this.

olliw42, 2020-06-24:
> Can be reopened if someone runs into a problem while trying to use this.

ähm, I did raise the point (in the other issue) because I did run into a problem while trying to use this ... and I would again if would try again ... 

it is exactly as dogmaphobic says, and the only possible conclusion is that the only possible reasonable way to do the version handling (if one wants to avoid painful efforts to get the various fields in sync) is via the filename, i.e., make it unique for each version.  I e.g. simply do it by adding a _vxx.xx to the name.

But the crucial point here is: the only possible conclusion is that the only possible reasonable way to do the version handling is via the filename. ;)

hamishwillee, 2020-06-24:
@olliw42 So your suggestion would be to add a note to the version name field indicating that if the XML is not hosted on the vehicle then the filename/location must be versioned?

olliw42, 2020-06-24:
my suggestions would be
* to draw the conclusion here and elsewhere where it also applies (aka learn a lesson)
* to decide for a unique versioning method and not two or three or whatever. Given the facts this only could be via the filename.

it is possible to decide from the filename (I think we settled for http// vs mavftp//) if the xml is hosted or not hosted and thus know if to use the filename and the version or to just use the filename as such and to ignore the version, but I would argue that this only brings in complexity and potential confusion. It's much easier to program, to ensure validity, and to maintain, if it is just one way for all. Handily, the filename version thing is not only the only possible reasonable way but also the simplest to maintain and the simplest for a GCS to handle. Thus, I would deprecate the version field or mention it is useless, and only go for versioning via the filename.

hamishwillee, 2020-06-24:
@dogmaphobic I tend to agree with @olliw42 here. Would you be amenable to this change (deprecate the version number and state that the filename of the URI must be versioned?)