mavlink/mavlink #1202: Proposal for CAMERA_STATUS message

olliw42, 2019-08-10:
The MAVLink camera microservice is already quite rich, but misses features such as e.g. reporting a camera's power status.

Not all cameras can be powered independently, but need to be operated from its internal battery. A prominent example of this type would be the GoPro Hero5,6,7 cameras.

It thus would be very convenient if that battery level would be reported to the user, and e.g. warnings be issued to the user if that level becomes too low. I note that such a feature is e.g. available with the 3DR Solo in combination with Solex.

Thus, the proposal is to add a CAMERA_STATUS message which provides such information.

The battery level is the one piece of data I currently have in mind, but it is reasonable to expect that  further might be coming up over time.

Another piece of data which one obviously wants to be reported is the SD card memory status. However, the camera microservice already provides this via the STORAGE_INFORMATION and 
CAMERA_CAPTURE_STATUS capacity fields. And this is the reason why I have not raised a PR, since there is thus the issue to be sorted out of how a CAMERA_STATUS message should fit into the scheme.

For instance, one could keep the SD card memory out of CAMERA_STATUS, and ask a GCS to request that message e.g. every 5 secs in addition to the already existing protocol. Or, one could add SD card info also to CAMERA_STATUS, and give it priority over the field in CAMERA_CAPTURE_STATUS. One also could argue to reuse one of the existing messages carrying power/battery info (if the believe is that there won't likely be further status info which one might report). This are just few possibilities, there are several further approaches which one can think of. So, this proposal implies having to define this point.

Cheers, Olli

PS: To demonstrate that the point is not just academic, but that there is a MAVLink camera micorservice implementation for the GoPro Hero5 available to practical use, I'd like to share this video:
https://youtu.be/rhzkxKO-40c

edwinhayes, 2019-08-25:
It's preferable to have one canonical way of conveying a given piece of information, to keep things simple and unambiguous.  So, is there actually anything which the existing `BATTERY*` and `STORAGE*` messages are unable to convey?

olliw42, 2019-08-26:
as said before, I think it depends on the longer term perspective/expectations.

If believe is that there won't be further useful data coming up in the future one could handle this by a BATTERY* message, if however expectation is that the battery level might not be the only data one may want to transmit in some future, a dedicated CAMERA_STATUS message should be the better option, as it can be easily extended thx to MAVLink2, and keeps the service simpler since not each new data item needs adding messages to the service.

Just as example, if that argument would have been applied before to the case of the storage information, one would not have the current situation of wanting to add a further message exchange.

edwinhayes, 2019-08-27:
I disagree, in that even if there was some other information which warranted a `CAMERA_STATUS` message (bool `lens_cap_forgotten`?), I still think that battery information should be sent using the existing `BATTERY*` messages and not in `CAMERA_STATUS`.

Having fields duplicated between battery and camera messages does not "make the service simpler"; the overhead of sending an additional message is minimal compared to the overhead of trying to manage software compatibility across multiple products against a protocol which does not have a canonical way of sending of single piece of data.  This type of ambiguity is what leads to many of the current Ardupilot vs PX4 compatibility issues.

olliw42, 2019-08-27:
a camera is not a battery, and all except of one field of the BATTERY* message would be relevant, and I argue that it is exactly the approach to misuse one message, originally designed for a particular purpose, for all sorts of purposes which brings in the ambiguity causing compatibility issues, since eventually no one knows anymore which field is used in which circumstance for what purpose.

edwinhayes, 2019-08-28:
A flight controller is not a battery either, yet it sends `BATTERY*` messages because it is reporting information _about a battery_.  This applies to any other device as well; hence why `MAV_BATTERY_FUNCTION` exists.

You can't possibly say that some fields of the battery message would be irrelevant; all you could say is that _on the cameras you are using personally_ they wouldn't be relevant.  In which case, just leave them empty - that's not my argument, it's the argument _you_ just gave to me in a different thread.

In general, all the usual functions which you might want to apply to an aircraft powertrain battery also apply to the battery in the camera: you might want the GCS to monitor its level and indicate when the level gets below a certain threshold, you might want to monitor the current consumption to allow detection of a fault in the device, you might want to monitor the temperature of the battery if you are operating in a hot climate.

As far as I can tell, arguing against using the existing battery messages means one of:

- You think that some of the fields and/or functionalities exposed by the battery message are not valid in the case of _any_ camera; or
- You think that these functions may be valid _for some cameras_, but should be implemented using a parallel mechanism because for some reason that's easier; or
- You think that in general MAVLink should create dedicated messages which express exactly the functionalities which you think are common in the most concise way possible, rather than creating generic messages which can be used in aggregate to convey different combinations of functions.

Which of these do you think has merit?

olliw42, 2019-10-10:
but as said, I really just would love to see battery charge info on the GCS screen. I now would be more interested in having this moving forward, i.e., a decision is been made. 

Certainly, if it's going to be the BATTERY_STATUS I will think that it's a mistake, but I also will think that it doesn't really matter what I think.

So, guys:

[ ] BATTERY_STATUS
[ ] CAMERA_STATUS

simple yes/no vote, please :) :) :)

edwinhayes, 2019-10-11:
> A battery on a flight controller is not like a battery on a camera (or any other component in general).

In what way?

> The batteries themselves certainly share many points, but the actions taken on pieces of data may be totally different and hence the data may be different.

_Actions_ have nothing to do with it: this message is for conveying state, not invoking an action.  So no, the data to be conveyed is not different _because_ the actions you want to invoke are different.  If the data to be conveyed is different, that's because there is some fundamental difference in the nature of the data, regardless of what you're doing with it.

> When you go through MAVLink then there is one pattern which repeats itself sticking out quite obviously: Messages are drafted with fields for some particular purpose, like for an autopilot.

I agree that this was something that happened in the past.  Actually, I think your own sentence actually illustrates part of the problem: "for an autopilot" is _not_ an example of "a particular purpose"!

The heartbeat/mode issue is one of coupling: data regarding the presence of a MAVLink component on the bus, and its operational _health_ should never have been coupled with information about the state of a navigational control loop, since it fails the easy test for coupling:
do all things which send (a) also need to send (b), _and vice versa_.

But the solution to this moving forward is for new protocol changes to enforce good coupling principles, not add more badly-coupled definitions:

Inclusion of battery information in a message conveying camera information also obviously fails this same test: do all things with battery information have camera information - no; do all things with camera information have battery information - no.

> So, what one actually would need is a clear definition of that enum for each component type, however component-specific enums are not existing in the MAVLink concept. And this was just for enums, you also find that pattern for other field types. Ergo: Over and over the same mayhay.

This line of thinking presupposes that a given MAVLink message conveys information about a specific _component_, which is not true.  Each MAVLink message conveys information about a specific logical function, the mixture of functions which are grouped into a particular component is not fixed.

So no, you don't need enums for each _component_ type, you need enums for each _function_, and that is what MAVLink has.

> here I very much disagree.

Ok then, provide an example.  What function might you want to provide in the case of a powertrain battery, but you would _fundamentally_ never want to provide in the case of a camera battery?

> This may be true if seen as isolated entity, but it becomes untrue if that entity is seen as an entity embeeded/functionally integrated into a larger entity, and not just an addendum to that larger entity. 

What does this even mean?

> For that very reason I was advocating in the BATTERY_STATUS thread that the BATTERY_STATUS message should be scaled back to it's primary purpose of being a report of that isolated entity called battery,

_What_ isolated entity called battery?

> I think that in general MAVLink should create messages which don't lead to situations like those described, and one way towards that is to logically separate what is logically different, which can mean that a message is not maximal generic in the sense that it is usefully to every and any entity but doesn't mean that the message wouldn't be generic in the sense that it wouldn't be useful for a broader class of entity types.

Yes, I agree, but the camera/battery case falls into the logically different category.

In addition I think that the various classes of components as reflected by the component ID or MAV_TYPE give some hint towards which entities might not logically fall into one set. That is, it has some appeal to me that every class may have it's (properly designed) XXXX_STATUS. Which doesn't mean that there couldn't be many more XXXX_STATUS than there are such classes.

I somewhat agree: messages should be separated by logical function, which is _commonly_, though not always, consistent with the range of publicly defined component IDs.  If there is a component ID allocated for 'CAMERA' and you can make a clear case that there is some status information which is generally relevant to components which call themselves CAMERAs _and not to other types_, then it makes sense for that information to be encapsulated into a `CAMERA_STATUS` message.  But battery information is not one of those things!  Neither is storage information.

> but as said, I really just would love to see battery charge info on the GCS screen. I now would be more interested in having this moving forward, i.e., a decision is been made.

In this case, you should _definitely_ be in favour of using the existing battery status message, because then what you want is already being supported!

olliw42, 2019-10-11:
> this message is for conveying state, not invoking an action.

I think this condenses where we fundamentally differ. 

Your statement is not true. If the message would have been used to ONLY convey a state but in no means to also invoke an action (and all actions e.g. be based on SYS_STATE in case of battery) then there wouldn't be any issue and we all would be happy. The truth and the problem was and is that it is not used that way, to just convey a state ... and apparently wasn't drafted that way when one is looking at how the states were speced.

Generally, states can incur actions, and actions can be based on states.

Therefore, the states need to be drafted properly, in the sense that it is well defined in which situations they occur or what a state means or what actions may result. If states are badly defined or arbitrarily used they become irrelevant or create mayhay.

Or, it should be made clear by spec that a state must not be used to invoke an action, in which case all is fine too.

This IMHO is so obvious, logical and consistent, and IMHO many of the present day issues can be traced back to violating this. And btw that's also how all reasonably successful specs in this world are drafted, they define states through actions (if a state invokes actions).

MAVLink is great, and has become a corner stone, and it deserves so. However, clearly, as a specification it doesn't do well, it is actually largely failing here. And what I describe is part of the reason.

> Ok then, provide an example. What function might you want to provide in the case of a powertrain battery, but you would fundamentally never want to provide in the case of a camera battery?

in the case of a battery on a flightcraft (which is looked at by the autopilot, which is what I phrased as "battery on an autopilot) I would want to be alerted very soon and the whole vehicle to return back or move itself into a safe position before the battery dries out, in case of a camera battery I would be happy if I get alerted and the camera shut off in the latest moment, but I certainly would not want whole vehicle to return back or move itself into a safe position just because the camera battery dried out ... to me this look like VERY different actions taken ... not sure I had to give that example

> In this case, you should definitely be in favour of using the existing battery status message

Isn't this exactly what I did ? :) To realize that anything else doesn't seem to lead to anywhere, and thus would be happy to go with it?

But: I don't do so for better insight but because it's the least resistive path. I would do so even though one would just repeat the very same mistakes of the past having lead to present day issues. The quick path is sometimes but most often not the best path. You won't win any chess game with that philosophy. So, the reason why I would be now happy to go with the battery status message are in fact a manifestation of why MAVLink fails as a specification.

> because then what you want is already being supported!

unfortunately this isn't so. This is not yet supported in e.g. QGC.

Because it's not in the spec. That is, one first - here and now - needs to decide, yes the BAT_STAT should be used in the camera micro service to convey the camera battery status, and when in the next step one could implement this into e.g. QGC. 

(the alternative would be to first implement it in e.g. QGC and when tell the spec it has to follow ... while this might have happened in the past, I think we can agree that this shouldn't be the right order of events, at least this order doesn't fell right to me, and that's why I'm here with this thread)

olliw42, 2019-10-11:
anyway, the previous two posts just underpin all of the previous, so let me pl repeat:

So, guys:

[ ] BATTERY_STATUS
[ ] CAMERA_STATUS

simple yes/no vote, please 

:) :) :)

edwinhayes, 2019-10-14:
> Your statement is not true.

My statement _is_ true.  You misunderstand how messaging protocols such as MAVLink work.

The purpose of status messages is _solely_ to convey information, not to invoke actions on the recipient.    The fact that the recipient _may_ decide unilaterally to perform some action based on the status that is receives is a totally different matter.

On the other hand, an example of a message which _is_ intended to invoke an action is `COMMAND_LONG`.  In the case of these messages, the decision to perform the action is not up to the recipient; the message itself invokes the specified action.

> Therefore, the states need to be drafted properly, in the sense that it is well defined in which situations they occur or what a state means or what actions may result.

No.  They don't.  This is bad design.  States encode states, not actions, and should be path independent.

> Or, it should be made clear by spec that a state must not be used to invoke an action, in which case all is fine too.

States do not invoke actions.  If a component decides to take action because of the state something reports, that is a totally different thing which is not related to the underlying protocol.

> in the case of a battery on a flightcraft (which is looked at by the autopilot, which is what I phrased as "battery on an autopilot) I would want to be alerted very soon and the whole vehicle to [...]

This is _not_ an example.  Your example just says that _you_ don't want a setup whereby the aircraft returns to home when the camera battery runs low; not that it's fundamentally impossible for someone to ever want this behaviour.

> unfortunately this isn't so. This is not yet supported in e.g. QGC.

Active work: https://github.com/mavlink/qgroundcontrol/pull/7575

> Because it's not in the spec.

Yes it is. Refer: https://mavlink.io/en/messages/common.html#MAV_BATTERY_FUNCTION

> simple yes/no vote, please

There is nothing to vote: the spec already includes a sane mechanism for reporting the battery status of devices onboard the aircraft which have batteries, so nothing needs to be done except perhaps to add an enum `MAV_BATTERY_TYPE_CAMERA`.

olliw42, 2019-10-14:
> The purpose of status messages is solely to convey information

frankly: LOL. I can't believe you're saying this. The spec is full of "actions".

> Active work: mavlink/qgroundcontrol#7575
> Yes it is. Refer: https://mavlink.io/en/messages/common.html#MAV_BATTERY_FUNCTION
> There is nothing to vote: 

I think you totally lost sight of what I'm trying to propose here. I'm talking about extending the camera micro service. Not to duplicate totally different stuff.

I also think it is pointless to continue the discussion and this thread, it takes too much bandwidth for zero result, so and I'm going to close it. IMHO this thread is an excellent example for why in this area the open source things are lagging so much behind. They are fun but lag.

closed 2019-10-14 by olliw42