mavlink/mavlink #1208: MAVLinkFTP: some (serious!?) issues

olliw42, 2019-08-16:
I found several issues with MAVLinkFTP. 

Some of them are in the docs, so I'm not sure I should have gone to mavlink-devguide , but they also touch implementations, so would probably better located yet elsewhere, so I though I come here to have it at least in a central place. Not ideal, agreed, but generally I believe that I have done my service to the community with reporting, right ;)

**1. Crc32 implementation in PX4 ok, but in Arduilot & MissionPlanner wrong!**

The PX4 reference implementation uses a crc32 which comes from the standard lib <crc32.h>. I couldn't explicitly test but there thus hardly can be doubt it's correct. 
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_ftp.cpp#L37
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_ftp.cpp#L865

In contrast, MissionPlanner and ArduPilot are using an implementation, which is quoted as "crc32 from Gary S Brown", but it does NOT produce the same results as all other Crc32 codes I found on the web and which I've tested, including those which also refer to "Gary S Brown". The used table is identical to that on the web, the polynomial code too, but the input and output crc are NOT! If I take any of the crc32 codes on the web and modify them accordingly they produce the same (incorrect) crcs as with the ArduPilot/MissionPlanner code.

From looking at the code, QGC appears to not use CalcFileCRC32, so nothing to say here.

Gary S Brown crc32 on the web:
 https://ess.cs.tu-dortmund.de/lockdoc-elixir/freebsd-lockdoc/lockdoc-v12.0-0.2/source/stand/libsa/crc32.c
http://www.creytiv.com/doxygen/re-dox/html/crc32_8c_source.html

ArduPilot crc32 working with MissionPlanner:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Math/crc.cpp#L121-L177
https://github.com/ArduPilot/ardupilot/blob/9ba11dd6b0b74388194589927a69e04f7765e9b9/libraries/GCS_MAVLink/GCS_MAVFTP.cpp#L704-L727

In short, all crc32 implementations I took from the web throw an crc exception when I use MissionPlanner for MAVLinkFTP file read, while when I use the ArduPilot crc32 implementation with incorrect input/outputs it works.

PS: I think the docs should more clearly say WHICH crc32 is to be used, given that e.g. ethernet and unix crc32 appear to be different.

**2. Nack Error codes in documentation shifted by one**

https://mavlink.io/en/services/ftp.html#error_codes

They should start from 0, not from 1. At least all implementations (PX4, QGC, MP, AP) do so.

**3. Reading a File operation incorrectly described in docs, OpenFileRO wrong**

The docs for OpenFileRO says "An ACK packet must include the allocated session and the data size of the file to be opened (size)" (https://mavlink.io/en/services/ftp.html#opcodes: ), and in the sketch for Reading a File further below it shows "ACK( session, size=len(file) )" (https://mavlink.io/en/services/ftp.html#reading-a-file) . 

This hardly can be correct given that the file size can be larger than the uint8_t of the size field, and indeed this is also NOT how it's implemented in PX4,AP,MP, where the file size is returned in the data field with size = 4.

for Px4 e.g.: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_ftp.cpp#L536-L537

Have fun. :)

hamishwillee, 2019-08-16:
Thanks @olliw42 - I have fixed item 2 - as you say the NAK codes were off by one :-( https://github.com/mavlink/mavlink-devguide/commit/c786431fdd14e45fa3d99c0af10cadd690d21958

hamishwillee, 2019-08-16:
@olliw42 W.r.t. 3 thanks for pointing that out. 

So from code it looks like the size of the file is returned in the ack in the data field (rather than in size field). So changes would be like:
```
ACK( session, size=len(file) 
```
To
```
ACK( session, data=len(file) )
```
Is that your understanding of the problem and fix?

olliw42, 2019-08-16:
ACK( session, size=4, data=len(file) )

> Is that your understanding of the problem and fix?

yes

olliw42, 2019-08-16:
btw, when you are at it: the offset in ListDirectory is not an offset into a file buffer, like in read/write, but merely an index into the list of names, i.e. offset = 0 is the first file in the directory, offset = 1 is the 2nd, and so on. A word like "seek" e.g. is highly confusing.  As a corollary, the buffer isn't filled up to max size, but only up to how many filenames+char+size+'\0' fit into it.

hamishwillee, 2019-08-16:
Thanks. Re 3, this is now tidied in https://github.com/mavlink/mavlink-devguide/commit/06df40d74b28a0cfc153193d406073a68b48e437

hamishwillee, 2019-08-16:
> btw, when you are at it: the offset in ListDirectory is not an offset into a file buffer, like in read/write, but merely an index into the list of names, i.e. offset = 0 is the first file in the [...]

So what it says now is:

> List files and directories in `<path>` from `<offset>`.
- Opens the directory (`path`), seeks to (`offset`) and fills the result buffer with NULL-separated filenames (files also include tab-separated file size) and directory names. Sends an ACK packet with the result buffer on success, otherwise a NAK packet with an error code.
- The directory is closed after the operation, so this leaves no state on the server.

So the interpretation I made here was that basically on the system getting the request you'd run a command to list all the files and directories. This would deliver them as big string (NULL separated) - or you'd create such a string from them. Then you'd seek within the string to an offset and send the chunk requested.

Can you confirm that you're saying offset the file/directory index. So you'd call this with index from 0 to n, each time getting a new file or directory. Eventually you'd get an empty string so you'd know the job was finished?

olliw42, 2019-08-16:
> So the interpretation I made here was that basically on the system getting the request you'd run a command to list all the files and directories. This would deliver them as big string (NULL [...]

this was exactly how I was initially interpreting the description, but this is NOT how it works in the implementations PX4,AP,MP

> So you'd call this with index from 0 to n, each time getting a new file or directory. 

no, it starts with asking for offset 0. The code may then put into the data e.g. 3 null-terminated entries, so the next call will be with offset 3. The code may then put into the data e.g. only 1 null-terminated entry, so the next call will be with offset 4. The code may then put into the data e.g. 5 null-terminated entry, so the next call will be with offset 9.  And so on.

>Eventually you'd get an empty string so you'd know the job was finished?

this could have been a possibility but that appears not to be how it's done. I would have to look again into the QGC and MP codes to see what the exact termination is, but I concluded to do it so: Eventually the offset will be the number of entries, which indicates that all entries have been fetched, and I return a NACK with EOF. This works with MP. QGC I couldn't test for obvious reasons.

olliw42, 2019-08-16:
and yet another btw, when you are it: The null terminated entry does not have the format 

name \t size in BCD \0 

but in fact 

type-char format name \t size in BCD \0

where type-char is 'D', 'F', or 'S'.

julianoes, 2019-08-16:
FYI @MatejFranceskin.

julianoes, 2019-08-16:
@olliw42 is the one we're addding to MAVSDK correct?

https://github.com/mavlink/MAVSDK/pull/774/files#diff-1b2f890b3c0c8b6934d0f16939eb0aa7

olliw42, 2019-08-16:
oh ... you're putting some responsibility on me LOL

I haven't looked up all of your code, but this is what I'd want to say:

the Crc32::add() function is exactly the one used by ArduPilot. The crux is, it can be made to work correctly iff (if and only if) the right crc input is set and the crc output is properly converted. In the way it is used it will however NOT produce the correct result (iff "correct" means in agreement with crc32 codes from the web).

Pl compare to the links I gave in the above. I reproduce one for your convenience:
```
	crc = ~0U;
	while (size--)
		crc = crc32_tab[(crc ^ *p++) & 0xFF] ^ (crc >> 8);
	return (crc ^ ~0U);
```
the 2nd and 3rd lines are identical to yours, but note the first and last lines

since you don't want to rely on me here, my suggestion would be that you take some of the crc32 codes on the web and test them against it yourself ... :)

olliw42, 2019-08-16:
btw, you might have guessed already, but just to say it, the next STorM32 alpha version due soon will have a MAVLinkFTP server to download the camera definition files. I believe to remember that you once said you might want to test this one day ... it is coming :)

julianoes, 2019-08-16:
> correct I think you should notice that your SDK as is doesn't work with PX4

However, it was tested against it and it worked so that's interesting. We'll look into it.

> STorM32 alpha version due soon will have a MAVLinkFTP server to download the camera definition files.

Nice!

olliw42, 2019-08-16:
> However, it was tested against it and it worked so that's interesting. We'll look into it.

interesting, indeed.

So. let me clearly recite the experimental fact I'm resorting to (I mentioned it in the above but maybe I should have been clearer): 

When I use the ArduPilot crc32, or a web variant where I fake the input/output, when my implementation works just fine with MissionPlanner. When in contrast I use any of the crc32 codes from the web, MissionPlanner complains with a crc exception. (well, I tested just 5 different web crc32's ... but this made me sufficiently confident the finding is general). Moreover, there is a clear difference in ArduPilot's crc32 code to those on the web.

olliw42, 2019-08-16:
@julianoes I just spend some more few minutes on your SDK code, and I really can't say I comprehend it, but this line https://github.com/mavlink/MAVSDK/blob/d9be07e64d174c7a71c92430dc31260ca34492b6/example/mavlink_ftp_client/mavlink_ftp_client.cpp#L157-L159 makes me want to say this:

For reading files MissionPlanner doesn't adhere to the spec as described in the docs in that it does NOT use a returned NACK with EOF as the marker for complete reading, but a line if(header.offset+header.size >= filesize). 
https://github.com/ArduPilot/MissionPlanner/blob/master/ExtLibs/ArduPilot/Mavlink/MAVFtp.cs#L1276-L1280

 I find that very unfortunate, as I find the spec's method to be significantly better.

as naive as I am, the cited line looks suspiciously similar to MP's approach. If that is so, I would encourage you to check if the spec's NACK-EOF approach might not be what you want to do.

julianoes, 2019-08-16:
Thanks @olliw42 I'll forward that to @MatejFranceskin.

olliw42, 2019-08-19:
I found this file which is in Nuttx: 
https://github.com/PX4/NuttX/blob/px4_firmware_nuttx-7.29%2B/libs/libc/misc/lib_crc32.c
https://gitlab.com/nuttx/NuttX/blob/master/libs/libc/misc/lib_crc32.c

it indeed uses the crc32 as you, MP & AP do, without the first and last lines of the code snippet 
this may explain why you found that the test against PX4 worked 
(and my statement that PX4's use of <crc.h> is evidence of it's correctness may be wrong !!!)

the confusion as regards to what the correct crc32 really is, is increasing ... I would find it strange if the majority of web sources would have this wrong ...

it appears to not be easy to retrieve the original code of Gary S. Brown, I found this
https://opensource.apple.com/source/xnu/xnu-792.13.8/bsd/libkern/crc32.c
also wikipedia (knowing it's a source with potential flaws, yet...) has it
https://en.wikipedia.org/wiki/Cyclic_redundancy_check
also some popular zips seem to have it
http://networkdls.com/Software/View/CRC32/
and so on ... indeed many if not most seem to have it ... Nuttx appears to be rather an exception ...

hamishwillee, 2019-08-20:
@olliw42 Re https://github.com/mavlink/mavlink/issues/1208#issuecomment-521909605

So to confirm (again). 
The difference is to the current text is that the offset is to the beginning of a file reference (not just anywhere in a string representing the file list) and that it must return the data for one or more complete files (null-terminator-separated).

So as you indicated you'd call with offset 0, count the number of retrieved files, then call at the new offset. Repeat until recipient NAKs you with EOF. 

Re https://github.com/mavlink/mavlink/issues/1208#issuecomment-521921774

```
type-char name \t size in BCD \0

where type-char is 'D', 'F', or 'S'.
```
Excuse my ignorance, but what is BCD, and what is "s" (I assume D = directory, F=File"

If you have a real example that would be awesome.

olliw42, 2019-08-20:
first off, I cannot comment on "must", I haven't analyzed the PX4/MP/AP codes in sufficient depth to tell exactly what all cases would be accepted. I can tell what it is NOT, what I did to get it working, and what my best understanding of the PX4/MP/AP codes is.

> The difference is to the current text is that the offset is to the beginning of a file reference 

no. I understand this to mean that if the databuffer would be thisisfile1.xml\0thisisfile1.xml\0 that the offset to the first file would be 0 and the offset to the second file would be 16. The offset to the first file is 0, but the offset to the 2nd file is 1, since it's the 2nd file. The offset to the 3rd file would be 2, and so on. It's simply an index into the list of filenames.

> and that it must return the data for one or more complete files (null-terminator-separated).

if it's a must I don't known, but this is what I do.

> So as you indicated you'd call with offset 0, count the number of retrieved files, then call at the new offset. Repeat until recipient NAKs you with EOF.

this reads correct, iff the call at the new offset is the index as explained before, and not the position into the data.

NOTE: for ListDirectory the NAK-EOF is crucial to find the end of the list. For ReadFile the spec also says that the end of the file should be determined from a NAK-EOF, but here it can also be determined from offset+data_size >= filesize, since the filesize was transmitted with the OpenFileRO. And this is exactly how MP does it, and also MAVSDK, which is SSSSAAADDDD

> Excuse my ignorance, but what is BCD, and what is "s" (I assume D = directory, F=File"

BCD just means the integer value as string, e.g. 1234 -> "1234" (https://de.wikipedia.org/wiki/BCD-Code)

no idea what 'S' exactly is, in the code the comments say 'skipped'. It's probably to mark dict entries like '.' and '..' and tell that they should not be displayed. It's not clear to me why they are then not just not transmitted, but ...

BTW: I'd like to mention that I actually LIKE the fact that for ListDirectory the offset is NOT what the current text says but just the index into the list, makes it much easier to code (but makes a NAK-EOF mandatory).

>If you have a real example that would be awesome.

not sure what exactly you would like to see here

hamishwillee, 2019-08-20:
Firstly, thanks, that is helpful. Secondly these docs were created entirely by inspection of PX4 and QGC who invented the format way in advance and therefore to my mind define the "standard". It's an opaque payload so anyone else can do what they like. That said, I suspect the intention was to iterate to a point and then define enums/messages that were not opaque ... and that last bit never happened. Upshot, if we can come up with better approach and get it in PX4, QGC and MAVSDK, then I am pretty sure it will be fine for ArduPilot to follow. 

>> If you have a real example that would be awesome.
> not sure what exactly you would like to see here

What I meant was a real example of what the string would look like. That would answer questions for me like whether the integer

olliw42, 2019-08-20:
> Secondly these docs were created entirely by inspection of PX4 and QGC 

well, the docs do not reflect the code behavior of the code as of today

> if we can come up with better approach and get it in PX4, QGC and MAVSDK

I think that - at least as regards the parts I did touch myself - what they do is VERY OK!  Really. No complains from my side.

(except of the minor point that they don't detect end of file by NACK-EOF but the length calculation, but this should be easy to correct, or easy to adapt the docs, depending on what will prevail)
(and of course the crc32 thing, but that's not a weakness, just a potential bug)

> That would answer questions 

your questions are easy to answer without:

> whether the integer is a Byte

again, I just can tell what I do: I do send the size as a number string, i.e. 1234 becomes "1234"

> includes string like MB in it.

I don't do this, I send the size in bytes. 

From the code I would not think adding MB would work.

>  Also whether the space you put after F is really there or not. 

ah... no, there are no blanks at all, in the above I just put them in for clarity

> I  think it will look something like:

nearly, except of the blank after the first F, which isn't there. So, this would be a real enough example

Fthisisfile1.xml\t34565\0Fthisisfile2.xml\t56754764756778\0

if the GCS receives that, it would increase the offset by two.

So, even more real example. Assume five files named

TestFile1.xml
TestFile2.xml 
TestFile3.xml
TestFile4.xml
TestFile5.xml

Now assume that upon the first call by the GCS with offset 0 , the drone would send back

FTestFile1.xml\t12345\0FTestFile2.xml\t55\0

then the GCS would send out a next call with offset 2. Now assume that the drone would send back

FTestFile3.xml\t22\0FTestFile4.xml\t768\0FTestFile5.xml\t11111\0

then the GCS would send out a next call with offset 5. The drone would see it's EOF since  5 >= number of files and send back a NACK-EOF.

quite simple actually.

hamishwillee, 2019-08-21:
Thanks very much @olliw42 - I have captured this in update https://github.com/mavlink/mavlink-devguide/pull/202 - waiting on review. 

If there are any other incorrect explanations please let me know. 

@julianoes Can you please keep me informed of what is happening for file reading re detecting end of operation. NOte that the docs already indicate using EOF is the right way to do this.

julianoes, 2019-08-21:
> @julianoes Can you please keep me informed of what is happening for file reading re detecting end of operation. NOte that the docs already indicate using EOF is the right way to do this.

No idea. I'll pass that on to @MatejFranceskin.

MatejFranceskin, 2019-09-17:
@olliw42 Thanks for pointing reviewing the code and apologies for not getting at it earlies. I was absent for a long time.
The specification for EOF says "Offset past end of file for ListDirectory and ReadFile commands."
I understood that you would get this error if you try to read past the end of file and not when you read exactly to the end of file. So instead of reading until "past" the end we stop at the end. Exactly on the last read without trying to push over the limit. This way we avoid exchange of two messages. What do you think - does it make sense?

olliw42, 2019-09-18:
@MatejFranceskin 
I'm not totally sure I understand what you're asking me. 

In principle any mechanism which works makes sense to me, the only problem I see is that whatever the spec is it should be documented so and when be adhered to in implementations.

I personally wouldn't be concerned about one or two more messages, FTP will typically be an exchange of many messages anyhow so one or two more shouldn't matter much. I would happily trade one/two messages for a more clean/comprehensible/reliable protocol.

I think the EOF makes totally sense, and if only for symmetry reasons. I mean, it is needed in ListDirectory, so it would feel strange to have a different mechanism for ReadFile, and also - and maybe especially also - because this EOF is so well known from all the other places like C, python, ....

hamishwillee, 2019-11-13:
What is still open in this?

olliw42, 2019-11-13:
point 1 !!!!

there are two crc32 around, one "correct" and one "incorrect", both referring to Gary S Brown as reference/source, but only the "correct" one is - by all accounts -Brown's, while the other appears to have a mistake, which however is widely copied. 

The spec refers to only crc32 without details, which - by all accounts - would refer to the correct one, however the "incorrect" one appears to be mostly used so far in MAVLink world (it will be interesting to see what QGC is using).

Obviously, the one wouldn't work with the other and vice versa.

hamishwillee, 2019-11-14:
OK, we should first specify the best/correct definition in a canonical way here https://mavlink.io/en/services/ftp.html so at least people aren't guessing. 

What's the best link for describing the algorthm in the line ?
```
CalcFileCRC32 | Calculate CRC32 for file at <path>.- Sends an ACK reply with the checksum on success, otherwise a NAK packet with an error code.
```
Ie "CRC32 algorithm (Gary S Brown) as implemented here. "

@MatejFranceskin @julianoes Do we have a test that verifies this all against QGC and which in theory ArduPilot could run against?

olliw42, 2019-11-14:
I'm not sure what the beast approach would be to do.

It certainly would be helpful if the doc would be clear and specific. The line "CRC32 algorithm (Gary S Brown) as implemented here." however is IMHO highly conflicting. First, all the implementations basically claim to be ""CRC32 algorithm (Gary S Brown)", yet differ. Second, it's actually difficult to trace what Gary S Brown's crc32 actually is. I couldn't find a source which I could directly link to him (like a publication or so). I "only" have the empirical observation that one more often finds "Gary S Brown" in connection with the "correct" crc32, and that the farer back one goes into history the more often one finds the "correct" crc32, and the oldest reference I could find was the "correct" crc32. So this part of the argument is somewhat anecdotical and not 100% factual.
The cleanest solution would probably be to write down the algorithm explicitley, and to mention that one needs to check carefully because not any routine, even not OS routines, being called crc32 are guaranteed to do all the same.

From a practical point, I think there is no free choice anymore, i.e., things have converged towards the "incorrect" crc32. It is used by MissionPlanner, by ArduPilot, and even the nuttx lib function used in PX4 appears to be the "incorrect" crc32, and I would conclude it is also used by MAVSDK ...

hamishwillee, 2019-11-15:
OK, in that case we should identify the specific libraries we have actually used, whether correct in terms of the original definition. @JonasVautherin @MatejFranceskin Can you confirm what libraries PX4, MAVSDK and QGC used for generating the CRC check in MAVTP - ie URL or similar we can point to?

MatejFranceskin, 2019-11-15:
MAVSDK uses the same implementation as PX4 so that they're compatible.
QGC doesn't use the function at all.

hamishwillee, 2020-06-22:
@JonasVautherin Can you confirm what library you use for the CRC calculation as per this thread. Hoping to close this issue by specifically identifying the implementation used and linking to it in the docs.

JonasVautherin, 2020-06-22:
I honestly don't know :see_no_evil:. @MatejFranceskin did you rely on the PX4 source code, or some PX4 documentation?

MatejFranceskin, 2020-06-22:
I relied on PX4 source code.

hamishwillee, 2020-06-23:
@olliw42 Its been a long time and I've lost track of this.

Am I correct that the CRC32 we are talking about is the [packet checksum](https://mavlink.io/en/guide/serialization.html#checksum)? This appears to say 

> The checksum is the same as used in ITU X.25 and SAE AS-4 standards (CRC-16-CCITT), documented in SAE AS5669A. See the MAVLink source code for the documented C-implementation.

What I think you're suggesting is an additional note that the documented C-implementation uses crc32  from the standard lib `<crc32.h>`.

Or am I completely lost?

olliw42, 2020-06-23:
it's been a long time for me too :)

what you mention is IMHO the crc for a MAVLink message, the crc under discussion here is the crc as used in MAVLink ftp, https://mavlink.io/en/services/ftp.html. It's probably a too simplistic statement, but the first is a crc16 and the other is a crc32.

hamishwillee, 2020-06-23:
Duh! OK, I think fixed in https://github.com/mavlink/mavlink-devguide/pull/258 - please comment!

olliw42, 2020-06-23:
This "with the same CRC32 algorithm as is used in the C standard library - **crc32.h**" is exactly what I cannot tell with competence! I can not even tell if such a statement can even be made, i.e., if there is such a  universal thing like "standard crc32 library". 

From the scan I did I got some doubts. But again I'm not competent to make strong statements (I only can observe in that matter)

One maybe could write "with the same CRC32 algorithm as is used in the C standard library used by the PX4 firmware"

julianoes, 2020-06-23:
I agree, we need to properly document what is to be used. I will follow up.

hamishwillee, 2020-06-23:
I am closing this. Discussion carried on in https://github.com/mavlink/mavlink-devguide/pull/258

closed 2020-06-23 by hamishwillee