mavlink/mavlink #1209: MAVLinkFTP: suggestion to add FileNotExisting to error codes

olliw42, 2019-08-17:
I suggest to add a NACK error FileNotExisting to the 'standardized' list of errors (https://mavlink.io/en/services/ftp.html#error_codes).

Reasons: This is obviously a very possible common error in OpenFileRO, RemoveFile, RemoveDirectory, TruncateFile, Rename, CalcFileCRC32. The lack of a FileNotExisting error leads to non-standard-esque, pretty hacky implementations.

For instance, currently a non existing file is reported by PX4 as a os system error code via FailErrno, which in MAVSDK leads to a code line like 
```
        // PX4 Mavlink FTP returns "File doesn't exist" this way
        if (sr == ServerResult::ERR_FAIL_ERRNO && payload->data[1] == ENOENT) {
            sr = ServerResult::ERR_FAIL_FILE_DOES_NOT_EXIST;
        }
```
(https://github.com/mavlink/MAVSDK/blob/d9be07e64d174c7a71c92430dc31260ca34492b6/src/plugins/mavlink_ftp/mavlink_ftp_impl.cpp#L133-L136)
in order to catch it and translate it into something useful. Obviously very implementation specific and not very standard-esque.

Also note that, somewhat ironically, there are errors for e.g. FileExists and FileProtected even though these equally well could have been handled by a os system error code via FailErrno, which suggests that not having also a FileNotExisting is more of an oversight than design.

This of course would affect existing implementations, but since MAVLinkFTP seems to be not widely adopted yet it's probably not very harmful. 

I certainly have seen that the standard says

> The encoding and content of the payload field are not mandated by the specification - and can be extension specific

and that therefore anyone can do whatever they want, invalidating any "standardize" argument. But I don't think that's upholdable nor smart. I mean, it would be obviously most sad - from a user perspective - if even the most likely combinations MissionPlanner/QGC<->PX4/ArduPilot wouldn't interoperate.

julianoes, 2019-08-19:
I think that addition would be reasonable.

hamishwillee, 2019-08-19:
Excellent idea. IMO the protocol should have been redone as non-opaque and making these enums, but whatever. The upshot though is that there is no more formal way for me to spec this change in messages. 

I have created https://github.com/mavlink/mavlink-devguide/pull/199 - what is best way for it to be named?

Before I push that "live" I'd like to see it in PX4 and/or QGC. @julianoes thoughts?

julianoes, 2019-08-19:
What do you think @MatejFranceskin?

hamishwillee, 2019-09-04:
@julianoes Can you go find @MatejFranceskin and see if we can get a canonical answer on this? 

I absolutely think this should go in. My concern is that if I update the docs and this is not done by the only "live" implementation it is a bit rubbish.

MatejFranceskin, 2019-09-04:
Yes, I agree. I will implement it.

hamishwillee, 2019-09-04:
THanks. I have merged the devguide update to match that. @julianoes, I guess MAVSDK should also now handle this error :-)

closed 2019-09-04 by hamishwillee

julianoes, 2019-09-04:
> I guess MAVSDK should also now handle this error :-)

@MatejFranceskin is taking care of it :smiley: