mavlink/mavlink #1210: Microservice documenting canonical behaviour for POSITION_TARGET messages.

edwinhayes, 2019-08-19:
There seems a lack of documentation regarding how the `SET_POSITION_TARGET` and matching `POSITION_TARGET` messages are supposed to function.  In particular, when is it expected for the getter/response form of the message to be published?  Touched on by #473 & #495, but can't find much since then.

My (optimistic) proposal to remedy this is by documenting a new micro-service which formalises the use of these messages by flight controllers.

In the first instance, I'd like to find out how these messages are treated by current implementations, to determine what commonality there actually is.  Could relevant people assist?

edwinhayes, 2019-08-19:
@tridge @peterbarker @DonLakeFlyer @rmackay9 @LorenzMeier @tilaktilak

peterbarker, 2019-08-19:
ArduPilot's treatment of these messages differs based on vehicle type, with varying degrees of support.

`SET_POSITION_TARGET_*` and `POSITION_TARGET_*` are essentially orthogonal.

The vehicle having received `SET_POSITION_TARGET` does not necessarily mean `POSITION_TARGET` will be emitted.  For example, if you are not in guided ("offboard") mode then you won't get the messages just because you sent `SET_POSITION_TARGET`.

The vehicle may also send `POSITION_TARGET` messages when `SET_POSITION_TARGET` has not been received - for example in Plane we send `POSITION_TARGET_GLOBAL_INT` any time you are not in `MANUAL` mode.  Copter sends it if the current mode has a valid waypoint target.

hamishwillee, 2019-08-19:
PX4 streams the position messages by default in [most streaming modes](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1639). Doesn't appear to be sent by other means. 

If we were writing this now I'd probably be hoping that:
- [MAV_CMD_SET_MESSAGE_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL) would set the interval
- [MAV_CMD_REQUEST_MESSAGE](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE) would ACK then emit the position message requested
- `SET_POSITION_TARGET_*` would set the position, and emit `POSITION_TARGET_*` whether or not the command succeeded.

But since this is not documented essentially flight stacks can do anything. If you want it to behave a particular way you'd document the expectation in a PR and we'd discuss the requirement in the dev call.

DonLakeFlyer, 2019-08-19:
This relates to a discussion around using SET_POSITION_TARGET from a ground station to move a vehicle. Since SET_POSITION_TARGET is just a regular old mavlink message the sender has no idea if the vehicle ever got the message and/or responded to it. Given that the question is whether it is possible to use POSITION_TARGET vehicle->gcs telemetry to create a verification/retry system for sending SET_POSITION_TARGET.

My feeling was trying to do this sort of thing from the ground station side was unreliable and possibly incorrect at times. If you need to have an ack for a command then you should use a COMMAND_LONG/INT variant which has ack capability built in. In the specific case of SET_POSITION_TARGET as it relates to moving a vehicle to a new location, MAV_CMD_DO_REPOSITION does what you want.

DonLakeFlyer, 2019-08-19:
FYI: All of which explains in more gory detail my request for ArduPilot to support MAV_CMD_DO_REPOSITION.

edwinhayes, 2019-08-19:
@DonLakeFlyer - `MAV_CMD_DO_REPOSITION` is only good for WGS84 frame; AFAIK there are no equivalents for other frames (for instance, local)?

@peterbarker - In addition to the stated behaviours, does AP support sending the `POSITION_TARGET` in response to `SET_MESSAGE_INTERVAL` / `REQUEST_MESSAGE`?

edwinhayes, 2019-08-19:
@hamishwillee - Well, I guess I'll see how much work it looks to be in order to get consistent implementations first, and then decide if its too hard basket.

DonLakeFlyer, 2019-08-19:
COMMAND_INT supports specifying the MAV_FRAME.

edwinhayes, 2019-08-19:
@DonLakeFlyer - Do the fields just become incorrectly named in that case?

DonLakeFlyer, 2019-08-19:
The fields of COMMAND_LONG/INT are named param5/6/7. Only the mavlink spec says that they really are for. In most cases the spec documents lat/lon usage in param5/6. But in reality depending on the MAV_FRAME the usage may differ. That said it requires the firmware to support that MAV_FRAME type. Don't know if anyone support local. QGC would never send anything in local coordinates to a vehicle.

edwinhayes, 2019-08-19:
Ok yeah I see.  I believe ArduCopter does support local?  And if PX4 doesn't, it might shortly, depending on how generous [one of our customers] is with their code.  And pursuant to the latter; we _will_ likely be wanting to make QGC support local coordinates (though whether you want to accept that into the mainline is a different obviously).

hamishwillee, 2019-08-19:
@DonLakeFlyer So assuming ArduPilot was to support MAV_CMD_DO_REPOSITION  would this imply that SET_POSITION... should be deprecated?

DonLakeFlyer, 2019-08-19:
> So assuming ArduPilot was to support MAV_CMD_DO_REPOSITION would this imply that SET_POSITION... should be deprecated?

Nope. They are two different use cases that sort of lead to the same result. They just get there in different ways.

edwinhayes, 2019-08-19:
I'm ambivalent about the idea that they are different use cases, but regardless I think it raises the question about what the correct behaviour should be when the two types of messages interact.  

For instance; if an onboard controller is sending a stream of `SET_POSITION`, and then a user sends a single `DO_REPOSITION` from the GCS, they would probably get an ACK, but then the set point would immediately be overridden?  The only way the GCS would be able to illustrate what is going on would be if it were to subscribe to the `POSITION` messages anyway?

DonLakeFlyer, 2019-08-19:
> They are two different use cases that sort of lead to the same result.

Which is some of the oddness with the mavlink spec which @edwinhayes is kind of getting at.

peterbarker, 2019-08-19:
On Mon, 19 Aug 2019, Don Gagne wrote:

> a command then you should use a COMMAND_LONG/INT variant which has ack capability built in. In the specific case of SET_POSITION_TARGET as it relates to
> moving a vehicle to a new location, MAV_CMD_DO_REPOSITION does what you want.

This message also has the advantage that it can also force a mode change, 
eliminating a race condition where you switch to mode-guided ("offboard") 
but don't get the vehicle a position to go to before you lose link.

peterbarker, 2019-08-19:
On Mon, 19 Aug 2019, Don Gagne wrote:

> FYI: All of which explains in more gory detail my request for ArduPilot to support MAV_CMD_DO_REPOSITION.

It does in Plane :-)

peterbarker, 2019-08-19:
On Mon, 19 Aug 2019, Edwin Hayes wrote:

> @DonLakeFlyer - MAV_CMD_DO_REPOSITION is only good for WGS84 frame; AFAIK there are no equivalents for other frames (for instance, local)?

Bleh.  I don't understand why the location was specified that way in the 
message,

And.... lat*1e7 and metres in the same message?!

> @peterbarker - In addition to the stated behaviours, does AP support sending the POSITION_TARGET in response to SET_MESSAGE_INTERVAL / REQUEST_MESSAGE?

Yes, both of those messages will elicit POSITION_TARGET_GLOBAL_INT and 
POSITION_TARGET_LOCAL_NED - but only if you are in a mode where that makes 
sense.

I'll just throw this in here... what these messages report isn't actually 
where the vehicle is currently headed - if it's doing avoidance it could 
very well be going directly away from the target.

Peter

tilaktilak, 2019-08-19:
> For instance; if an onboard controller is sending a stream of SET_POSITION, and then a user sends a single DO_REPOSITION from the GCS, they would probably get an ACK, but then the set point would [...]

Regarding this case, it's similar to a concurrent access to a resource. Not sure the solution is to use the same SET_POSITION message for both onboard controller and user because it will be impossible to know from where the order is coming from and then impossible to implement any mechanism for concurrent access.

edwinhayes, 2019-08-19:
> I'll just throw this in here... what these messages report isn't actually
where the vehicle is currently headed - if it's doing avoidance it could
very well be going directly away from the target.

That's another aspect that should have been captured by specification; whether the stated target should reflect internal behaviours.

The behaviour you describe would mean that collision avoidance behaviours implemented by internal or external mechanisms (w.r.t. the flight controller) report the target position differently, which seems pretty non-obvious.

edwinhayes, 2019-08-25:
Ok then, pursuant to drafting a PR for discussion: what would people prefer for the case, as Peter mentioned above, that the time arrives that the flight controller is supposed to publish `POSITION_TARGET` but it's not a mode where it's actually navigating to that target.

Should it:
a. Ack the command, then not publish.
b. Nack the command.
c. Publish the message with some fields set to mark the fact it's not being used.

And, per the last comment; should the published values of `POSITION_TARGET` reflect when the aircraft is doing something like (internal) collision avoidance such that it's not navigating directly to the position specified (via 'SET_POSITION_TARGET', 'DO_REPOSITION' or otherwise)?  I believe it always should.

edwinhayes, 2019-08-27:
> what is trying to be accomplished here and why. 

Even if everyone agrees that the `SET_TARGET` mechanism should not be used for guided mode commands _from a GCS_, the manner in which these messages are being used and published is clearly not consistent between the stacks, and open to a bunch of ambiguities.

So that alone is motivation for seeing if its possible to get agreement on what the standard behaviour should be, totally unrelated to a GCS.

> If a vehicle is using offboard control I'd assume all bets are off as to being able to control it correctly from a gcs.

Why not?  This is a self-defeating argument; it's only the case at the moment because there is no formalised standard for how such elements should interoperate.

It seems pretty obvious to me that there are a large number of use cases where a user might want to be able to tell an aircraft to navigate to a particular position, despite that aircraft also having some onboard intelligence, so it makes sense to me to consider these cases.

At the moment, since the stacks don't even agree on when to send `POSITION_TARGET`, a GCS wouldn't even be able to reliably detect that there _was_ an onboard controller, and thus tell the user that goto mode wasn't going to work properly?

edwinhayes, 2019-08-29:
@DonLakeFlyer - At the present time, there would be no reason for a compliant firmware implementation to NACK a `COMMAND_LONG` if it was also getting `SET_POSITION` messages; both would be perfectly valid messages for a flight controller to respond to while in guided mode.  The flight controller can't, from those messages along, determine what the user's intent was.

Hmm, maybe the situation where there are possible things issuing guided mode commands should be addressed with something similar to the `CHANGE_OPERATOR_CONTROL` service, such that the flight controller should only be listening to instructions from one thing at a time?

hamishwillee, 2019-11-13:
Did we end up with anything useful out of this?

The current docs indicate to me that:
- `POSITION_TARGET_*` should be emitted in response to any `SET_POSITION_TARGET_*` IFF the vehicle is being position controlled/is in a position control mode. I would expect that otherwise the message would be ignored and nothing would be emitted. This seems like what ArduPilot does.
- As stated, this reflects target as set by `SET_POSITION_TARGET_`, not a final destination. It is up to whatever is sending that target to define the destination.
- [MAV_CMD_DO_REPOSITION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION) looks far more like a GCS command. It isn't clear that the mode _has_ to be offboard/guided for this to work, but the either way it should be rejected if not supported in the mode with `MAV_RESULT_TEMPORARILY_REJECTED`
  - There doesn't seem to be any corresponding message for the actual destination. I would think this useful as it would allow the `SET_POSITION_TARGET_` to provide avoidance over the top of the guided command.

DonLakeFlyer, 2019-11-13:
> I'm kind of at a loss as to what is trying to be accomplished here and why. My point of view:
> 
> * A gcs should use COMMAND_LONG commands to tell the vehicle to do things.
> * A gcs should never rely on POSITION_TARGET telemetry to decide to do things associated with the commands from step 1.

This is still my opinion.

hamishwillee, 2019-11-14:
@DonLakeFlyer I am inclined to agree. The messages serve different purposes (or are at least being used that way). SET_POSITION_TARGET_ is used for fine control setpoint control by a companion, while `MAV_CMD_DO_REPOSITION` is effectively a large scale goto command.
I can see them working together for obstacle avoidance.

Aside from whatever problem it is @edwinhayes wants to solve though, it would be good to specify what `MAV_CMD_DO_REPOSITION` does if the setpoints start coming in (I would expect these to reflect the actual internal setpoint state).

edwinhayes, 2019-11-14:
All I was really looking for was a consistent set of documentation and behaviour between all the major MAVLink speakers.  There remain a number of ambiguities:

> POSITION_TARGET_* should be emitted in response to any SET_POSITION_TARGET_* IFF the vehicle is being position controlled/is in a position control mode.

What exactly defines a "position control mode"?

> As stated, this reflects target as set by SET_POSITION_TARGET_, not a final destination. It is up to whatever is sending that target to define the destination.

Is this actually useful/safe behaviour?

> but the either way it should be rejected if not supported in the mode with MAV_RESULT_TEMPORARILY_REJECTED

But what is the canonical way for a flight controller to know whether or not it should accept/reject reposition commands and position targets?  Both these messages are applicable in guided mode.

Anyway, this is no longer in scope for me, so feel free to close this ticket.  However, I think it should be noted that that these sort of ambiguities are a legitimate barrier to commercial software developers looking to develop safe and reliable systems.

hamishwillee, 2019-11-15:
Yes, all this is a problem. Let me see if I can find another champion offline. If not then I am going to close, because I don't believe it can be resolve "in the abstract".

hamishwillee, 2019-11-17:
A point provided by @julianoes is that on PX4 [MAV_CMD_DO_REPOSITION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION) is obeyed in HOLD mode, not offboard. So the clash does not come up.

Part of the confusion here is that "Guided mode" is ArduPilot's equivalent of offboard, and it is also a QGC term for the kind of mode that can receive GCS high level commands like `MAV_CMD_DO_REPOSITION`.

Not sure how to untangle this. On way might be to add note that they can't reliably be used concurrently and that PX4 handles reposition mode in HOLD.