mavlink/mavlink #1212: Proposal for AIS message

IamPete1, 2019-08-21:
[AIS ](https://en.wikipedia.org/wiki/Automatic_identification_system)is more or less ADSB for ships, so the [ADSB ](https://mavlink.io/en/messages/common.html#ADSB_VEHICLE)message would probably be a good place to start. There is quite abit of information available via AIS see [here ](https://gpsd.gitlab.io/gpsd/AIVDM.html) for all the messages used within the AIS system, that is i think far to much info, to start with anyway.  

[This page ](https://gpsd.gitlab.io/gpsd/AIVDM.html)gives a more approachable overview. I think the two AIS messages of interest are [message 1 ](http://www.bosunsmate.org/ais/message1.php) this is sent regularly by few seconds by vessels and gives information  about the vessels ID no, location speed and heading. [Message 3](http://www.bosunsmate.org/ais/message5.php) is sent every few minuets and gives more information about the vessel, such as size, draught, name, destination ect.

I am by no means a AIS expert (or MAVLink), but it seems some what redundant to duplicate all the AIS messages here, maybe we could do a ADSB type message for location and speed ect. and pas-through the existing AIS messages somehow?

auturgy, 2019-08-21:
agree. I think there’s a discussion on the forum from quite some time ago on this topic: I’d love to see it move forward

auturgy, 2019-08-21:
fyi https://discuss.ardupilot.org/t/ais-basic-collision-avoidance/30461

hamishwillee, 2019-08-21:
@IamPete1 We discussed in dev call. None of us are AIS experts either, but we like the idea. We did go around this with OpenDroneID recently (see WIP messages starting with [OPEN_DRONE_ID_*](https://mavlink.io/en/messages/common.html#OPEN_DRONE_ID_BASIC_ID)) and they ended up having to create their own messages for various reasons.

Anyway, we suggest you do a comparative analysis. If you think it can be done with ADSB then by all means do so. If not, create a PR and we can look at it.

auturgy, 2019-08-21:
I’d prefer AIS be independent of ADSB, as whilst a comparison is easy to make, it really is a separate use case. For USV, we need to apply UNCLOS / ROTR, so avoidance behaviour for AIS will be different from ADSB contacts.

Regards,

James

hamishwillee, 2019-08-21:
THanks @auturgy - apologies for misrepresenting you! @IamPete1 then suggest PR :-)

hamishwillee, 2019-09-04:
I'm closing this now. Question asked and answered.

closed 2019-09-04 by hamishwillee

IamPete1, 2019-09-04:
I'm working on a PR, should be in in this week