mavlink/mavlink #1217: Decode the encapsulated image data

Marxlp, 2019-08-24:
I am trying using **pymavlink** to parse the `ENCAPSULATED_DATA` message sent by **PX4FLOW** and I get some data like below. There are 17 array, each array is a payload in one mavlink package. As the [document](https://mavlink.io/en/services/image_transmission.html) said, current mavlink implement only transfers *jpeg* data. Since the **jpeg** start with **ff d8**, the format of the data below can't be identify. Does anyone have any advice?
```python
[array('B', [0, 0, 65, 73, 73, 73, 73, 73, 73, 74, 75, 75, 76, 74, 75, 77, 76, 75, 76, 77, 77, 78, 78, 78, 79, 80, 79, 79, 80, 79, 81, 80, 81, 81, 80, 82, 81, 82, 82, 81, 82, 81, 82, 83, 83, 84, 84, 84, 85, 80, 62, 76, 85, 86, 85, 85, 86, 85, 86, 87, 86, 87, 88, 87, 87, 87, 67, 73, 73, 73, 75, 74, 74, 73, 75, 74, 74, 75, 76, 77, 75, 77, 77, 78, 77, 77, 78, 79, 79, 78, 79, 80, 80, 80, 81, 80, 81, 81, 81, 81, 81, 81, 82, 84, 84, 83, 83, 84, 84, 84, 84, 85, 83, 78, 60, 78, 85, 86, 86, 85, 85, 86, 86, 86, 87, 86, 88, 87, 87, 87, 68, 72, 73, 73, 73, 72, 73, 74, 74, 75, 75, 76, 76, 76, 76, 76, 77, 78, 78, 77, 77, 78, 78, 78, 79, 80, 81, 80, 80, 80, 81, 80, 79, 82, 81, 80, 81, 82, 82, 83, 82, 82, 83, 83, 83, 84, 84, 79, 61, 77, 83, 84, 84, 86, 86, 86, 86, 85, 86, 86, 86, 86, 86, 88, 67, 72, 73, 72, 72, 73, 74, 73, 74, 75, 75, 74, 75, 77, 77, 76, 77, 77, 77, 78, 77, 78, 79, 79, 79, 79, 80, 78, 79, 79, 80, 81, 80, 80, 80, 81, 82, 82, 83, 83, 82, 82, 82, 83, 83, 84, 85, 77, 60, 78, 84, 84, 84, 84, 86, 86, 85, 85, 86, 86, 87]), array('B', [1, 0, 87, 86, 86, 69, 73, 74, 73, 73, 74, 74, 74, 75, 74, 75, 76, 76, 77, 76, 78, 78, 78, 78, 77, 79, 79, 80, 79, 80, 78, 80, 81, 80, 81, 80, 81, 81, 81, 80, 83, 82, 82, 82, 83, 83, 83, 84, 84, 85, 84, 85, 78, 61, 81, 85, 84, 85, 86, 85, 85, 86, 87, 86, 86, 86, 88, 87, 88, 71, 73, 72, 73, 73, 74, 75, 74, 74, 75, 76, 75, 76, 77, 77, 77, 77, 78, 77, 78, 79, 79, 79, 79, 80, 80, 80, 81, 80, 80, 80, 81, 82, 82, 81, 81, 82, 82, 82, 83, 82, 82, 82, 84, 83, 83, 85, 76, 61, 81, 84, 85, 86, 87, 85, 86, 86, 87, 86, 87, 88, 87, 87, 88, 71, 73, 74, 74, 74, 75, 75, 74, 75, 75, 75, 76, 76, 77, 77, 77, 77, 77, 78, 79, 78, 78, 79, 79, 80, 80, 79, 80, 80, 80, 81, 81, 81, 82, 81, 82, 82, 83, 83, 83, 83, 84, 83, 84, 83, 85, 85, 76, 62, 81, 85, 86, 87, 86, 86, 85, 86, 86, 86, 85, 87, 87, 86, 87, 72, 73, 74, 73, 75, 73, 75, 75, 75, 74, 76, 76, 76, 77, 77, 78, 78, 77, 77, 78, 78, 79, 78, 79, 81, 80, 79, 80, 81, 80, 81, 81, 81, 80, 82, 82, 82, 82, 81, 82, 81, 83, 83, 85, 86, 84, 85, 72, 63, 82, 85, 85, 85, 85, 86, 85, 87, 86]), array('B', [2, 0, 86, 86, 87, 87, 87, 89, 72, 73, 73, 73, 75, 77, 76, 74, 74, 76, 76, 77, 77, 77, 78, 78, 78, 78, 79, 78, 80, 78, 80, 80, 81, 81, 81, 80, 80, 81, 81, 82, 81, 84, 83, 83, 83, 82, 84, 85, 84, 84, 84, 85, 86, 86, 85, 74, 65, 82, 85, 86, 87, 87, 86, 86, 86, 86, 87, 87, 89, 88, 89, 88, 74, 73, 74, 74, 75, 75, 77, 75, 76, 76, 76, 76, 76, 78, 77, 78, 79, 78, 78, 78, 79, 78, 79, 79, 80, 80, 80, 81, 81, 81, 82, 81, 83, 82, 83, 83, 83, 83, 83, 83, 84, 84, 83, 84, 85, 84, 85, 71, 66, 84, 85, 86, 86, 86, 86, 85, 87, 87, 87, 88, 87, 87, 88, 88, 73, 72, 73, 74, 73, 74, 75, 76, 74, 75, 76, 76, 76, 75, 77, 77, 77, 77, 77, 77, 78, 78, 78, 78, 79, 78, 80, 80, 81, 79, 80, 81, 81, 82, 82, 81, 83, 81, 82, 82, 82, 83, 83, 85, 83, 85, 83, 70, 67, 82, 84, 85, 85, 84, 85, 86, 86, 85, 86, 86, 87, 87, 87, 87, 74, 74, 75, 76, 76, 76, 76, 76, 77, 77, 77, 78, 77, 78, 78, 79, 78, 79, 80, 78, 80, 79, 81, 81, 82, 81, 82, 82, 81, 81, 82, 82, 83, 83, 83, 83, 85, 84, 83, 84, 84, 86, 86, 86, 86, 87, 85, 69, 68, 85, 86, 86, 86, 88, 86]), array('B', [3, 0, 88, 87, 87, 89, 88, 89, 89, 89, 88, 73, 72, 74, 72, 74, 75, 75, 74, 75, 75, 76, 76, 76, 76, 76, 76, 77, 77, 78, 77, 78, 79, 79, 78, 80, 80, 80, 79, 81, 79, 80, 82, 81, 81, 82, 82, 81, 84, 81, 84, 83, 83, 84, 84, 85, 84, 83, 68, 69, 82, 85, 85, 86, 86, 86, 86, 85, 87, 86, 86, 89, 89, 87, 87, 73, 73, 73, 73, 74, 75, 74, 75, 76, 74, 75, 75, 76, 75, 75, 77, 77, 77, 78, 76, 78, 79, 79, 79, 79, 79, 80, 81, 80, 81, 82, 82, 81, 81, 81, 82, 82, 83, 83, 83, 84, 83, 85, 83, 84, 85, 83, 68, 70, 83, 84, 85, 84, 86, 85, 85, 86, 86, 86, 87, 87, 89, 88, 87, 73, 73, 74, 73, 73, 74, 74, 74, 75, 74, 74, 75, 74, 76, 76, 78, 77, 77, 78, 78, 79, 79, 79, 79, 79, 78, 80, 80, 81, 81, 80, 81, 80, 82, 83, 82, 82, 84, 83, 82, 84, 83, 84, 84, 86, 85, 84, 67, 71, 85, 85, 86, 84, 86, 86, 86, 86, 86, 87, 87, 88, 88, 86, 87, 72, 72, 73, 73, 73, 73, 74, 74, 74, 74, 75, 75, 75, 76, 76, 77, 77, 75, 77, 79, 79, 79, 78, 79, 80, 80, 80, 81, 80, 80, 81, 81, 82, 81, 82, 82, 82, 83, 82, 82, 83, 84, 83, 84, 84, 85, 82, 65, 72, 84, 84, 86]), array('B', [4, 0, 86, 84, 85, 86, 86, 85, 85, 87, 87, 87, 88, 87, 72, 73, 73, 73, 73, 74, 73, 75, 75, 74, 74, 76, 75, 76, 76, 77, 78, 79, 78, 78, 79, 79, 79, 79, 80, 80, 80, 81, 80, 80, 81, 82, 82, 82, 83, 82, 82, 84, 84, 84, 84, 84, 85, 84, 85, 86, 83, 65, 74, 84, 84, 86, 85, 87, 86, 87, 87, 87, 86, 87, 87, 88, 87, 87, 72, 73, 73, 73, 73, 73, 74, 74, 74, 75, 75, 75, 75, 75, 75, 76, 76, 77, 78, 77, 77, 79, 78, 80, 80, 81, 80, 80, 80, 82, 81, 81, 82, 83, 83, 82, 83, 83, 82, 83, 83, 83, 83, 85, 84, 84, 82, 62, 74, 83, 84, 85, 85, 85, 86, 87, 87, 87, 85, 86, 88, 87, 88, 87, 73, 74, 73, 73, 74, 74, 74, 74, 74, 75, 75, 74, 76, 76, 75, 77, 78, 77, 78, 78, 78, 80, 79, 80, 79, 80, 81, 80, 81, 81, 80, 81, 81, 81, 82, 82, 83, 83, 83, 84, 84, 84, 85, 85, 84, 86, 81, 63, 75, 84, 85, 85, 85, 86, 85, 86, 86, 86, 88, 88, 88, 88, 87, 88, 72, 73, 74, 73, 73, 73, 75, 75, 75, 74, 75, 75, 76, 76, 76, 77, 78, 78, 78, 79, 79, 80, 80, 79, 80, 80, 81, 81, 80, 80, 81, 82, 81, 82, 82, 82, 82, 83, 83, 83, 84, 85, 84, 84, 85, 84, 79, 62, 76]), array('B', [5, 0, 85, 84, 85, 86, 87, 84, 87, 85, 87, 88, 89, 88, 88, 86, 88, 73, 73, 74, 73, 74, 75, 74, 75, 75, 76, 76, 75, 76, 76, 76, 77, 78, 78, 79, 80, 79, 79, 80, 80, 80, 82, 82, 82, 82, 82, 82, 82, 82, 82, 83, 82, 83, 83, 83, 85, 84, 86, 84, 84, 85, 84, 80, 63, 77, 84, 84, 86, 86, 85, 87, 85, 86, 87, 86, 87, 88, 88, 87, 89, 73, 74, 73, 73, 74, 73, 74, 75, 75, 76, 76, 75, 75, 78, 76, 77, 78, 79, 78, 79, 79, 79, 80, 80, 81, 79, 82, 83, 81, 81, 81, 82, 82, 82, 83, 81, 83, 83, 84, 83, 84, 84, 85, 84, 85, 85, 79, 62, 78, 85, 86, 86, 87, 86, 86, 86, 88, 87, 88, 87, 88, 88, 89, 87, 73, 73, 73, 74, 74, 74, 74, 74, 75, 75, 76, 76, 76, 76, 76, 77, 77, 77, 78, 79, 79, 79, 79, 79, 80, 80, 80, 81, 80, 81, 80, 81, 81, 82, 83, 82, 83, 83, 82, 83, 83, 83, 84, 84, 85, 84, 77, 61, 78, 85, 85, 86, 85, 85, 85, 86, 87, 87, 87, 87, 88, 88, 89, 89, 74, 73, 74, 74, 74, 74, 74, 75, 74, 76, 76, 76, 77, 77, 78, 78, 78, 78, 80, 80, 80, 79, 81, 81, 80, 81, 81, 81, 82, 81, 81, 82, 82, 83, 82, 81, 83, 83, 83, 82, 84, 85, 83, 85, 83, 85]), array('B', [6, 0, 76, 63, 80, 86, 85, 86, 86, 86, 85, 88, 89, 88, 89, 88, 88, 88, 89, 89, 72, 73, 74, 74, 73, 74, 75, 75, 75, 75, 76, 77, 76, 77, 78, 78, 79, 79, 79, 79, 78, 79, 80, 81, 80, 81, 80, 80, 81, 81, 81, 82, 81, 82, 81, 82, 82, 82, 83, 83, 84, 84, 83, 84, 83, 84, 76, 62, 79, 85, 86, 86, 86, 86, 86, 88, 87, 87, 88, 88, 89, 89, 88, 88, 74, 74, 74, 73, 74, 75, 75, 75, 76, 75, 76, 77, 76, 77, 77, 77, 77, 78, 78, 79, 79, 79, 80, 79, 80, 80, 81, 81, 82, 80, 82, 81, 83, 82, 81, 83, 83, 82, 83, 83, 84, 255, 84, 84, 85, 85, 73, 62, 81, 85, 86, 86, 87, 87, 85, 87, 88, 87, 88, 88, 88, 90, 88, 89, 75, 73, 74, 74, 76, 76, 77, 77, 77, 77, 76, 77, 76, 78, 79, 79, 79, 78, 79, 80, 80, 80, 80, 81, 80, 81, 80, 82, 81, 82, 82, 83, 81, 83, 83, 83, 83, 83, 84, 85, 85, 84, 85, 85, 86, 85, 73, 64, 82, 85, 86, 87, 87, 88, 88, 88, 88, 88, 89, 88, 89, 89, 89, 90, 73, 74, 75, 74, 76, 75, 76, 76, 76, 76, 77, 76, 77, 77, 77, 78, 78, 79, 79, 80, 79, 80, 79, 81, 81, 81, 81, 81, 81, 80, 81, 82, 81, 82, 84, 81, 82, 83, 83, 84, 83, 85, 85]), array('B', [7, 0, 84, 84, 85, 74, 62, 83, 85, 85, 87, 87, 86, 88, 87, 88, 89, 88, 88, 90, 88, 89, 88, 75, 75, 74, 75, 75, 75, 76, 76, 77, 78, 77, 77, 77, 77, 78, 78, 78, 79, 79, 80, 79, 79, 82, 81, 80, 80, 81, 81, 81, 81, 81, 83, 82, 82, 82, 84, 84, 84, 84, 84, 86, 85, 85, 85, 84, 85, 73, 64, 84, 87, 86, 88, 87, 87, 89, 88, 89, 89, 89, 90, 89, 90, 87, 88, 76, 75, 76, 76, 76, 77, 76, 76, 78, 77, 77, 78, 78, 78, 78, 79, 79, 80, 80, 79, 80, 80, 81, 81, 80, 80, 81, 81, 81, 81, 82, 83, 83, 84, 84, 83, 83, 84, 84, 85, 85, 84, 85, 86, 86, 86, 71, 65, 85, 87, 87, 88, 88, 88, 88, 87, 88, 89, 88, 90, 91, 90, 90, 89, 74, 74, 75, 74, 75, 76, 75, 76, 76, 77, 77, 77, 77, 78, 77, 78, 79, 77, 79, 79, 79, 80, 80, 80, 80, 80, 80, 82, 81, 82, 80, 82, 81, 82, 83, 82, 83, 83, 83, 83, 84, 83, 84, 85, 85, 85, 69, 65, 84, 87, 87, 8 [... cut ...] , 233, 36, 160, 66, 223, 221, 125, 232, 51, 90, 88, 116, 87, 95, 244, 237, 32, 194, 21, 64, 197, 28, 209, 252, 73, 43, 64, 151, 76, 64, 66, 118, 36, 70, 160, 140, 167, 189, 65, 251, 7, 49, 1, 51, 126, 158, 24, 233, 116, 101, 119, 180, 202, 183, 159, 159, 74, 232, 10, 235, 230, 46, 114, 174, 64, 56, 82, 84, 199, 203, 164, 127, 7, 131, 118, 144, 231, 188, 94, 163, 1, 58, 175, 6, 130, 26, 52, 180, 145, 144, 176, 192, 84, 15, 194, 38, 229, 216, 114, 23, 180, 244, 255, 50, 41, 12, 190, 215, 44])]
```

LorenzMeier, 2019-08-24:
Look at the image type in the message. The payload in this case are just raw intensities.

closed 2019-08-24 by LorenzMeier

Marxlp, 2019-08-24:
Thanks , @LorenzMeier.
I checked the data type, and it's 3, `MAVLINK_DATA_STREAM_IMG_RAW8U` ( if I don't misunderstand the `common.xml`). Another problem comes up. The size of data received is 253 x 17 = 4301,  while the image has width x height = 64 x 64 = 4096 bytes.

Marxlp, 2019-08-24:
I finally solve this problem by remove data out of  the section 4096 after read the [FLOW](https://github.com/PX4/Flow). The result is shown below and mycode also pasted below. Thanks again @LorenzMeier.
[image]
```python
#!/usr/bin/env python2
#coding:utf-8
"""
Author:MarxLP
This script shows how to read px4flow image stream.
"""

from array import array
import PIL
import io
from pymavlink import mavutil
import numpy as np
import matplotlib.pyplot as plt
import pdb

# the PX4FLOW is plugged on the computer
PORT = '/dev/ttyACM0'
BAUDRATE = 115200
def main():
    con = mavutil.mavlink_connection(PORT, BAUDRATE)
    target_system = con.target_system
    msg = con.wait_heartbeat()
    # send image request TODO
    while True:
        msg = con.recv_match(type='DATA_TRANSMISSION_HANDSHAKE', blocking=True)
        packets_num = msg.packets
        payload = msg.payload
        image_size = msg.size
        image_data = array('B',[])
        w, h = msg.width, msg.height
        print('w:{0}, h:{1}, image_size:{2}, packets_num:{3}'.format(w,h,image_size, packets_num))
        for i in range(packets_num):
            img_msg = con.recv_match(type='ENCAPSULATED_DATA', blocking=True)
            #pdb.set_trace()
            image_data.extend(img_msg.get_payload()[2:])
        img_data = np.array(image_data[:w * h]).reshape((w,h))
        plt.imshow(img_data, cmap='gray')
        plt.show()
        # just one shot for test
        break

if __name__ == '__main__':
    main()
```