mavlink/mavlink #1218: Issue with camera definition URI and MAVLink FTP, and proposal to resolve

olliw42, 2019-08-25:
the camera microservice provides a mechanism for downloading a camera definition .xml file from a URI specified in the `CAMERA_INFORMATION` message. 

Ideally, from a usability perspective, this file would be stored on the camera component itself and be downloaded via MAVLinkFTP; the URI may be e.g. `mavlinkftp://filename.xml` (as suggested here https://github.com/mavlink/qgroundcontrol/issues/7453#issuecomment-493212569). The reasons for why this is desirable I think need not be discussed.

This however creates an issue, since the requester, such as a GCS, may not actually support MAVLinkFTP. The issue is that the camera component does not know if the requester supports MAVLinkFTP or not, but the proper URI it should provide in the `CAMERA_INFORMATION` message depends on this: If the requester supports MAVLinkFTP it wants to send a `mavlinkftp://` URI, so one can benefit from that feature, while if the requester does not, it wants to send e.g. a `http://` web URI, in order to not make the whole mechanism to not work.

Note that supporting MAVLinkFTP is NOT a requirement of the camera microservice specification, and this case is hence within spec and hence by definition real !

So, basically, what one wants to have is a mechanism which allows a camera component to send a mavlinkftp URI if that is supported or a normal URI as fall back if not.

Several mechanisms which would not need touching the messages come to mind, however I failed to see one which is really "nice". For instance, the camera component designer could add a parameter which allows a user to set the URI, which is however nasty for the user, also it may require her to do so whenever the GCS being used is changed. One could (finally) establish a mechanism which allows the camera component to detect which type the requester is and somehow know from this if it supports ftp or not, which obviously is also nasty in many ways. One also could require the camera component to have a specific parameter which the requester could set to tell the camera component which URI to send. And so on. As said, IMHO all not "nice".

Another approach, which I'd like to propose herewith, could be to use the capability flags, and add a flag which tells the requester/GCS if the camera component supports MAVLinkFTP or not. In either case the camera component would send its normal URI, e.g. `http://path/to/file/filename.xml`, but in case the FTP flag is set the requester would understand this URI to mean `mavlinkftp://filename.xml`.

This wouldn't break any existing implementation, and resolve the issue. 

The drawback would be that the definition xml would have to be located in the root directory of the camera component. One could partially resolve that if one would add e.g. two FTP flags, so that one could specify a truncation depth of 0,1,2. In the above example truncation depth 0 would result in `mavlinkftp://filename.xml`, depth 1 in  `mavlinkftp://file/filename.xml`, and depth 2 in `mavlinkftp://to/file/filename.xml`. More depth levels probably don't make sense.

Open for discussion :)

hamishwillee, 2019-08-26:
So to summarise:

It is desirable to host the camera definition with the camera and deliver it via mavftp (because it will always be up to date). But this is a problem because if you specify a mavftp URL and the GCS does not support MAVFTP then the file is inaccessible.

Options 
- add support for primary/fallback URLs in the API. In simplest form, this could just be comma separated URLs that the GCS was expected to try in order until it succeeds.
- mechanism for camera to determine what GCS supports and send URL it can use
- Provide mechanism such that GCS can tell if Mavftp is supported (or not) and transpose a generic URL into a mavftp URL
- Accept that there might be GCS that can't read the file.

@dogmaphobic Do you have an opinion? I tend to think the first one makes most sense.

olliw42, 2019-08-26:
I think the first one is an excellent idea if there isn't an issue with the 140 chars limit. It would break existing implementations however (I myself wouldn't be concerned but you guys often are). I'm not totally sure about side effects, but potentially this could be avoided by sending '\0' separated URLs. For implementations trying to extract the list it's then slightly harder to find the end of the list, could be something like a double '\0\0'. Anyway, I like this idea.

the third I guess is kind of what I suggested. Not that elegant and flexible as one, but non-breaking and easy enough.

the fourth would mean that the camera microservice won't always work, i.e. essentially break the whole thing in pieces, and thus would not consider it an option

olliw42, 2019-08-26:
just for completeness, one further option would be to make MAVLinkFTP support a required part of the camera microservice spec. I'm not sure if this would be widely appreciated however.

julianoes, 2019-08-26:
I would vote to make MAVLinkFTP required when implementing the camera definition protocol because the implementations that I am aware of already support the camera definition protocol, namely QGC and MAVSDK (very soon).

All that is required is to make them understand the mavlinkftp:// URL.

olliw42, 2019-08-26:
the service has two sides, called GCS and camera in the docs, and "implementation" can mean either or both of them

if you mean by "implementation" only the GCS side, i.e. that the spec should be that GCS is required to support MAVLinkFTP, but that the camera may or may not, then I think this makes lots of sense. 

If the spec would however also require the camera to support MAVLinkFTP, then this could be a REAL issue:  I would then like to emphasize again that this by extension most likely would imply that a similar requirement is also needed for other cases too where a definition file is involved, such as in a potential COMPONENT_INFORMATION mechanism. This could be a **VERY** though requirement for many components !!! Sending a COMPONENT_INFORMATION is easy enough, but doing the MAVLinkFTP, even if just the basics needed for listing the dir and reading a file adds substantial complexity and resource requirements.

julianoes, 2019-08-26:
> if you mean by "implementation" only the GCS side

That's how I understood your question. I think that on the other hand, as a camera/gimbal/component you can assume that the ground station has mavlinkFTP and http capabilites. So if as a component you can do mavlinkFTP, then you should do mavlinkFTP. If you can't, then you might as well hope for http.

olliw42, 2019-08-26:
GREAT

if that would be agreeable to everyone then this would have found a quick and easy solution (just needs to be made definite by adding to the spec)
it's of course the easiest for a component if a component can rely on some smartness on the other end of the link :)

olliw42, 2019-08-27:
my understanding of the conclusion too

hamishwillee, 2019-08-27:
Thanks @dogmaphobic ! So to summarise, you are happy for us to document a requirement that a GCS "must" support both HTTP and MAVLink FTP.

closed 2019-08-28 by LorenzMeier

olliw42, 2019-08-28:
GREAT!
many thx

hamishwillee, 2019-08-28:
OK, I have updated two docs for this fix: https://github.com/mavlink/mavlink-devguide/pull/205 and https://github.com/mavlink/mavlink/pull/1221

PR #1221 (2019-08-28): CAMERA_INFORMATION - url must support mavftp
Fixes #1218