mavlink/mavlink #1219: Gentle reminder on some issues

olliw42, 2019-08-25:
for some of the open issues/PRs I've raised I would highly appreciate it if they would be settled in a not to distant future:

* behavior on received PARAM_VALUE/PARAM_EXT_VALUE/PARAM_EXT_ACK messages
https://github.com/mavlink/mavlink/issues/1181
seems to have been found to make sense
is crucial to know for any component implementation if it can rely on this

* COMPONENT_INFORMATION
https://github.com/mavlink/mavlink/pull/1176
???? seems to be stuck on all ends
advice for how to move forward highly appreciated, or if to abandon if obsolete
could also influence design of other upcoming INFORMATION-type messages, such as GIMBAL_INFORMATION and alike

* Update docs related to AUTOPILOT_VERSION
https://github.com/mavlink/mavlink/issues/1175
the docs are still as before and do not reflect that AUTOPILOT_VERSION can't be relied on, comments in https://github.com/mavlink/mavlink/issues/1156 appear all to still apply

* FTP: crc32
https://github.com/mavlink/mavlink/issues/1208
nasty, I know, but really unclear currently what to do in actual component implementations

* CAMERA_STATUS message
https://github.com/mavlink/mavlink/issues/1202
no response yet at all. Comment would be much appreciated, and if it's just a 'no'.

* type of component
the issue of how to make it possible to determine the type/brand of a component/GCS, like it's possible for the autopilot, is lingering around in several places, and may have impact in several topics

Many thx, and sorry for poking. :)

hamishwillee, 2019-08-25:
> behavior on received PARAM_VALUE/PARAM_EXT_VALUE/PARAM_EXT_ACK messages #1181
seems to have been found to make sense - is crucial to know for any component implementation if it can rely on this

I updated the post. 

The type of component and #1175 are on me. I am well aware of them but they won't make it to the top of my priority list any time soon. Keep pinging if you like though.

olliw42, 2019-08-26:
>The type of component and #1175 are on me. I am well aware of them but they won't make it to the top of my priority list any time soon.

fair enough. Many thx for the feedback, it's helpful to know.
I got an impression already that the type of component thing won't be resolvable soon, so added it since it's distributed in many places. #1175 like some others I listed clearly affect implementation, but I believe there had been a general consensus, so I can at least go with what I believe the general  consensus is :). It's a bit uncomfortable though to not know exactly what the spec is in this basic aspect.
Thx so far.

olliw42, 2019-09-02:
first post updated to reflect current situation (or what I believe the current situation is)

julianoes, 2020-12-03:
@olliw42 I'm just looking at the different CRC32 implementations and so far it all seems to be the same implementation, and should give the same results although I haven't verified them all.

Could you point me to the differences you have seen? Thx.

https://github.com/ArduPilot/MissionPlanner/blob/7870f55a9806ce4d42d1b6381ad69a4a548338ab/ExtLibs/ICSharpCode.SharpZipLib/Checksum/Crc32.cs#L101

https://github.com/ArduPilot/ardupilot/blob/20c4f1950bec41c124319a75f93430878b94db4b/libraries/AP_Math/crc.cpp#L206

https://github.com/mavlink/qgroundcontrol/blob/4c872ca4abc68382afc00c2317ec9bfdf4431b44/src/QGC.cc#L138

https://github.com/PX4/PX4-Autopilot/blob/229f02a4f9491ca557b83f7e36160821b825e362/platforms/posix/src/px4/common/lib_crc32.c#L108

julianoes, 2020-12-03:
@olliw42 ok, all of the above seem to be using the same algorithm, however, as you rightfully pointed out earlier. The CRC algorithm used does not seem to be the one that the web seems to agree upon (aka Gary Brown). The implementation we use starts from `0x0` instead of `0xFFFFFFFF` and is missing the final xor out step with `0xFFFFFFFF`. So, it's all consistent but wrong :sob:.

hamishwillee, 2020-12-09:
Closing this as "fixed". If there is more to be done, I think a new issue worthwhile creating at this point.

closed 2020-12-09 by hamishwillee

olliw42, 2020-12-09:
@julianoes 

sorry for not having see that earlier ... I mistakenly had my git notifications turned off LOL

yes, this was my "observation" too, that these two versions seem to exist, and that both are widely used, i.e., for each you can projects implementing it, and that what is used in our field is not Gary Brown.

great to hear that it is at least consistently used incorrectly !!!!

I didn't intended to doubt that there is a good reason for using the modified version, but it is great that you identified the explicit reason!

very glad to see this resolved, that's the sort of little nasty thing which however takes lots of time to resolve. 
Great job, sir. Thx.