mavlink/mavlink #1220: Pl explain how "payload length is defined by remaining message length" is working

olliw42, 2019-08-26:
messages such as [V2_EXTENSION](https://mavlink.io/en/messages/common.html#V2_EXTENSION) and the recently added [TUNNEL](https://mavlink.io/en/messages/common.html#TUNNEL) carry opaque data in a payload field of variable length, and as regards the length determination it is said that 

> Variable length payload. The payload length is defined by the remaining message length when subtracting the header and other fields.

However, I really don't see and/or can't figure out how this should or can work, and hence would highly appreciate an explanation of how to do this in practice.

The issue - IMHO - and which I don't get around without fooling - is that payloads may have valid zeros at the end, but the MAVLink2's empty byte payload truncation chops them off and modifies the message size length accordingly. I can't see any reasonable method to retrieve the original valid payload size from the received message. So, I argue, the procedure as described in the spec fails, and a payload_length data field is needed to completely and correctly retrieve the intended payload.

I note that in #1194 I had suspicion that this might not work and a payload_length field be needed, but assumed by the precedence of V2_EXTENSION and that it wasn't commented on further that I'm wrong and it's not needed. Maybe my fault if I wasn't insisting enough in this point.

thx for any insight.

olliw42, 2019-08-27:
sure, there are workarounds (I'm actually doing your first method)

BUT

1. The statements as such are - IMHO - simply incorrect, and should (must) thus - IMHO - be corrected (independent on any workaround).

2. The payload length is - IMHO - quite different to things like a sequence number and such since there is no useful application of the message without knowing the payload length, that is ANY application will have to have this information! So I consider it a very basic requirement for the working of the message, which I think is justification to add such a field outside of the opaque structure. 

You also can look at it this way: Anyone will have to consume one byte of the payload for that purpose, and it thus doesn't make much sense to not put it outside and canonicalize the payload length handling.

as I said, it was probably my fault that I wasn't more pressing this point and more insistingly expressing my concerns that the described method might not actually work.

olliw42, 2019-08-27:
well, I would have done that if I would be sure to 100.00% that I'm right and that the original procedure doesn't work, but I'm sure to only 100%-epsilon ... I'm worried I miss a point. That is, I would highly appreciate if someone could/would explicitly confirm/disconfirm what the "payload length is defined by remaining message length" method isn't working. So that there is a clear starting point for the discussion.

hamishwillee, 2019-08-27:
I had nothing to do with defining either of these messages so I could be wrong about the intention but ....
if the payload format is opaque (entirely up to you), then you can define that it must terminate with some marker that isn't zero to prevent truncation, or you can include a payload length as part of your private protocol.

hamishwillee, 2019-08-27:
I have no strong opinion. Suggest you create a PR that does what you want and we can review.

hamishwillee, 2019-08-28:
Yes docs should also be improved to note the impact of zero byte truncation, because this is non-obvious. And yes, they aren't really any more "variable length" than any other message. I wouldn't say the messages "don't work". Obviously they do work and can be made to work - it is just the limitations need to be properly covered.

I agree that the messages would be better with a length field - less guessing and fiddling around for users. Can't be done as an extension without losing zero byte truncation. We'd need a new message.

olliw42, 2019-08-29:
many thx!

see #1222

closed 2019-09-02 by olliw42