mavlink/mavlink #1225: MAVLink Tunnel: Two points to discuss

olliw42, 2019-09-02:
first off, the `TUNNEL` message is working great, as intended. A brief demo is here: https://youtu.be/FRyZ7VZ5lvY.

The following two points are thus not issues MAVLink necessarily has to be concerned about, but it would be nice if MAVLink would provide some help to ease things.

**1. Component ID, GUI heartbeat**

Let's exemplarily consider a system GUI <-serial-> AutoPilot <-serial-> STorM32.

Since the `TUNNEL` message should not be broadcasted, but the target mechanism be used, the GUI obviously must be assigned a component ID. In addition, the GUI needs to emit a heartbeat by istelf, so that the AutoPilot can open the channel and route the messages between GUI and STorM32.

Currently I'm using for the GUI:
component ID = MAV_COMP_ID_USER42 (= 66)
heartbeat.type = MAV_TYPE_GCS
heartbeat.autopilot = MAV_AUTOPILOT_INVALID

This, as said, works fine. However, it would be great and make things easier if - like with other components - a "tunnel server" could be assigned a `MAV_TYPE` and a default component ID. The idea here is somewhat similar to the `MAV_COMP_ID_UDP_BRIDGE` and `MAV_COMP_ID_UART_BRIDGE` components, where it obviously also made sense to provide default component IDs. Since the type of a component is first determined by a `MAV_TYPE` and secondarily by the component ID, the concept implies that also a `MAV_TYPE` should be defined, which also would clarify the heartbeat.

So, my proposal would be to add to the spec a component ID and `MAV_TYPE`:

MAV_COMP_ID_TUNNEL_SERVER = 242
MAV_TYPE_TUNNEL_SERVER = 34

I chose the name "TUNNEL_SERVER" only as example, any other which you guys would find more appropriate would be fine. IMHO one default component ID would be totally sufficient, like with the bridge IDs.

**2. Procedure(s) for registering codes for payload_type**

The TUNNEL message has a field payload_type, and it suggested that software creators can register blocks of types as needed, and involves a file https://github.com/mavlink/mavlink/tunnel-message-payload-types.xml. The procedures and mechanism for doing this were however explicitly left open so far.

Currently I'm using
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_IN = 33333
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_OUT = 33334

but I would be interested of course to get codes registered. So, it's time to discuss the registration procedures and mechanisms.

I do not have a final proposal here. In my mind two steps can be distinguished: 

**2.1 Adding new codes to the MAV_TUNNEL_PAYLOAD_TYPE enum:** 
This obviously can be easily handled by asking for PRs. A slight issue arises from the promise that software creators can register "blocks of types". E.g., I only use two codes now but would want to register let's say 5 codes, or 10. This implies that the purpose and names of the remaining 3 (or 7) codes are not known, and may change over time. My proposal would be thus to allow that. That is, the initial registration of e.g. 5 codes could be like

MAV_TUNNEL_PAYLOAD_TYPE_STORM32_UNDEFINED1 = XXX
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_UNDEFINED2 = XXX+1
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_UNDEFINED3 = XXX+2
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_UNDEFINED4 = XXX+3
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_UNDEFINED5 = XXX+4

The software creator would be allowed to change the name of any of these via a PR at some later time, but only once (i.e. only undefined codes can be changed). 

**2.2 Specifying the entry in the tunnel-message-payload-types.xml file:**
In principle this file is not needed for the working of the TUNNEL message, so adding to the MAV_TUNNEL_PAYLOAD_TYPE enum could be seen as the key step in registering. Therefore, this point IMHO can be discussed separately from 2.1 - and maybe we should do so: At this point in time it's not clear to me what purpose this file should finally serve, I could see several meaningful options, so your input is needed.

Looking forward to hear your opinions.

olliw42, 2019-09-07:
somewhat understandably there hasn't be much response. So, in order to push that forward, I'd like to make specific suggestions, and for better clarity I've put them separately in the next two posts.

olliw42, 2019-09-07:
**Tunnel Component ID**

I still would find it highly convenient if one default component ID could be assigned to a tunnel communication partner. I however concluded that it doesn't really need a new MAV_TYPE; the available ones are good for the time being. 

I thus suggest adding MAV_COMP_ID_TUNNEL_NODE = 242, and have submitted a corresponding PR:

olliw42, 2019-09-07:
**Procedure(s) for registering codes for payload_type**

I suggest this procedure/rules:

1. The first 200 codes are reserved for MAVLink, just in case.
2. A software creator can register a block of codes by means of PR. Per default a block of e.g. 25 codes is registered. But this can of course be more or less than the default. 
3. The block of codes is added to the enum as MAV_TUNNEL_PAYLOAD_TYPE_XXXXX_RESERVEDYY, where XXXXX indicates the software creator or software, and YY is an increasing number starting at 0. This level of registration just means that a block of codes is reserved for the software creator and blocked for others, but the software creator hasn't made any specification as to its use. No entry to the .xml file is needed.
4. A code can be given an explicit name by the software creator, again by means of a PR. However, in that case there must be a complete description of the structure of the payload being added to the .xml file. This level of registration makes the payload protocol associated to the code open to the public, i.e. other implementations may use tunnel messages with this payload protocol to communicate with the said software.
5. Once a code has become explicit, its name AND its payload structure cannot be changed anymore, for forward and backward compatibility.

Comments: 
* The size of the default block of 25 can be considered. On the one hand there is no reason to not give "big" blocks, with 25 one can accommodate 1300 software creators, on the other hand there is not much reason to have many codes, since a finer granularity can always be achieved within the payload protocol.
* I would suggest that the code blocks are aligned to multiples of the default (e.g. 25), that is each block starts at a code devisible by 25.
* One could allow a very careful means to deprecate a code and later reuse it, like with messages, but this probably shouldn't be less strict as for messages. I wouldn't propose such a thing here.
* The xml markup in the xml file isn't specified by this, but that can be deferred to a later point without harm.

I accordingly have submitted a corresponding PR for registering 10 codes for the STorM32 gimbal controller:

BTW: One "issue" I see with the enum is that it potentially becomes very big and may blow up the web page documentation. I wonder if there could be a mechanism which would make the reserved codes show up as blocks in the doc webpage. But that's another topic.

edwinhayes, 2019-09-09:
> I wonder if there could be a mechanism which would make the reserved codes show up as blocks in the doc webpage. But that's another topic. Such a mechanism could be however handy also elsewhere, e.g. shrink the highly redundant MAV_COMP_ID_USERXX fields.

Ahh, yeah, those are basically my fault sorry; I'm gonna try to do something about them.

hamishwillee, 2019-09-09:
@edwinhayes I'm thinking about proposing a mechanism for specify enum values as a range.

edwinhayes, 2019-09-09:
@hamishwillee Seems sensible.  I was going to propose something orthogonal; the addition of a tag indicating that a field is intended to be overridden (with the corresponding change in the generator/validator to match).

hamishwillee, 2019-09-09:
@edwinhayes I don't quite get how that would work; the idea of reserving something is so that you get a clear warning when you try to override it. Essentially what I want to do is just make it very easy to reserve blocks so you get that warning without all the hassle of generating and rendering individual entries.

edwinhayes, 2019-09-09:
@hamishwillee But you don't get a _warning_, you get an _error_.  Which means that even if you have your own dialect, you have to keep referring to a component as MAV_COMP_ID_USER23, rather than something meaningful like MAV_COMP_MY_SPECIAL_COMPONENT.

hamishwillee, 2019-09-09:
Thanks @edwinhayes . That is a good point.

hamishwillee, 2019-09-09:
@olliw42 Thanks for taking the time to make a proposal - it is always easier to comment on an idea than come up with your own.

All this is very logical/robust. The only concern I have is the one you raised about rendering of large blocks. Given that the gatekeeper for getting into the reserved range is a PR could we go more lightweight approach? [note, I am aware it is not as robust!]

Specifically 
1. A software creator can register a block of codes by means of PR.
   - The first entry(s) must be for a real implementation - i.e. no speculative reservation because we might "do this one day".
   - No need to add the whole range as items. They can do this as a comment in XML. Yes I know this isn't really "reservation"
   - if you choose to reserve the whole block that is OK too. 
   - New blocks must be added in units of the block size and must be adjacent. I.e. you can't come along and decide to put your new block mid way through the unallocated reserved range. 
   
I prefer 10 for block allocation, but not precious about it. 

Yes to all the rest. The point at which a name can't be changed is the point where the protocol is published in the (as yet) undefined XML.

olliw42, 2019-09-09:
> it is always easier to comment on an idea than come up with your own.

yeah, I know :). 

The "problem" here is that someone must have had once some ideas when the V2_EXTENSION message was set up as it is ... I would have found it interesting to hear that ... and I also didn't wanted to collide. However, this maybe has happened too far in the past and memory is lost :)

> The only concern I have is the one you raised about rendering of large blocks.

following the discussion, I'm not sure if "rendering" means only the presentation in the web documentation or also the listing in the code (for C e.g. in common.h).

I'm talking only about the rendering on the web, but not the code. IMHO, in the code all the blocks must be fully represented, for each registered code there should be an entry in the enum. Anything else didn't actually cross my mind, to be honest. For the purpose of the PR, listing them all looks correct to me. 

> Given that the gatekeeper for getting into the reserved range is a PR could we go more lightweight approach? 

I don't have a very strong opinion here, but I'm not sure I like that much. It's more lightweight as regards the number of lines in the code, but not more lightweight as regards the procedures and maintenance. I tried to come up with something which is just plain simple, yet works.

> The first entry(s) must be for a real implementation - i.e. no speculative reservation because we might "do this one day".

in this point I guess we have a somewhat different opinion. In some sense, the very fact that blocks can be reserved implies that many codes are speculative in the sense of "we might use it one day".

Also, it actually has some development advantages if the codes can be reserved but do not need to be fixed in stone immediately. I read your statement to imply that for the first code rules 4 & 5 apply. I would find this a bit too demanding. For instance, one could not really do a beta release. "Beta" to me means public testing, but public means registered code.

If the purpose was to "just" avoid that now all sorts of folks show up and cry "Me too, me too" without having an real application then there is a point in here. I think that this rather could be handled by means of accepting a PR. One could e.g. ask for some "evidence". If it's e.g. producer you cooperate with then this could be evidence enough. If it's an external or even hobbyist like me one could ask for some solid evidence such as a video, a alpha release, or such.

> New blocks must be added in units of the block size and must be adjacent. I.e. you can't come along and decide to put your new block mid way through the unallocated reserved range.

absolutely. This was what I tried to express with "_I would suggest that the code blocks are aligned to multiples of the default (e.g. 25), that is each block starts at a code devisible by 25_"

I like to mention that it is under your full control through the standard PR mechanisms. So some "you" may come along but you always can rectify.

> I prefer 10 for block allocation, but not precious about it.

yeah, this was my first impulse too ... but then I calculated that with 25 one still can support 1300 software creators ... so we probably could even give away blocks of 100 without any damage in future ... hence I settled in between, on 25 LOL

olliw42, 2019-09-09:
so, let me try to improve the enum thing:

maybe we can do the blocking also be means of the code name:

Instead of adding `MAV_TUNNEL_PAYLOAD_TYPE_XXXXX_RESERVED0` to `MAV_TUNNEL_PAYLOAD_TYPE_XXXXX_RESERVEDMAX`, we simply add only one enum `MAV_TUNNEL_PAYLOAD_TYPE_XXXXX_RESERVED0_TO_MAX`. E.g., if the block size is 25 I could ask for an initial entry

MAV_TUNNEL_PAYLOAD_TYPE_STORM32_RESERVED0_TO_24 = 200

This would make it very explicit what the allocated block range is.

If I would then want to register a first named code, the enum could be changed to 

MAV_TUNNEL_PAYLOAD_TYPE_STORM32_CODENAME = 200,
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_RESERVED1_TO_24 = 201

with the 2nd next named code to 

MAV_TUNNEL_PAYLOAD_TYPE_STORM32_CODENAME = 200,
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_ANOTHERCODENAME = 201,
MAV_TUNNEL_PAYLOAD_TYPE_STORM32_RESERVED2_TO_24 = 202

and os on

hamishwillee, 2019-09-10:
@olliw42 
- Yes it would be nice to know what the intent for V2_EXTENSION was. But I don't think it was necessarily though out much further than what has been written and in any case I am happy to clarify this now. 
- My concerns about allocation of individual entries for blocks are that it is easy to make copy/paste errors, and that it is inefficient information transfer - common.xml page is already very big both for reading and for download (which matters in places like China). 
  Weighed against that is the reason for having reserved, which is to allow the tooling to detect accidental clashes, and the fact that in reality we won't have too many such blocks. 

For now I say we do this right - as you you suggested originally.

But we switch to 10 as the default block size to reserve. Also, we make it clear that even though you don't have to reserve the whole block, you do have to reserve on the decade boundary. Seem reasonable?

hamishwillee, 2019-09-10:
I would also quite like to move the default location of https://github.com/mavlink/mavlink/tunnel-message-payload-types.xml to out of the root of mavlink/mavlink. Not sure best name for a containing folder - perhaps something like "payload_schema"

olliw42, 2019-09-10:
10 is perfectly fine with me
and the spec allows to ask for bigger blocks if someone really needs, now in multiple of 10, should be perfectly fine
(I've changed it in the above post so that there is a sort of correct reference)

as regards the file location, I think I can't really be of much help ... this really should fit you guys preferences
the one though which crosses my mind is how it would work with PR-ing if it's not in mavlink/mavlink, would it be then an extra git repo? So, one would have to do two PRs, one in mavlink/mavlink for the enum, and one in this other git for the xml? I guess I would consider this inconvenient, also from your side, as enum and xml changes could come together, see 4

hamishwillee, 2019-09-10:
No, I was thinking of in a sub folder rather than in the root of mavlink/mavlink. It's just a tidyness thing since we might have a number of these definitions.

olliw42, 2019-09-10:
ah, ok, sure, this makes of course lots of sense

so potentially there might be more such kind of files pop up over time, not only for V2_EXTENSION and TUNNEL with their payloads, but who knows what kind of files ... so something quite generic might be good

'xml', 'xmls' 'xml_files', 'support_files', 'auxiliary_files', 'schema_files', 'definition_files', 'definition_xml_files', ...

hamishwillee, 2019-09-10:
@olliw42 FYI, captured this in https://github.com/mavlink/mavlink-devguide/pull/207

I expect you'll have some comments. I consider it a draft. Which I don't expect to submit until it's had some review from a number of people.

hamishwillee, 2019-09-10:
Re https://github.com/mavlink/mavlink/issues/1225#issuecomment-529792762

Yes!

Probably definition_files but have a readme in there explaining what they should be.  I guess though we have no definition files, and no pull yet to create them. I'd rather not invent work until it is needed.

olliw42, 2019-09-10:
so should I change PR #1233 to include the new location https://github.com/mavlink/mavlink/definition_files/tunnel-message-payload-types.xml, or is that PR fine then as is?

yeah, no definition files to go into definition_files yet ;)

olliw42, 2019-09-10:
thx for mavlink/mavlink-devguide#207

olliw42, 2019-09-10:
> I consider it a draft. Which I don't expect to submit until it's had some review from a number of people.

great. It indeed would be great to have more opinions, to get it right, ideally it would be useful to many ... and there is no pressure or hurry

I was just a bit 'pressuring' to get the payload codes registered ... there are currently about 10 lose ends in the STorM32's MAVLink code section and I of course like to get them closed step by step :)

THX

closed 2019-09-20 by hamishwillee