mavlink/mavlink #1242: INSTALLATION ISSUE

HaroldSP, 2019-10-01:
ISSUE installing Mavlink 1/10/2010

Please add detailed instructions in README file or installation guide

System: 

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic

Pre-install:

#!/bin/bash

# sudo apt install build-essential manpages-dev git automake autoconf 
# gcc --version 
# sudo apt-get install python3-pip 
# sudo apt install python-pip 

# pip install future 
# sudo apt-get install python-dev libxml2 libxml2-dev libxslt-dev 
# pip3 install lxml 
# sudo pip2 install -U future lxml 
# sudo apt install xmlto 

# sudo apt-get install python-matplotlib 

#sudo apt-get install cmake 

#pip install pymavlink 
# cd mavlink/pymavlink
# python setup.py install

Configuring:

#rm ../CMakeCache.txt
mkdir mavlink-build
cd mavlink-build
cmake ..

Result:
-- Configuring done
-- Generating done
-- Build files have been written to: $HOME/mavlink/build

Errors in make command:

IOError: [Errno 2] No such file or directory: '$HOME/mavlink/message_definitions/v0.9/test.xml'
CMakeFiles/test.xml-v0.9.dir/build.make:60: recipe for target 'test.xml-v0.9-stamp' failed
make[2]: *** [test.xml-v0.9-stamp] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/test.xml-v0.9.dir/all' failed
make[1]: *** [CMakeFiles/test.xml-v0.9.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

hamishwillee, 2019-10-01:
Thanks. I **think** this should have been fixed last night by https://github.com/mavlink/mavlink/pull/1241 . Did you test this using a current build?

HaroldSP, 2019-10-03:
Thanks, it seems fine now!

closed 2019-10-03 by HaroldSP