mavlink/mavlink #1244: Pl clarify compid = 0 (MAV_COMP_ID_ALL) in the message header

olliw42, 2019-10-03:
the current docu is somewhat confusing on the matter, but it seems that it supports using MAV_COMP_ID_ALL (= 0) for the component ID in the message header.

This to me looks weird, since naively I would have thought that any component in a MAVLink system should have a unique component ID, and thus send out messages only with the component ID field in the header be populated accordingly. Having a component sending out messages with component ID = MAV_COMP_ID_ALL = 0 in the header doesn't make sense to me. 

Note that this is not about having target component ID = 0, which means broadcasting and of course makes lots of sense. Note also that in the routing rules there is no mention whatsoever of a message component ID = 0.

I've stumbled across this since QGC appears to (occasionally !?) send out messages with component ID = 0 in the message header, specifically e.g. the requestCameraInformation. 

My naive expectation effectively means that there can't be messages with msg.compid = 0, but if I add code to STorM32 which rejects all messages with msg.compid = 0 based on my naive expectation I find that QGC's requestCameraInformation command is rejected and the MAVLink camera service be broken. My debug output also explicitly shows that QGC's requestCameraInformation command has msg.compid = 0.

So, the above issue appears to not only be academic, but be of some relevance. 

If component ID = MAV_COMP_ID_ALL = 0 in the header does make some sense, I would wish that the docu could be improved to better explain what the usage is. If it doesn't and indeed is effectively forbidden, i.e. allowed only for target, the docu could say this.

thx for all your work on MAVLink

hamishwillee, 2019-10-04:
IMO your "naive" impression is correct. System and component IDs are used for addressing, and 0 is used to indicate broadcast. Obviously this means that if a system or component identifies itself as having an address of 0 in the sender component id this means that the component is not separately addressable. [Again note, this is sender id in the packet fields not target id in the payload]

This is clearly a mis-use but is it a big problem?
- For receiving broadcast messages, clearly not.
- If receiving a message or command then all components on the receiving system will receive the message. This could have unintended consequences depending on the command.
- If sending a command the ACK will be seen by everything on the commanding system. Probably not a problem.

I don't think semantically a component should ever need to identify itself as MAV_COMP_ID_ALL and this would break things. 

In terms of docs, we would need to clarify this in https://mavlink.io/en/guide/serialization.html - which states 0 is valid. 

As to the specific response - i.e. dropping the packet, I'm not sure. Arguably this might be best handled at the generator level - e.g. creating a runtime exception or similar.

@julianoes Does this make sense?

hamishwillee, 2019-10-04:
@olliw42 Either way this is a bug for QGC camera implementation. Can you please create a defect report.

olliw42, 2019-10-04:
many thx

if it's a big problem or not is also really implementation specific. For instance, I was adding a line if(msg.compId==0) return very high up in my parser since it made case checking for some things further down easier, and that's how I found the issue. If compId=0 is effectively not allowed by spec this should have been a valid implementation, but currently it makes the things not working ... "not working" is maybe not exactly a non-problem :D

I will drop an issue in QGC

When I was mentioning that the docs are somewhat confusing, I was referring to here: https://mavlink.io/en/messages/common.html#MAV_COMPONENT. 
I think it should just bluntly be said that 0 is for target ID but not source ID.

hamishwillee, 2019-10-06:
Thanks. Added comment in https://github.com/mavlink/mavlink/pull/1245 but further work will need to be done in other docs.

olliw42, 2019-10-07:
since this appears to be agreed, I can close I guess

closed 2019-10-07 by olliw42