mavlink/mavlink #1251: PLAY_TUNE message should be non-ambiguous

JonasVautherin, 2019-10-17:
I am not so convinced about the [PLAY_TUNE](https://mavlink.io/en/messages/common.html#PLAY_TUNE) message in the specs. It reads:

```
target_system    | uint8_t   | System ID
--               | --        | --
target_component | uint8_t   | Component ID
tune             | char[30]  | tune in board specific format
tune2 **         | char[200] | tune extension (appended to tune)
```

My point here is that `tune in board specific format` is a "non-spec": it says "we have a message that says that there is no standard way to do that".

What PX4 implements is using the [QBasic Play](https://www.qbasic.net/en/reference/qb11/Statement/PLAY-006.htm) syntax:

```
Octave and tone commands:
 Ooctave    Sets the current octave (0 - 6).
 < or >     Moves up or down one octave.
 A - G      Plays the specified note in the current octave.
 Nnote      Plays a specified note (0 - 84) in the seven-octave
            range (0 is a rest).

Duration and tempo commands:
 Llength    Sets the length of each note (1 - 64). L1 is whole note,
            L2 is a half note, etc.
 ML         Sets music legato.
 MN         Sets music normal.
 MS         Sets music staccato.
 Ppause     Specifies a pause (1 - 64). P1 is a whole-note pause,
            P2 is a half-note pause, etc.
 Ttempo     Sets the tempo in quarter notes per minute (32 - 255).

Mode commands:
 MF          Plays music in foreground.
 MB          Plays music in background.

Suffix commands:
 # or +      Turns preceding note into a sharp.
 -           Turns preceding note into a flat.
 .           Plays the preceding note 3/2 as long as specified.
```

And it makes perfect sense to me. I believe that the specs should read: "tune in the qbasic play format". That would make it unambiguous.

hamishwillee, 2019-10-17:
You _can_ only make it unambiguous if all systems that already use this message use the same tune format. You _should_ only make this unambiguous if this is the best format for tunes, will be so forever, and existing systems that might want to use this message don't implement some other valid alternative. Is that the case?

I doubt it. QBasic already feels like a bit of a dead format - at least it is hard to find a decent player for windows (no, I do not count the one here: http://dev.px4.io/master/en/advanced/system_tunes.html)

Probably best option is a new message that allows you to specify the format. Possibly also allows you to set an arbitrary length.

Field Name | Type | Description
--- | --- | ---
target_system | uint8_t | System ID
target_component | uint8_t | Component ID
format | uint8_t | Enum, with 0 being QBasic
index | uint8_t | 0 for 1st buffer, 1 for 2nd buffer in multi-buffer definition.
maxindex | uint8_t | Number of buffers-1
tune | char[whatever is left] | Tune data

There probably also needs to be an ACK to indicate that the message was supported - ie so that the sender knows that the receiver can use the specified format and that it received the second part of the buffer. 

If there is no ack and no need for longer tunes we would want a mechanism to determine what formats are supported (and we might want that anyway). That could just be a command and associated info message that gets emitted with the supported types.

Thoughts?

JonasVautherin, 2019-10-18:
> QBasic already feels like a bit of a dead format

Doesn't mean it's not good enough :sunglasses:. I find that the syntax pasted above is pretty complete.

> Is that the case?

At least [APM](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/ToneAlarm.cpp#L33-L35) kind of does it, with one [small extension](http://firmware.ardupilot.org/Tools/ToneTester/#MFT240L8%20O4aO5dc%20O4aO5dc%20O4aO5dc%20L16dcdcdcdc):

```
// Tunes follow the syntax of the Microsoft GWBasic/QBasic PLAY
//   statement, with some exceptions and extensions.
// See http://firmware.ardupilot.org/Tools/ToneTester/
```

The extension being:

```
Extensions/variations:

  MB MF  The MF command causes the tune to play once and then stop. The MB command causes the
         tune to repeat when it ends.
```

> Enum, with 0 being QBasic

Because APM and PX4 seem to agree on QBasic, I think we should consider keeping only QBasic. So that clients (QGC/MAVSDK/...) can confidently say that they know how to play a tune.

> Possibly also allows you to set an arbitrary length.

Possibly, if we have to create a new message. But I guess the 30/230 chars of the current one may be enough for most cases (in case we don't need to create a new message).

> There probably also needs to be an ACK to indicate that the message was supported.

Same as above. If we can stick to QBasic, I guess we don't need that either.

LorenzMeier, 2019-10-18:
Could you send a PR to fix up the docs for the message?

JonasVautherin, 2019-10-18:
> Could you send a PR to fix up the docs for the message?

When we agree on what it should say, sure!

hamishwillee, 2019-10-20:
@LorenzMeier Are you OK with **hard coding** QBasic format as the "MAVLink defacto standard"? 

Personally I am no - I'm not sure that anyone designing a flight stack now would necessarily choose this format. Normally for this sort of breaking change we'd create a new message or extension.

hamishwillee, 2019-10-22:
@JonasVautherin FYI, according to @auturgy ArduPilot has already flipped between several formats and sees no benefit in fixing the version. 

He also didn't see any particular benefit in specifying or getting the supported version because this can be known from the firmware. I argue that it is useful for interoperability with unknown systems.

LorenzMeier, 2019-10-22:
We can't afford keeping payload data formats undefined. It needs to be a clear thing. Given how it is being used I stand by using QBASIC for this go-around. We can still add a format field later that defaults to 0 = QBasic without that addition being a breaking change.

hamishwillee, 2019-10-22:
@LorenzMeier I disagree. If you specify QBASIC for _this_ go-around it is a breaking change. I don't disagree that the formats should be defined but there are millions of ways around that - starting with adding a format field now.

JonasVautherin, 2019-10-22:
> He also didn't see any particular benefit in specifying or getting the supported version because this can be known from the firmware.

To me it is exactly the point of a standard. Taking HTML as an example: I could make a browser that understands `<g>` ("gras", which is the equivalent to "bold" in french) instead of `<b>`. And I could argue that if you implement a website that should work on my browser, it's easy for you to know that I use `<g>` and not `<b>`. But the fact is that you don't want to know about my browser.

> I argue that it is useful for interoperability with unknown systems.

I would say the opposite: if you standardise, then it works with unknown systems. If you don't, then by definition you can't support unknown system. How could I possibly support the custom tune format of a system I don't know in MAVSDK?

> If you specify QBASIC for this go-around it is a breaking change.

Yes, that's possible, and I can't argue against that. Except that as a rule, we should never add a message that says "write your custom payload here" in the first place. If there is a good way to do one thing, we can standardise it. If there isn't, then maybe it should not be part of the standard.

I don't see a kind of tune that you may need and that could not be described by the QBasic syntax :sweat_smile:.

hamishwillee, 2019-10-22:
@JonasVautherin This isn't enough of a big deal for me to continue arguing. Create a PR that does what you want and we'll go with that. If someone does come along with a different message format they want to support we'll create PLAY_TUNE_V2 to handle that.

JonasVautherin, 2019-10-23:
I modified your PR (#1255) to show what I meant. I realize I should have done that from the beginning on, it would have been simpler.

My point with that issue was just to suggest that we agree on saying "the string format should be QBasic", instead of allowing other formats. If we ever get into the situation where we need another format, then it will still be time to add a "format" extension like you did. But if that's not necessary, let's not open that door.

auturgy, 2019-10-23:
fyi
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/MMLPlayer.cpp
This is trivia. But you need to consider that it isn’t if or when: it’s as soon as you merge.

Regards,

James

hamishwillee, 2019-10-23:
I already gave up on this. But if everyone would prefer to update to a new message then that is easy enough. Definition could be as simple as below. YOu'd mark the original as superseded. 

```
    <enum name="TUNE_FORMAT">
      <description>These flags encode the MAV mode.</description>
      <entry value="0" name="TUNE_FORMAT_QBASIC1_1">
        <description>Format is QBasic 1.1 Play: https://www.qbasic.net/en/reference/qb11/Statement/PLAY-006.htm.</description>
      </entry>
    </enum>

    <message id="301" name="PLAY_TUNE_V2">
      <description>Control vehicle tone generation (buzzer).  Default is QBasic 1.1.</description>
      <field type="uint8_t" name="target_system">System ID</field>
      <field type="uint8_t" name="target_component">Component ID</field>
      <field type="uint8_t" name="format" enum="TUNE_FORMAT">Tune format.</field>
      <field type="char[240]" name="tune">Tune definition.</field>
    </message>
```

JonasVautherin, 2019-10-23:
> I already gave up on this. But if everyone would prefer to update to a new message then that is easy enough.

I really thought that since APM and PX4 seemed to be on the same page, it was easy to just formalize that in the specs. Not sure if APM and PX4 will put resources into moving to a new message just for the sake of formalizing that.

Seems like status quo is simplest. MAVSDK implementation works for PX4 and may work for APM, and I guess we'll see if it ever gets a bug report.

Sorry for the inconvenience :sweat_smile:.

closed 2019-10-23 by JonasVautherin