mavlink/mavlink #1254: Control YAW in GPS vial MAVLINK

bigboy061293, 2019-10-22:
I have a question about feeding GPS into Cube running Ardupilot 3.6.11 (or 4.0 beta). 

My GPS module return NMEA message with HDT sentence on heading information which I want to feed into Ardupilot. I have examined the code https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_MAV.cpp  seeing that yaw value is accessible but this GPS_INPUT does not contain yaw information https://mavlink.io/en/messages/common.html#GPS_INPUT

Anyone experiences this before? Thanks!

amilcarlucas, 2019-10-26:
The ardupilot codebase uses a mavlink fork that contains the missing yaw information.
I have opened a PR against this upstream repository here: https://github.com/mavlink/mavlink/pull/1206

So until that one gets merged, the information will still be missing in this repo and in mavros.

amilcarlucas, 2019-11-08:
Got merged now, so this is solved.

closed 2019-11-08 by amilcarlucas