mavlink/mavlink #1259: Narrowing conversion error in c_library_v2 common

lkm1321, 2019-10-31:
Hi, 

I'm getting narrowing conversion error with the master branch release in https://github.com/mavlink/c_library_v2. Commit hash bbbc401c95e526bf6e87afc8e97d95ef3207f3d7

It seems a valid error, as the header is trying to convert int 258 to uint8_t (which will overflow). 

/mavlink/include/common/common.h:27:0: error: narrowing conversion of ‘258’ from ‘int’ to ‘uint8_t {aka unsigned char}’ inside { } [-Wnarrowing]

The error goes away in commit hash 
618a74eb748db200e268bfbdacbca612ddc98bc1

amilcarlucas, 2019-11-08:
I think this got solved 3 days ago.

hamishwillee, 2019-11-13:
@lkm1321 This can be closed now right?

lkm1321, 2019-11-14:
I haven't had a chance to check the latest version. I am tracking a fixed
hash for now. But if you're sure, feel free to close.

closed 2019-11-15 by hamishwillee