mavlink/mavlink #1260: Mavlink Camera: Pl clarify Timelapsed Image Capture

olliw42, 2019-10-31:
The camera microservice documentation is pretty clear on how image capturing works for missions, but is confusing/illogical for interactive user-initiated image capture through a GUI. The issue is gaining relevance by the fact that QGC's implementation doesn't appear to follow any "rule", as I've learned from recently digging somewhat into QGC's code.

1) Issue with docs

The image capture sequence for interactive user-initiated image capture through a GUI is illustarted by a scheme, there the capturing is triggered by a MAV_CMD_IMAGE_START_CAPTURE (interval, count/forever), which obviously through interval/count/forever allows timelapsed capturing, but then only exactly one picture should be taken: https://mavlink.io/en/services/camera.html#camera-capture-status. This is strange in two aspects: It makes sense only if only a single-shot is triggered with MAV_CMD_IMAGE_START_CAPTURE, and 2ndly it doesn't describe a procedure of how to take timelapsed pictures.

2) Issue with QGC

The docs seem to be clear that after each taken photo the camera should emit a CAMERA_IMAGE_CAPTURED. QGC however doesn't even accept and process this message, but only processes CAMERA_CAPTURE_STATUS messages, which however is not requested in timelapsed mode, only in video and single-shot photo. Yet it counts up the taken photos, which is obvously based only on its internal timer, and not by feedback from the camera, and which hence will necessarily run out of sync with the camera with time. And since it doesn't get CAMERA_CAPTURE_STATUS messages it also does not update the storage device information (available free memory on card) druing the timelapse.

Thus: 

As regards GUI initiated timelapsed image capturing, there is unclearness as regards 

* if timelapsed capturing should be done based on single-shot by the GUI or by the interval mechanism available through MAV_CMD_IMAGE_START_CAPTURE,

* and what data/messages the camera should report back or should be requested by the GUI.

Several approaches come to mind:

* The GUI repeatedly triggers single-shot photos based on it's internal timer, the camera reports back a CAMERA_IMAGE_CAPTURED based on which the GUI increases its photo counter, and the GUI requests a CAMERA_CAPTURE_STATUS after each shot to update e.g. the available memory 

* The GUI triggers timelapsed capturing using MAV_CMD_IMAGE_START_CAPTURE with respective interval/count values, the camera reports back a CAMERA_IMAGE_CAPTURED, and the GUI repeatedly requests CAMERA_CAPTURE_STATUS based on some timing rational of its own brew

* The GUI triggers timelapsed capturing using MAV_CMD_IMAGE_START_CAPTURE with respective interval,count values, and the camera reports back BOTH a CAMERA_IMAGE_CAPTURED and a CAMERA_CAPTURE_STATUS after each shot

Ideas?

hamishwillee, 2019-11-13:
@dogmaphobic Can you please help

For [the docs](https://mavlink.io/en/services/camera.html#still-image-capture), the implication is that for  _interactive user-initiated image capture through a GUI_ you can only capture a single image at a time
- is that correct?
- presumably this means timelapse is not supported?

This also implies that the sequence diagram for this case is slightly wrong, because it implies that you can specify a repeat/interval (and above we're saying you can't).
[image]

Anyway Gus, can't fix the docs until those above clarified. Thanks.

hamishwillee, 2019-12-09:
@dogmaphobic Ping ^^^