mavlink/mavlink #1262: Mavlink c++ generation

AlexisTM, 2019-11-04:
When compiling the following program, including `common.hpp` from https://github.com/mavlink/mavlink/commit/ba8a5b3dac343b728e5ae5fd151b2ae97eda784b

```
#include <mavlink_v2/common/common.hpp>

int main() {
  return EXIT_SUCCESS;
}
```

With c++11 standard and gcc 7.4

```
> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.4.0-1ubuntu1~18.04.1' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)
```

I get the following errors.

```log
In file included from /home/alexis/ros/src/mavlink_driver/src/test.cpp:1:0:
/home/alexis/ros/install/include/mavlink_v2/common/common.hpp:24:0: error: missing initializer for member ‘mavlink::__mavlink_msg_entry::target_component_ofs’ [-Werror=missing-field-initializers]
 constexpr std::array<mavlink_msg_entry_t, 188> MESSAGE_ENTRIES {{ {0, 50, 9, 0, 0, 0}, {1, 124, 31, 0, 0, 0}, {2, 137, 12, 0, 0, 0}, {4, 237, 14, 3, 12, 13}, {5, 217, 28, 1, 0, 0}, {6, 104, 3, 0, 0, 0}, {7, 119, 32, 0, 0, 0}, {8, 117, 36, 0, 0, 0}, {11, 89, 6, 1, 4, 0}, {20, 214, 20, 3, 2, 3}, {21, 159, 2, 3, 0, 1}, {22, 220, 25, 0, 0, 0}, {23, 168, 23, 3, 4, 5}, {24, 24, 30, 0, 0, 0}, {25, 23, 101, 0, 0, 0}, {26, 170, 22, 0, 0, 0}, {27, 144, 26, 0, 0, 0}, {28, 67, 16, 0, 0, 0}, {29, 115, 14, 0, 0, 0}, {30, 39, 28, 0, 0, 0}, {31, 246, 32, 0, 0, 0}, {32, 185, 28, 0, 0, 0}, {33, 104, 28, 0, 0, 0}, {34, 237, 22, 0, 0, 0}, {35, 244, 22, 0, 0, 0}, {36, 222, 21, 0, 0, 0}, {37, 212, 6, 3, 4, 5}, {38, 9, 6, 3, 4, 5}, {39, 254, 37, 3, 32, 33}, {40, 230, 4, 3, 2, 3}, {41, 28, 4, 3, 2, 3}, {42, 28, 2, 0, 0, 0}, {43, 132, 2, 3, 0, 1}, {44, 221, 4, 3, 2, 3}, {45, 232, 2, 3, 0, 1}, {46, 11, 2, 0, 0, 0}, {47, 153, 3, 3, 0, 1}, {48, 41, 13, 1, 12, 0}, {49, 39, 12, 0, 0, 0}, {50, 78, 37, 3, 18, 19}, {51, 196, 4, 3, 2, 3}, {52, 132, 7, 0, 0, 0}, {54, 15, 27, 3, 24, 25}, {55, 3, 25, 0, 0, 0}, {61, 167, 72, 0, 0, 0}, {62, 183, 26, 0, 0, 0}, {63, 119, 181, 0, 0, 0}, {64, 191, 225, 0, 0, 0}, {65, 118, 42, 0, 0, 0}, {66, 148, 6, 3, 2, 3}, {67, 21, 4, 0, 0, 0}, {69, 243, 11, 1, 10, 0}, {70, 124, 18, 3, 16, 17}, {73, 38, 37, 3, 32, 33}, {74, 20, 20, 0, 0, 0}, {75, 158, 35, 3, 30, 31}, {76, 152, 33, 3, 30, 31}, {77, 143, 3, 3, 8, 9}, {81, 106, 22, 0, 0, 0}, {82, 49, 39, 3, 36, 37}, {83, 22, 37, 0, 0, 0}, {84, 143, 53, 3, 50, 51}, {85, 140, 51, 0, 0, 0}, {86, 5, 53, 3, 50, 51}, {87, 150, 51, 0, 0, 0}, {89, 231, 28, 0, 0, 0}, {90, 183, 56, 0, 0, 0}, {91, 63, 42, 0, 0, 0}, {92, 54, 33, 0, 0, 0}, {93, 47, 81, 0, 0, 0}, {100, 175, 26, 0, 0, 0}, {101, 102, 32, 0, 0, 0}, {102, 158, 32, 0, 0, 0}, {103, 208, 20, 0, 0, 0}, {104, 56, 32, 0, 0, 0}, {105, 93, 62, 0, 0, 0}, {106, 138, 44, 0, 0, 0}, {107, 108, 64, 0, 0, 0}, {108, 32, 84, 0, 0, 0}, {109, 185, 9, 0, 0, 0}, {110, 84, 254, 3, 1, 2}, {111, 34, 16, 0, 0, 0}, {112, 174, 12, 0, 0, 0}, {113, 124, 36, 0, 0, 0}, {114, 237, 44, 0, 0, 0}, {115, 4, 64, 0, 0, 0}, {116, 76, 22, 0, 0, 0}, {117, 128, 6, 3, 4, 5}, {118, 56, 14, 0, 0, 0}, {119, 116, 12, 3, 10, 11}, {120, 134, 97, 0, 0, 0}, {121, 237, 2, 3, 0, 1}, {122, 203, 2, 3, 0, 1}, {123, 250, 113, 3, 0, 1}, {124, 87, 35, 0, 0, 0}, {125, 203, 6, 0, 0, 0}, {126, 220, 79, 0, 0, 0}, {127, 25, 35, 0, 0, 0}, {128, 226, 35, 0, 0, 0}, {129, 46, 22, 0, 0, 0}, {130, 29, 13, 0, 0, 0}, {131, 223, 255, 0, 0, 0}, {132, 85, 14, 0, 0, 0}, {133, 6, 18, 0, 0, 0}, {134, 229, 43, 0, 0, 0}, {135, 203, 8, 0, 0, 0}, {136, 1, 22, 0, 0, 0}, {137, 195, 14, 0, 0, 0}, {138, 109, 36, 0, 0, 0}, {139, 168, 43, 3, 41, 42}, {140, 181, 41, 0, 0, 0}, {141, 47, 32, 0, 0, 0}, {142, 72, 243, 0, 0, 0}, {143, 131, 14, 0, 0, 0}, {144, 127, 93, 0, 0, 0}, {146, 103, 100, 0, 0, 0}, {147, 154, 36, 0, 0, 0}, {148, 178, 60, 0, 0, 0}, {149, 200, 30, 0, 0, 0}, {162, 189, 8, 0, 0, 0}, {230, 163, 42, 0, 0, 0}, {231, 105, 40, 0, 0, 0}, {232, 151, 63, 0, 0, 0}, {233, 35, 182, 0, 0, 0}, {234, 150, 40, 0, 0, 0}, {235, 179, 42, 0, 0, 0}, {241, 90, 32, 0, 0, 0}, {242, 104, 52, 0, 0, 0}, {243, 85, 53, 1, 52, 0}, {244, 95, 6, 0, 0, 0}, {245, 130, 2, 0, 0, 0}, {246, 184, 38, 0, 0, 0}, {247, 81, 19, 0, 0, 0}, {248, 8, 254, 3, 3, 4}, {249, 204, 36, 0, 0, 0}, {250, 49, 30, 0, 0, 0}, {251, 170, 18, 0, 0, 0}, {252, 44, 18, 0, 0, 0}, {253, 83, 51, 0, 0, 0}, {254, 46, 9, 0, 0, 0}, {256, 71, 42, 3, 8, 9}, {257, 131, 9, 0, 0, 0}, {258, 187, 32, 3, 0, 1}, {259, 92, 235, 0, 0, 0}, {260, 146, 5, 0, 0, 0}, {261, 179, 27, 0, 0, 0}, {262, 12, 18, 0, 0, 0}, {263, 133, 255, 0, 0, 0}, {264, 49, 28, 0, 0, 0}, {265, 26, 16, 0, 0, 0}, {266, 193, 255, 3, 2, 3}, {267, 35, 255, 3, 2, 3}, {268, 14, 4, 3, 2, 3}, {269, 109, 213, 0, 0, 0}, {270, 59, 19, 0, 0, 0}, {299, 19, 96, 0, 0, 0}, {300, 217, 22, 0, 0, 0}, {310, 28, 17, 0, 0, 0}, {311, 95, 116, 0, 0, 0}, {320, 243, 20, 3, 2, 3}, {321, 88, 2, 3, 0, 1}, {322, 243, 149, 0, 0, 0}, {323, 78, 147, 3, 0, 1}, {324, 132, 146, 0, 0, 0}, {330, 23, 158, 0, 0, 0}, {331, 91, 230, 0, 0, 0}, {332, 236, 239, 0, 0, 0}, {333, 231, 109, 0, 0, 0}, {334, 135, 14, 0, 0, 0}, {335, 225, 24, 0, 0, 0}, {340, 99, 70, 0, 0, 0}, {350, 232, 20, 0, 0, 0}, {360, 11, 25, 0, 0, 0}, {365, 36, 255, 0, 0, 0}, {370, 98, 73, 0, 0, 0}, {371, 161, 50, 0, 0, 0}, {375, 251, 140, 0, 0, 0}, {380, 232, 20, 0, 0, 0}, {385, 147, 133, 3, 2, 3}, {390, 156, 238, 0, 0, 0}, {9000, 113, 137, 0, 0, 0}, {12900, 197, 22, 0, 0, 0}, {12901, 16, 37, 0, 0, 0}, {12902, 181, 31, 0, 0, 0}, {12903, 149, 24, 0, 0, 0}, {12904, 238, 21, 0, 0, 0}, {12905, 56, 21, 0, 0, 0}, {12915, 67, 252, 0, 0, 0} }};

/home/alexis/ros/install/include/mavlink_v2/common/common.hpp:24:0: error: missing initializer for member ‘mavlink::__mavlink_msg_entry::target_component_ofs’ [-Werror=missing-field-initializers]
/home/alexis/ros/install/include/mavlink_v2/common/common.hpp:24:0: error: missing initializer for member ‘mavlink::__mavlink_msg_entry::target_component_ofs’ [-Werror=missing-field-initializers]
[... one per message ...]

/home/alexis/ros/src/mavlink_driver/src/test.cpp: In function ‘int main(int, char**)’:
/home/alexis/ros/src/mavlink_driver/src/test.cpp:3:14: error: unused parameter ‘argc’ [-Werror=unused-parameter]
 int main(int argc, char** argv) {
              ^~~~
/home/alexis/ros/src/mavlink_driver/src/test.cpp:3:27: error: unused parameter ‘argv’ [-Werror=unused-parameter]
 int main(int argc, char** argv) {
                           ^~~~
In file included from /home/alexis/ros/install/include/mavlink_v2/common/../message.hpp:35:0,
                 from /home/alexis/ros/install/include/mavlink_v2/common/common.hpp:16,
                 from /home/alexis/ros/src/mavlink_driver/src/test.cpp:1:
/home/alexis/ros/install/include/mavlink_v2/common/../msgmap.hpp: In instantiation of ‘void mavlink::MsgMap::msg_swap_memcpy(_T&, _Tin) [with _T = short unsigned int; _Tin = unsigned int]’:
/home/alexis/ros/install/include/mavlink_v2/common/../msgmap.hpp:113:3:   required from ‘void mavlink::MsgMap::operator<<(_T) [with _T = unsigned int]’
/home/alexis/ros/install/include/mavlink_v2/common/./mavlink_msg_heartbeat.hpp:59:16:   required from here
/home/alexis/ros/install/include/mavlink_v2/common/../msgmap.hpp:73:9: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
   buf = *static_cast<const _T *>(static_cast<const void *>(&data));
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
/home/alexis/ros/install/include/mavlink_v2/common/../msgmap.hpp:123:3:   required from ‘void mavlink::MsgMap::operator<<(_T) [with _T = double]’
/home/alexis/ros/install/include/mavlink_v2/common/../msgmap.hpp:139:9:   required from ‘void mavlink::MsgMap::operator<<(const std::array<_Tp, _Nm>&) [with _T = double; long unsigned int _Size = 16]’
/home/alexis/ros/install/include/mavlink_v2/common/./mavlink_msg_wheel_distance.hpp:54:16:   required from here
/home/alexis/ros/install/include/mavlink_v2/common/../msgmap.hpp:73:9: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]

```

julianoes, 2019-11-04:
It's "just" warnings, I think:
```
[-Werror=strict-aliasing]
   buf = *static_cast<const _T *>(static_cast<const void *>(&data));
```

AlexisTM, 2019-11-04:
Closed in favour of https://github.com/ArduPilot/pymavlink/issues/365

closed 2019-11-04 by AlexisTM