mavlink/mavlink #1264: More support for ICE monitoring

mwrightE38, 2019-11-07:
I think there needs to be more support for EFI's 

Devices like

hamishwillee, 2019-11-14:
@mwrightE38 Yes, that would be great. This is an open source project so what normally happens is that people who are invested in an area contribute a pull request for the messages they need. There are a bunch of people who then help sanity check.

A good starting point might be [these messages](https://github.com/ArduPilot/mavlink/compare/eba3a31e6649cf98d39d000df93ec428a040bf28...9559d86bd182147c494e3af0d1428bcbd2b7c3eb#diff-cf76872a518a69b492e8b757b86a3928) proposed in the ArduPilot project. They have been working with a number of EFI systems including [Megasquirt](http://megasquirt.info/) and [ecotrons](https://www.ecotrons.com/small_engine_fuel_injection_kit/) so this set should be a fairly representative useful set. 

Are you interested in reviewing and following up with a PR?

mwrightE38, 2019-11-14:
@hamishwillee that's close to what I've implemented in one of my branches.

I'm just not familiar with the exact structure of where EFI messages would be best put based on the current project structure. (in common or in each different dialect where useful) I'm happy to place a PR if someone can give feedback on where they'd like it.

hamishwillee, 2019-11-15:
Hmmm. Perhaps the best way is to take the ArduPilot EFI messages into common. Is there anything you need that they are missing?

@auturgy What's the best way to get the EFI messages into common? I have no particular concern about them other than the id used (mavlink 1). I also "wonder" why floats were used in some cases - e.g. engine load as a percentage.

hamishwillee, 2019-11-19:
@auturgy Is there a good way to progress this with ArduPilot ... and confirm "ECU Index" is an index number for multiple ICE?

If not, then I guess we start with a PR. @mwrightE38 it is worth waiting a week or so for Auturgy. IF he doesn't come back we'll talk about a PR for this in common.

auturgy, 2019-11-24:
ECU Index is the instance number, for multiple ecu’s.
The message ID is of course debatable, but imho this application is worth burning a ML1 number on.
Additional fields can be added as extensions.
Megasquirt and the CanberraUAV homebrew “EMU ECU” have been tested.
https://github.com/ArduPilot/ardupilot/pull/12816
https://github.com/ArduPilot/ardupilot/pull/12874
Ecotrons hasn’t been tested (the one we have is CAN based, so a bit more work).
WRT pulling up to common: the message needn’t be stack specific, but the development/implementation has been occurring in ArduPilot so far.

hamishwillee, 2019-11-24:
Thanks @auturgy.

Right thing to do now is pull it in as-is with new desired fields as extensions. Then we can discuss - it might be that with the learning of ArduPilot's development we can do better in common. 

@mwrightE38 You happy to create PR for the discussion?

hamishwillee, 2019-12-09:
@mwrightE38 Ping ^^^

mwrightE38, 2019-12-12:
@hamishwillee Sorry about the delay. Sure I'll make a pr within the next few days.

amilcarlucas, 2019-12-13:
The PR is already here: #1281

hamishwillee, 2019-12-15:
@amilcarlucas That's a PR for ardupilot. @mwrightE38 is interested in common, with some additional fields. I figure we bring it into comment then discuss.

auturgy, 2019-12-15:
There have been a couple of extensions added post @amilcarlucas sync PR, and probably a couple more to go.
@mwrightE38 what firmware are you using?

hamishwillee, 2019-12-15:
FYI, above, the extra fields @mwrightE38 was interested in are oil pressure, oil temp, and fuel pressure.

auturgy, 2019-12-16:
@mwrightE38 I’m assuming you know this, but for clarity, the Power4Flight device you’ve linked to above doesn’t appear to support either UAVCAN (it states Currawang compatible, which means Piccolo CAN) or MavLink (they use their own serial protocol) - although I haven’t got a copy of the ICD to confirm. That doesn’t mean we shouldn’t add the relevant messages, but does mean that testing against hardware would be precluded at this stage. Is my interpretation of the Power4Flight document correct?

hamishwillee, 2020-01-08:
@mwrightE38 See the EFI extensions @auturgy links to above. Is there anything else you need in this message?
If you don't need this in a particular flight stack "urgently" probably a good idea for us to let ArduPilot evolve this. If it suits your purpose we can pull it into common.

mwrightE38, 2020-01-08:
@auturgy Yes you're correct that device doesn't currently support mavlink but there is a plan in the future to add Mavlink or UAVCan (I'd assume UAVCan will be used) as a supported protocol. I tossed in a small p4flight serial to mavlink converter in between the intelliject and the autopilot when I used it. The real reason I was advocating for the messages is to be put in is for GCS monitoring.

@hamishwillee Sorry about the delay. No urgent need for me at all just to help out other folks who need it. Fields look good. That's everything that would need monitored real time plus enough to do integration. This can be closed.

closed 2020-01-08 by mwrightE38