mavlink/mavlink #1270: Altitude units are not specified

hdino, 2019-11-15:
I had a look at the different coordinate frames, but couldn't find any documentation of the altitude units used. See for example [https://mavlink.io/en/messages/common.html#MAV_FRAME](https://mavlink.io/en/messages/common.html#MAV_FRAME).

hamishwillee, 2019-11-19:
The units are defined by the messages they are used in - e.g. https://mavlink.io/en/messages/common.html#POSITION_TARGET_LOCAL_NED

closed 2019-11-19 by hamishwillee

hdino, 2019-11-19:
@hamishwillee Then some messages lack this information. Examples:

[MISSION_ITEM](https://mavlink.io/en/messages/common.html#MISSION_ITEM) - only specified for LOCAL frames.

[COMMAND_LONG](https://mavlink.io/en/messages/common.html#COMMAND_LONG) in conjunction with [MAV_CMD_NAV_WAYPOINT](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_WAYPOINT), [MAV_CMD_NAV_LOITER_UNLIM](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_UNLIM), [MAV_CMD_NAV_LOITER_TURNS](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TURNS), [MAV_CMD_NAV_LOITER_TIME](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LOITER_TIME), [MAV_CMD_NAV_TAKEOFF](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_TAKEOFF) etc.

In addition, the frame is not specified in these cases, so it's not clear whether the altitude is specified above WGS84, ground, MSL or something else.

Same applies to `pressure_alt` in [HIGHRES_IMU](https://mavlink.io/en/messages/common.html#HIGHRES_IMU) and [HIL_SENSOR](https://mavlink.io/en/messages/common.html#HIL_SENSOR).