mavlink/mavlink #1271: AUTOPILOT_VERSION ( #148 ) Extension

Stifael, 2019-11-18:
**Issue**
At Yuneec, we have several different product-configuration for the same type of board (similar to fmu-v2 [here](https://github.com/PX4/Firmware/tree/master/boards/px4/fmu-v2)), where each configuration is captured by one .cmake file and named via the `label` tag (https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v2/default.cmake#L6).
We want to capture that `label` and send it via the [AUTOPILOT_VERSION](https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION) message. However, it is not obvious how to do that with the current mavlink `AUTOPILOT_VERSION` message without extending the message with a new field.

Stifael, 2019-11-18:
FYI @julianoes

Stifael, 2019-11-18:
My current hack is to use the `product_id`-field because it captures the same information as `board_version`-field.

hamishwillee, 2019-11-18:
> However, it is not obvious how to do that with the current mavlink AUTOPILOT_VERSION message without extending the message with a new field.

Why not extend it with a new field? Are we at the 255byte limit on the message?

What would you call the field? `sw_config`? My understanding is that you can't use the other fields because none of those numbers change for different variants of components that might be built into the firmware. Is that right?

hamishwillee, 2019-11-20:
Well, I can't see any other way to do it unless we agree that each sw_config is a new product: in which case the product ID would be correct :-). So next step would be a PR. That would get more eyes on it, so possibly at that point there would be disagreement.

hamishwillee, 2019-11-27:
OK @Stifael I'm closing as answered. If you create a PR we can move on from there.

closed 2019-11-27 by hamishwillee

durka, 2021-10-21:
> I don't know enough here to understand how that is used or whether we need one or both bits of this information.

To be clear, the CONFIG_CDCACM_PRODUCTID is also in the product ID field :D 

Given that the board version field is currently filled by PX4, not sure how this impacts backward compatibility...

hamishwillee, 2022-02-24:
@peterbarker Did this go anywhere/should this be kept open or closed?

If the information should go somewhere, then would this affect the design of the general component information - https://mavlink.io/en/messages/development.html#COMPONENT_INFORMATION_BASIC ? That's still in development.xml, so if we need to change it, now would be a good time.

peterbarker, 2022-02-24:
> @peterbarker Did this go anywhere/should this be kept open or closed?
>
> If the information should go somewhere, then would this affect the design of the general component information - [...]

ArduPilot's populating the top 16 bits of that field with the board ID, if that's what you're asking...