mavlink/mavlink #1275: MAVLinkFTP: Require support of compressed (text) files

olliw42, 2019-12-02:
The MAVLink micro services involve a number of different files, which may be stored in the vehicle so that they can be fetched with MAVLinkFTP.

However, they can be very verbose, e.g. when they are text based, and thus consume much memory on the vehicle. For instance, the Camera micro service involves a camera definition XML file, COMPONENT_INFORMATION involves a component definition XML file.

The proposal is to specify at least one compressed format, which must be supported by any MAVLinkFTP client, such as GCSes. I would propose this to be .zip.

For instance, if it is .zip, a camera definition file might be stored on the vehicle as cameraxyz_def.xml.zip and the MAVlinkFTP client would automatically unzip it into cameraxyz_def.xml before processing.

To give you some numbers from my own "experience":
* The STorM32 has currently 3 camera and 2 component definition xml files stored which together have a size of ca 250 kB, and which would compresses to slightly less than 20 kB with .zip. (I currently use my own compression scheme for storing them since the MCU's memory is only 256kB, but it would be much easier if it just could store the zips)
* I've tentatively created a component definition xml file for ArduPilot, and it has a size of 710 kB, but would compress to 64 kB.

Note that some of the files one typically would NOT want to store on a on-board SD card (if the component has any) since it wouldn't then be easily possible to upgrade these files with the "normal" firmware upgrade method of flashing the firmware to the MCU.

hamishwillee, 2019-12-02:
Sounds like a good idea to me. Generally the gimbal or whatever will have pre-compressed, and it doesn't seem particularly costly for a GCS to have a library to decompress. @julianoes @dogmaphobic Any thoughts on this?

julianoes, 2019-12-03:
> cameraxyz_def.xml.zip and the MAVlinkFTP client would automatically unzip it into cameraxyz_def.xml 

I think that makes sense.

hamishwillee, 2019-12-09:
I very much like the end goal, but the FTP protocol itself feels like it might not be the right place. Mostly because it is a generic file transfer protocol and adding specific behaviour for zip file handling and extraction is odd. 

The "right" place feels like the protocol that uses the service - e.g. camera and component information.

I'm not sure how to progress this.

olliw42, 2019-12-10:
I actually had the same consideration and initially attached it to the micro services in my mind. 

However, the ftp service is potentially used by several other services, micro or not (like camera and component-information), so in a code one would put that handling in a place wrapping the mavtfp call, and not in several code places attached to the respective services. That's what made me attach it to MAVFTP.

On the other hand, one could argue that this zip handling should be available also for other URLs than only mavftp://, i.e. also http://. For http it's however not really crucial, it's more a nice to have, but anyway, one could also require this by GCSes. Which would make it an additional code piece called whenever a URL is handled ...

... which would suggest that it should be added to anywhere there a xml file (or a text-based file in general) is mentioned, like http:// and mavft:// are mentioned ...

for cam_definition in CAMERA_INFORMATION e.g. 

"cam_definition_uri | char[140] |   |   | Camera  definition URI (if any, otherwise only basic functions will be  available). HTTP- (http://) and MAVLink FTP- (mavlinkftp://) formatted  URIs are allowed, and the file can be also compressed in .zip format (and any GCS that implements the Camera Protocol must support these options)."

and similar in COMPONENT_INFORMATION, and so on ...

a bit verbose maybe.

just for though

dogmaphobic, 2019-12-10:
> so in a code one would put that handling in a place wrapping the mavtfp call, and not in several code places attached to the respective services.

I tend to agree. The original FTP commands are older than most people here. Adding something to _set_ compression enabled/disabled would make sense. It deviates from the original FTP commands but it does it by extending it rather than changing.

hamishwillee, 2019-12-11:
I get the point (and did think about it). 

I'm just not convinced that FTP is the right layer. Mostly because as already noted, we'd quite like this to work for HTTP too. 

This feels like a library that systems share, that in turn talks to FTP or whatever.  I don't see this library as part of MAVLink per-se. It would just be words like the ones Ollie suggested for the definition URL. 

> Adding something to set compression enabled/disabled would make sense. It deviates from the original FTP commands but it does it by extending it rather than changing.

Yes but in an FTP context, I'd assume compression enabled meant "compress/decompress the existing file for transmission on the channel". What we're talking about here is "if the file is compressed as zip then decompress it on the other end". Just seems weird.

dogmaphobic, 2019-12-11:
Oh, I was thinking compression at the protocol level, like when you enable compression with ssh. If a client doesn't handle a gzip stream, it won't ask for it. I'm not sure that makes a whole lot of sense for HTTP. The idea is that the original FTP client in QGC was awfully broken and useless any way. Something new will have to be written. We could as well add this while at it.

hamishwillee, 2019-12-11:
@dogmaphobic Yes, that makes sense. 

This idea is to store the camera definition as a zip and auto-extract it on the GCS at point of opening; so about saving space on the drone/gimbal/camera end.

julianoes, 2019-12-11:
@dogmaphobic I think this is not necessarily about saving during the transmission but storage of the xml files on a microprocessor. I agree though that we could add that to MAVLink FTP in general as well.

@olliw42 to make this easy considering dependencies the easiest, as @dogmaphobic suggests, is probably just to use gzip (and not actual zip archives).

I gave this a try by just using libz (already a dependency of QGC and MAVSDK) with an example camera definition xml file:
https://gist.github.com/julianoes/c8cfff19b4df49fbd7d3a217dcb6fc5f#file-reame-md

olliw42, 2019-12-11:
many thx, sir.

I need to digest properly before I can respond. Zip is by far more common on Win, and I'm on Win LOL

julianoes, 2019-12-11:
7zip can do it:
[image]

olliw42, 2019-12-11:
thx @julianoes 

I looked at it, and it is indeed not difficult to do gzip-ing. I found several easy ways. 7z is certainly one common. For the few test cases I have, the compression rates are essentially identical, so gzip doesn't win a lot here but also doesn't lose. So, altoghether, gzip looks like a very good suggestion to me.

One thought however. I think one should not make such decisions totally dependent on the likings of one particular implementation. And I wonder if there couldn't be a potential future advantage/benefit/gain of zip's folder structure: It would allow us to have several .xml files and bundle them into one .zip file, which the GCS would then extract and understand that it should combine these xmls into one joined xml structure in code. Being able to bundle different xmls may help maintaining of the xmls. For instance, let's assume that in some near or distant future the xml may contain several top entries such as <parameters>...</parameters>, <flightmodes>...</flightmodes>, <missions>...</missions>, and so on. Putting them all into one xml would result in a big thing, while keeping them in three would make it easier. As mentioned earlier, a parameter xml for ArduPilot is of size of 710 kB, which isn't great to look at.

I just wanted to throw in that thought, so that we don't lose a potential opportunity.  

If you guys decide aginst it however I will happily agree and you should then also decide to go with gzip.

:)

julianoes, 2019-12-12:
That's an interesting idea with the multiple files. However, there is also the versioning for the respective files and docs on how the xml should be structured. Having multiple files will just make this even more complicated, so I'm not sure if that's really required.

olliw42, 2019-12-16:
@julianoes 

I've stored now the camera xml files also in gzip format in the STorM32 controller, for downloading per MAVFtp. Works great, but in the course of that I "discovered" the following, which I'm not sure I can make sense of:

I compressed into .gz using two procedures:
(A) using 7zip and the menu as you showed it. The .gz file size is then 2049 bytes.
(B) Using this python code (where xml_content holds the xml-file content)
```
import gzip
import io
with gzip.open(filename+'.xml.gz', 'wb', compresslevel=9) as output:
     with io.TextIOWrapper(output, encoding='utf-8') as enc:
            enc.write(xml_content)
```
The .gz file size is then 2121 bytes. (since I autogenerate the xml's using a py script, it's convenient to also do the zipping in the script)

Now, then I unpack the .gz file with 7zip I get a FOLDER which contains the xml file, and this is so in BOTH cases!

The point of this is that from what you wrote in the above I got the impression that then I unpack a .gz I would get the file immediately without any folder, as gzip doesn't have any folder structure. And because that's not how I packed it. Secondly, it's not clear to me what complications this means for handling inside QGC. And thirdly, if .gz also has a folder structure one could store two and more files into it, and my last post wouldn't apply to gzip.

??

olliw42, 2019-12-16:
btw, the added benefit of supporting zipped files is the much faster download with MAVFtp ... in the case of my files ca. 10x faster since the compression rate is ca 10x ... e.g., the download of the ca. 110kB uncompressed comp_info definition file takes 20-25 seconds, so a factor 10 is a factor :):):)

julianoes, 2019-12-17:
> Now, then I unpack the .gz file with 7zip I get a FOLDER which contains the xml file, and this is so in BOTH cases!

I wonder if it's just the 7zip unpacking that automatically creates a folder or if it's actually packed that way.

olliw42, 2019-12-17:
the folder has exactly the same name as the file, i.e., if the file is XYZ.xml I'll get a folder XYZ.xml with a file in it named XYZ.xml. Also, that file name is stored in the .gz file as I can see when I open the .gz with an hex editor, but there is only that one name and not two names for folder and file. This maybe (!?!?!?!?!) may suggest it indeed doesn't store a folder structure but that this is just what 7zip does.

The gzip py I used is said on the web is using zlib. You were saying that QGC is using libz but I find a folder zlib in the QGC libs folder, so this was probably just a misspelling. So whatever I created is likely what QGC can use.

I think one also should double check with MissionPlanner, not only QGC. I find it has a folder 7zip and so maybe (!?!?!?!) also already be able to handle .gz files. Could be a bit of a stretch however.

I'm also not sure if one should decide that based on already linked libs, I mean, how difficult can it be to use a lib which does .zip.

The advantages of supporting compressed xml are however real and obvious now (easier to store on host, much faster download), and something one really really wants :D:D:D.

julianoes, 2019-12-17:
I think the libz dependency is not a problem for QGC and should not be for MissionPlanner either.

olliw42, 2019-12-17:
the question would be if a .zip capable lib dependency would be a problem for either

question: are you talking about zlib or libz ? Both seem to be existing.

julianoes, 2019-12-17:
> the question would be if a .zip capable lib dependency would be a problem for either

Aha, I think that's harder but I don't know if it would be a problem.

> question: are you talking about zlib or libz ? Both seem to be existing.

I think I'm talking about zlib.

olliw42, 2019-12-17:
:)

julianoes, 2020-01-08:
@olliw42 let's use `.gz` for gzip compressed stream for now and then we could still add `.zip` to support an archive with folder/multiple files later. Ok?

olliw42, 2020-01-08:
perfectly fine with me, gz is good :)

hamishwillee, 2020-01-08:
Good plan. What are next steps? "Someone" to implement .gz automatic decompression on camera/component definition files in QGC?

olliw42, 2020-01-09:
I think the next step should be to write it into stone ... äh into the docs ...

hamishwillee, 2020-01-12:
Writing it into docs is a bit pointless unless we know "someone" is committed to updating QGC .

olliw42, 2020-01-12:
I respectfully disagree

1. I generally can't see any correlation of MAVLink with a particular GCS or flight stack or component or whatever
2. MAVFtp is also in the specs but not usable in QGC, and this argument applies to several other topics, e.g. that a GCS must support MAVFtp urls for xml's, and so on ...
3. It is much less pointless to have something in the spec so that GCSes, flight stacks, components can follow up than to put something in some specific GCS, flight stack, componet which is NOT in the specs ...
4. you are putting the chicken egg problem into place, and not resolving it :)

this btw may also give you hints for what the stumbling stone is, namely MAVFtp in QGC ... 

IMHO :)

(and as regards the "someone" I unfortunately have to tell that one potential "someone" concluded that momentarily it doesn't make sense to produce further PRs when old & simple PRs are still flying around)(not your fault, THX for your pings !!, but it's the situation) :)

hamishwillee, 2020-01-13:
@olliw42 What I am trying to avoid is "great ideas" making their way into the spec but not being really tested and hardened. The way to avoid that is to have a stakeholder that is committed to the idea of pushing an implementation through. 

So while you're right that I'm not breaking the chicken/egg problem, I'd rather that than fill the spec with what ends up being cruft. And yes, I understand why you wouldn't want to commit to this yet. I wouldn't if I was you either.

Let me try unlock this.

@dogmaphobic We'd like to make it mandatory for a GCS that supports the camera protocol to automatically handle camera definition files that are stored as .gz compressed files (as recognised by the .gz file extension). We have checked (see thread) that this can be done without adding additional dependencies to QGC.
If I update the docs, can you commit to doing the QGC side code sometime in the next 6 months?

olliw42, 2020-01-13:
I hear you
there is no WIP thing for this sort of stuff, like for COMPONENT_INFORMATION and alike ... that's true

as regards QGC, the main issue as mentioned is IMHO the dependency on mavftp ... not sure if anyone wants to commit to this really ... both dogmaphobic and donalakeflyer made some tough comments on the existing QGC implementation if I remember correctly, gave me the impression that it's a bit of a bigger effort than just doing a qUncompress ...

what's up with the SDK folks, they don't do camera and mavftp microservices ?!?!?! :D:D:D

hamishwillee, 2020-01-13:
Good point re SDK. @julianoes Would you be able to add this support to the SDK (or arrange it?). What are your thoughts on getting this in QGC? Having a real implementation is enough for us to decide it is "sane" but frustrating for end users if we put this in but then QGC does not.

olliw42, 2020-01-13:
so, I thought just for sake of a proof of principle I could try to ungzip a camera definition file which is called via a http url.

I first tried Qt's qUncompress, which however failed. The Qt docs say one needs to prepend the data with the expected file size, but I have no idea how or where one could get info from. https://doc.qt.io/qt-5/qbytearray.html#qUncompress. So no idea how this possibly could work.

I then scanned the web for zlib things and found //https://stackoverflow.com/questions/2690328/qt-quncompress-gzip-data. However, when I add this code QtCreator complains and doing #include <zlib.h> does not help as it QtCreator says it's not known. So, it doesn't exactly look like QGC already has support for zlib.h.

Anyway, the exercise demonstrated to me that this is beyond my competences, and I should leave it to the experts. So much concerning one "someone".

:)

julianoes, 2020-01-17:
I quickly tried out `qUncompress` and could not figure it out. I kept getting `qUncompress: Z_DATA_ERROR: Input data is corrupted` even though I did set the header length.

On Linux for the QGC build all I have to do is:
```
#include <zlib.h>
```

And probably for linking later something like `LIBS += -lz` will be necessary in qgroundcontrol.pro.

Now for Windows I don't know, I only found this:
https://stackoverflow.com/questions/44522073/z-lib-problems-while-porting-qt-creator-projet-to-windows#answer-44614728

hamishwillee, 2020-04-27:
Closing. 

A great suggestion, but not a requirement on MAVFTP; rather the requirement belongs in specific protocols that specify the URL. This is already specified in the docs, and I've created a fix for the common.xml messages here: https://github.com/mavlink/mavlink/pull/1355

Let's hope the GCS will update soon.

closed 2020-04-27 by hamishwillee