mavlink/mavlink #1280: Pl add discussion about COMPONENT_INFORMATION to dev call agenda, if possible

olliw42, 2019-12-09:
In the next dev call, this Wed, I would like - if possible - discuss a bit more thoroughly how one could progress with COMPONENT_INFORMATION.

The COMPONENT_INFORMATION message has been merged to mavlink as WIP, but - at least as much as I can see - it's currently totally unclear what the best ways would be to bring it to fruition.

Notes:

As regards putting topics on the dev call agenda, the wiki says "_Add topics that you'd like to discuss and issues you'd like to unblock._" but I couldn't find any further explanation of how one would do that. So, I miss-use this issue here, sorry for that.

I'm actually not sure the topic belongs into the mavlink dev call, in light of rroche's recent "Meet the Volunteers" thread one could think that the "Community Management" level would be more appropriate but I couldn't find more about it, hence I'm here. Also sorry for this. 

Putting up this issue was the best effort I could see. :)

hamishwillee, 2019-12-09:
Works for me. I added the label and will add to agenda.

olliw42, 2019-12-10:
thanks so much. Very cool.

just to avoid any mistakes on my side: 

The meeting starts at 9:00 AM CET = 9:00 AM MET = 9:00 Zürich time = 9:00 Berlin time, right?

(
I ask since here https://mavlink.io/en/about/support.html it says 9:00 AM CET: 

[image]

but in https://www.dronecode.org/calendar/ 

[image]

it shows it at 8:00 - 9:00, and explicitely says "The calendar below shows events for the Zurich timezone." at the top and "Terminanzeige in der Zeitzone: Mitteleuropäische Zeit - Zürich" at the bottom, so not UTC but 8:00 AM MET = 8:00 AM CET since CET = MET 

!?!?!?!?
)

hamishwillee, 2019-12-10:
6pm tomorrow for me in Melbourne. I.e. tomorrow but an hour before now. Looking at GMT it is now 8am, so 7am GMT tomorrow.

I have nothing to do with that calendar. @mrpollo :-)

olliw42, 2019-12-10:
LOL ... so I have to do some math ...

GMT is UTC, and MET = CET = UTC+1.

so 7:00 AM GMT would mean 8:00 AM MET/CET/Zürich/Berlin time !?!?!?!

so, the drone code calender would be right, and the mavlink-doc https://mavlink.io/en/about/support.html entry would be off by 1 hour ??

would you mind double checking if the mavlink-doc entry is correct/wrong?

julianoes, 2019-12-10:
It starts at 8am during winter and 9am in summer :smile:. So tomorrow 8:00.

olliw42, 2019-12-10:
many thx, sir.

I'm glad I asked

The byproduct is that the info in the mavlink docs isn't correct, since it's currently winter and we have MET = UTC+1 = GMT+1 = CET, and it currently should thus say 8:00 AM CET.

See you tomorrow.

olliw42, 2019-12-11:
THANKS a lot for the very helpful and open discussion today. Much appreciated.

:)

(and close)

closed 2019-12-11 by olliw42