mavlink/mavlink #1285: Proposal: Deprecate all non MISSION_*_INT usage with a breaking change

DonLakeFlyer, 2019-12-24:
As it currently stands there is a hole in the mavlink spec with respect to _INT usage. There is a capability bit on the vehicle side. But there is no clear way to know whether the GCS side will support uploads of MISSION_ITEM_INT items. The only sideways sort of way to do that would be for the vehicle to remember if the GCS did a MISSION_REQUEST_INT. To both deal with that and remove complexity from the mavlink spec I propose a breaking change to only allow support for the _INT variants of the protocol.

* This would mark MISSION_ITEM, MISSION_REQUEST and MAV_PROTOCOL_CAPABILITY_MISSION_INT as deprecated.
* A gcs is required to support the _INT based protocol sequence.
* If a gcs receives old non _INT requests it always respond with _INT responses.This provides an interim transition path to the future while these messages still exist. I've tested this in QGC and it works fine against PX4 and ArduPilot.

This reflects the reality of PX4/ArduPilot/QGC/MP of today which supports the _INT based protocol. I don't see any reason to maintain the non _INT protocol moving forward. So gets reduce complexity and get rid of it.

Related to #8122

LorenzMeier, 2019-12-24:
That makes total sense. We need to tighten up the overall protocol.

hamishwillee, 2020-01-06:
@auturgy Any comment?

hamishwillee, 2020-01-09:
@auturgy My understanding of your comments in dev call was that:
- The problem was caused by ArduPilot not implementing the spec correctly. QGC works around this problem already for ardupilot. If people other implement the protocol correctly there will be no problem. So there isn't much value in doing this.
- That @WickedShell uses the float versions.

Is that correct? 

To me the value (to me) in this is simplifying the work for future implementations. Essentially a docs change that removes the old non-INT path. I believe I understand your objection as "not best use of time" rather than "bad idea". Can you please comment directly, because if it is the former, then I'd like to do the work.

WickedShell, 2020-01-09:
I should clarify that, I don't use the float versions, but it can be argued that they actually better represent some situations such as local NED much better then the int does. (Not that most systems have sensors that can read that accurately)

DonLakeFlyer, 2020-01-09:
> To me the value (to me) in this is simplifying the work for future implementations.

Yup, that is why I proposed it. Simpler is better.

auturgy, 2020-02-05:
Lets move to a PR and have a look at the impact - I think it'll be ok, as the lost local NED precision isn't useable anyway

amilcarlucas, 2020-04-02:
Isn't the only place that uses both _INT and NED variant the old Gimbal messsage?

hamishwillee, 2020-06-02:
To progress this, PR created. See https://github.com/mavlink/mavlink/pull/1389

PR #1389 (2020-06-25): Deprecate all non MISSION_*_INT usage
Fixes #1285

To progress this, have created PR. Docs will have to be updated too. Original description below

----

As it currently stands there is a hole in the mavlink spec with respect to _INT usage. There is a capability bit on the vehicle side. But there is no clear way to know whether the GCS side will support uploads of MISSION_ITEM_INT items. The only sideways sort of way to do that would be for the vehicle to remember if the GCS did a MISSION_REQUEST_INT. To both deal with that and remove complexity from the mavlink spec I propose a breaking chang [... cut]
closed 2020-06-25 by hamishwillee