mavlink/mavlink #1286: Proposal: Support for optional values in MISSION_ITEM_INT.[x|y]

DonLakeFlyer, 2019-12-24:
As the spec currently stands the x and y values of MISSION_ITEM_INT have no mechanism to indicate an optional value. With MISSION_ITEM this is commonly done by setting the value to a NaN. Without this it limits the parameter usage for mission items which are not location based and use x/y for other things than a lat/lon. Even given that I would contend that the ability to support am optional value in lat/lon can be useful as well. For example a TAKEOFF item that has optional location with it. 

We can accomplish this by spec'ing a max int value in x/y indicates the lack of an optional value.

The reason to spec that at the MISSION_ITEM_INT level as opposed to each individual message is to give a gcs enough information to handle this internally. At least for QGC it stores mission items with doubles internally and converts them as they come on. In order to do that correctly it needs to know that for any MISSION_ITEM_INT a max int value in x/y means optional. And it should convert it to a NaN on the way in.

LorenzMeier, 2019-12-24:
How about INT32_MAX?

DonLakeFlyer, 2019-12-25:
> How about INT32_MAX?

Yup

hamishwillee, 2020-01-09:
@auturgy Any thoughts on this one?

hamishwillee, 2020-02-06:
@DonLakeFlyer 
FMI, can you give me an example of an optional parameter in a command? 
i.e. For an update _message_ I get this - we use NAN to indicate perhaps that this sensor does not output a particular bit of information. But I don't get it for commands. 

Do either of these work better for you ... ?

A modification of the params:
> `<field type="float" name="param4"> PARAM4, see MAV_CMD enum.` **NaN indicates an optional value.**`</field>`
>  `<field type="int32_t" name="x">PARAM5 / local: x position in meters * 1e4, global: latitude in degrees * 10^7.` **INT32_MAX indicates an optional value.**`</field>`

Or in the description

>Message encoding a mission item. This message is emitted to announce the presence of a mission item and to set a mission item on the system. The mission item can be either in x, y, z meters (type: LOCAL) or x:lat, y:lon, z:altitude. Local frame is Z-down, right handed (NED), global frame is Z-up, right handed (ENU). **NaN or INT32_MAX may be used to indicate optional parameters in float and integer params (respectively).**, See also https://mavlink.io/en/services/mission.html.

DonLakeFlyer, 2020-02-06:
> FMI, can you give me an example of an optional parameter in a command?

Look at MAV_CMD_DO_ORBIT. That has NaN support in param 5/6/7

hamishwillee, 2020-02-14:
Thanks. See #1303 for proposed wording. You need to review :-)

PR #1303 (2020-02-26): common: NAN and INT32_MAX for default param values
Fixes #1286 

This attempts to address #1286 by clarifying that `NaN` and `INT32_MAX` may be used in MAV_CMD params to indicate optional/default values - e.g. a taking yaw from vehicle position rather than the specified value. 

I cover `MISSION_ITEM`, `MISSION_ITEM_INT` and `MAV_CMD`.

I don't cover the implication that if you explicitly specify INT32_MAX for param 5 (say) you are mandating the command is only use in an integer-based message. 

Do we need to update COMMAND and COMMAND_INT?

@auturgy FYI
closed 2020-02-26 by hamishwillee