mavlink/mavlink #1287: mavlink.io website not accessible

mtbsteve, 2020-01-03:
Seems like the https://mavlink.io/en/ site is down. Thanks for fixing.

mhkabir, 2020-01-03:
Works for me

mtbsteve, 2020-01-03:
Its certainly not working. Tried it from different internet access points.
[https://1drv.ms/u/s!AnKeW8KMoCcyw0tQgy2ajwtcjVPA?e=Fmfahu](url)

LorenzMeier, 2020-01-03:
Works also from here. I'm closing this for now assuming it is a local issue.

LorenzMeier, 2020-01-03:
Thank you for taking the time to report it!

closed 2020-01-03 by LorenzMeier

vooon, 2020-01-03:
Works for me too.

mtbsteve, 2020-01-03:
@LorenzMeier It is not accessible from my environment:
- iPhone - Chrome and Safari no access by LTE/German Telekom nor via wifi
- PC - Chrome, Internet explorer neither via wifi nor LAN
- Android Tablet -Chrome neither LTE nor wifi
PS: I live in Germany... and the rest of the internet works well...

hamishwillee, 2020-01-05:
Works from Australia.
This is hosted on github pages, so should work anywhere github works.

mtbsteve, 2020-01-05:
ok I am probably the only person on earth who cannot access the mavlink.io website. 
Anyway at least I can access https://github.com/mavlink/mavlink-devguide
and read the documentation from there so I am good for now.  
Still it would be nice if someone could find out what makes this site behavior  different from the rest of the internet.

hamishwillee, 2020-01-05:
I hope so :-). FMI can you see http://docs.px4.io/master/en/index.html ?

mtbsteve, 2020-01-05:
Yes I can confirm that http://docs.px4.io/master/en/index.html
works perfectly well for me.

hamishwillee, 2020-01-05:
That's odd. They live on the same infrastructure. I don't know how to help debug this for you.

devurandom, 2020-02-15:
#1289 is a duplicate.

Reason for this is that GitBook changed their product and DNS scheme:

> In short, CNAME to hosting.gitbook.com.
>
> If you are a v1 user, you may have been used to using www.gitbooks.io , which only works for legacy.gitbook.com and will not work for new users.
>
> Note: GitBook legacy will soon be deprecated. Contact us to help you migrate your content.

mtbsteve, 2020-02-15:
Thanks for the clarification @devurandom !
@LorenzMeier @hamishwillee would be great if you could migrate the mavlink.io content accordingly.

hamishwillee, 2020-02-17:
@dagar ^^^^