mavlink/mavlink #1288: CAMERA_INFORMATION ( #259 ) and COMPONENT_INFORMATION ( #395 ) version description is wrong

reedbn, 2020-01-07:
The CAMERA_INFORMATION firmware_version field description is wrong. Currently, [common.xml line 5706](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L5706) reads:

`<field type="uint32_t" name="firmware_version">Version of the camera firmware (v &lt;&lt; 24 &amp; 0xff = Dev, v &lt;&lt; 16 &amp; 0xff = Patch, v &lt;&lt; 8 &amp; 0xff = Minor, v &amp; 0xff = Major)</field>`

Which renders as

> Version of the camera firmware (v << 24 & 0xff = Dev, v << 16 & 0xff = Patch, v << 8 & 0xff = Minor, v & 0xff = Major)

However, consider the following:
```
0xAABBCCDD << 24 = 0xDD000000 & 0xFF = 0x00
0xAABBCCDD << 16 = 0xCCDD0000 & 0xFF = 0x00
0xAABBCCDD <<  8 = 0xBBCCDD00 & 0xFF = 0x00
0xAABBCCDD <<  0 = 0xAABBCCDD & 0xFF = 0xDD
```

This suggests that 3 of the 4 parameters are ALWAYS decoded as 0, which makes no sense. Prioritizing the bit-and doesn't do any favors either. I also discovered the same issue for COMPONENT_INFORMATION ( #395 ), where both firmware_version and hardware_version have this undoubtedly incorrect decoding suggestion.

I suspect the `&lt;` were supposed to be `&gt;`, such that 0xAABBCCDD would be decoded as Major DD, Minor CC, Patch BB, Dev AA, but can't find any specific references or examples to confirm this. Please update these three incorrect descriptions to indicate the actual desired encoding/decoding.

olliw42, 2020-01-08:
I noted this too and if this were to be interpreted as actual code then IMHO you were right, but I don't think this is meant to be actual code but pseudo code to hint towards what is meant

I would translate it into actual code as 
(Dev  & 0xff) << 24 + (Patch & 0xff) << 16 + (Minor & 0xff) << 8 + (Major & 0xff)

there is also AUTOPILOT_VERSION with 32-bit wide version fields, but without any explicit statements, but it be affected too

hamishwillee, 2020-04-26:
Either way it is confusing. Can you guys propose a wording or PR that would fix this?

reedbn, 2020-04-27:
@hamishwillee revising the comment to read similar to @olliw42's code in their January 8 comment would suffice, since the code would then be plausible (if not correct). The full revised comment would read something like:

`Version of the camera firmware, encoded as: (Dev & 0xff) << 24 | (Patch & 0xff) << 16 | (Minor & 0xff) << 8 | (Major & 0xff)`

hamishwillee, 2020-04-27:
@dogmaphobic You happy for me to modify  [CAMERA_INFORMATION.firmware_version](https://mavlink.io/en/messages/common.html#CAMERA_INFORMATION) to `Version of the camera firmware, encoded as: (Dev & 0xff) << 24 | (Patch & 0xff) << 16 | (Minor & 0xff) << 8 | (Major & 0xff)`

It is more clear than the current version.

(Thanks @reedbn )

dogmaphobic, 2020-04-30:
More clear would depend on the point of view. Were I to show either one to my wife, she would think the cat was messing with the keyboard in either case :P 

Yup, it looks fine to me.

hamishwillee, 2020-05-01:
@dogmaphobic Sure. But I'm designated idiot here, and I think this is just that little bit less ambigous.

@reedbn Can you PR this change into the messages? Rememebering that the XML needs to use lt and gt markup, not the carets.

reedbn, 2020-05-02:
Turned out there were 4 instances, so I replaced all 4 in the pull request.

hamishwillee, 2020-05-03:
Thanks very much.

hamishwillee, 2020-05-12:
Fixed. Thanks!

closed 2020-05-12 by hamishwillee