mavlink/mavlink #1289: mavlink.io website offline

MaxBec, 2020-01-13:
Hey guys,

the mavlink.io website seems to be offline. Can you confirm that?
[image]

hamishwillee, 2020-01-13:
It is working for me now, from Australia.

dayjaby, 2020-01-27:
Still not working from Germany.

hamishwillee, 2020-01-28:
I can access the site through a German proxy server:

[image]

As I understand the Internet, that pretty much means that the problem is local to specific users, and not something we can deal with. I'm therefore closing - though happy for you to reopen if you can think of anything that we can do to further debug or help with this.

hamishwillee, 2020-01-28:
And if needed, the content can be found here: https://github.com/mavlink/mavlink-devguide/tree/master/en

you can build locally by adapting the instructions here: http://dev.px4.io/master/en/contribute/docs.html#gitbook-documentation-toolchain - at least on Ubuntu.

closed 2020-01-28 by hamishwillee

devurandom, 2020-02-15:
```
❯ drill mavlink.io
;; ->>HEADER<<- opcode: QUERY, rcode: SERVFAIL, id: 43793
;; flags: qr rd ra ; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 0 
;; QUESTION SECTION:
;; mavlink.io.  IN      A

;; ANSWER SECTION:
mavlink.io.     109     IN      CNAME   www.gitbooks.io.
www.gitbooks.io.        3409    IN      CNAME   cdn.gitbook.com.
```