mavlink/mavlink #1302: Feature:  Offboard Altitude + Attitude + Yaw-rate

Stifael, 2020-02-06:
Note: This issue is a feature discussion

### Goal
Find a solution that is future proof and therefore no hack solution is required at this stage.

### Feature Definition
Being able to control the drone via offboard (companion computer, desktop computer...) through `attitude`-setpoint, `altitude`-setpoints and `yaw-rate` setpoints.

The `attitude`-setpoint can either be `quaternion`, `roll/pitch/yaw` or just `roll/pitch` or all of them. In principle only roll/pitch is required because heading will be controlled through yaw-rate-setpoints.

Altitude control should support altitude-commands relative to takeoff position and velocity-setpoints in z-direction. 

### Possible options
I already discussed possible options, but we did not really find a consent. 

#### Option A: Use two mavlink messages
One idea was to use two existing messages such as [SET_ATTITUDE_TARGET](https://mavlink.io/en/messages/common.html#SET_ATTITUDE_TARGET) and [SET_POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED). 

*Advantage*: No need to implement a new message

*Disadvantage*: These messages currently have no dependencies. This means that the autopilot, who has to process the messages for validity, cannot decide if certain fields are missing or not. This means that quite some changes has to be done on the autopilot side against the current implementation where each offboard-mavlink-message contains the full information. 

#### Option B: Create a new mavlink message
The new message would contain all the information required to successfully run attitude-altitude-yaw-rate control. This option aligns with the current offboard-mavlink-implementation, where each mavlink-message contains all the information required to successfully execute the setpoints. 

*Advantage*: Easy to implement
*Disadvantage*: requires a new message and might not cover other combination of setpoints (are there any which actually would make sense?)

#### Option C: virtual sticks
`Virtual sticks` is similar to what DJI has (please don't be hang up on the name `Virtual sticks`). The idea is that the [MANUAL_CONTROL](https://mavlink.io/en/messages/common.html#MANUAL_CONTROL) message can be mimicked via an offboard-message. This means that depending on the desired-flight mode, x,y,z and r have different meanings, exactly as it is done with the existing `MANUAL_CONTROL` message. 

*Advantage*: Depending on the autopilot, this option has the benefit that all the manual-control-modes can be reused without major changes to the autopilot code-base. Furthermore, continuous integration could finally also include all the manual-control-modes, because the control-pipeline should be the same

*Disadvantage*: please give me feedback

hamishwillee, 2020-02-06:
@Stifael Is there anyone in the community you'd like to bring into the discussion? It might be useful to bring this to the next MAVLink meeting - would you be able to attend? @auturgy FYI.

With respect to option A vs B we tend to suggest that if multiple existing messages can be used that is desirable. Reasons to start thinking about a single message are things like:
- Impact of timing/streaming rates of information from separate messages on the solution working.
- Ability to change existing messages without breaking existing users.

Option C is interesting idea :-)

elpraesidente, 2020-02-07:
Yes, I can attend.

Stifael, 2020-02-07:
yes, I can attend.

auturgy, 2020-02-19:
An option not addressed is creating extensions to an existing message.  That might permit the adaption of SET_ATTITUDE_TARGET with the extra fields you need

hamishwillee, 2020-02-19:
As discussed, best direction we can give is to try a PR with extensions needed added to existing message. We can then discuss that. If that doesn't meet general approval then we'll work towards alternatives.

julianoes, 2020-03-04:
I like @auturgy's suggestion.