mavlink/mavlink #1304: ROI supported message

dakejahl, 2020-02-15:
Currently there is no way for an autopilot to let a ground station know if ROI is supported or not. This would be useful to know so that a ground station could enable/disable UI elements relating to ROI.

I am thinking this would be implemented as a question the GCS asks the autopilot when it connects. 

Thoughts? Suggestions?

hamishwillee, 2020-02-17:
@julianoes Does the new gimbal API help with this? My guess is that it could tell you if ROI is supported, but not necessarily that it is "not supported" (i.e. in theory an MC without a gimbal could set an ROI).

My thinking is that this is useful. Can we justify a MAV_PROTOCOL_CAPABILITY?

julianoes, 2020-02-19:
@dakejahl could question. We discussed it in the call and found two cases:

1. Case without gimbal for multicopter:
Here it's not clear. The only thing we can think of is to try the command and then the autopilot would return `RESULT_UNSUPPORTED`.
We could also think of a general mechanism (like brain-stormed in the [mavlink microservice proposal](https://docs.google.com/document/d/1zK_JcJbhtbRA3W0MgnXSNWMUxOcBu6vXocrjZVMXki4)) where you could query support of a message and/or command in general to see if it is supported, since it's a general problem and not just ROI.

2. Case with gimbal:
For setups with gimbals I will add a capability flag to the gimbal manager indicating whether the gimbal supports ROI to the [new gimbal messages](https://github.com/mavlink/mavlink/pull/1174).

My crappy suggestion would be to assume that ROI is possible if it's a multicopter or if the new gimbal protocol is used and telling you that it ROI is supported.

hamishwillee, 2020-02-25:
Thanks @julianoes . 

@dakejahl FYI ... just to clarify, the general mechanism (1) would be some way to query whether any particular command or message is supported. E.g. some way of evaluating if a command would succeed or fail, without the side effect of actually executing the command. I think that is something we may well need, but would not wait on for this discussion.

What command is used for setting an ROI out of missions? MAV_CMD_DO_SET_ROI?

hamishwillee, 2020-03-04:
@dakejahl Comments?

dakejahl, 2020-03-07:
> where you could query support of a message and/or command in general to see if it is supported, since it's a general problem and not just ROI.

I think this is the right way to do it, but obviously that is a whole topic in itself. I'm in no rush for this, as custom builds have a mechanism to enable/disable this in QGC and that is my use case.

hamishwillee, 2020-04-01:
FYI we are looking at case 1 above: https://github.com/mavlink/mavlink/issues/1304#issuecomment-588114751 - ie ability to test if particular command is supported or not.

hamishwillee, 2020-04-29:
@dakejahl This will be solved by the work we're doing in the component information file metadata: https://github.com/mavlink/mavlink/issues/1346

 Essentially you will be able to query for what MAV_CMDs are supported in missions, and separately what MAV_CMDS are supported in diffrent modes.

A GCS will therefore be able to know cannonically if MAV_CMD_DO_SET_ROI is supported in each of those cases. 

It will also be possible via the gimbal API v2 to query for presence of a gimbal.

hamishwillee, 2020-05-12:
I am closing. Yes it is not fixed yet, but this will be done.

hamishwillee, 2020-07-12:
We're still working on support for component metadata. Very much coming soon.

closed 2024-09-22 by dakejahl

hamishwillee, 2024-10-02:
Thanks for closing. We still don't know if any particular command or message is supported - all we can do right now is query and look at the ack, which of course has side-effects if the command is supported.
The component information metadata is now part of the standard and could be used to document support but nothing has been agreed or implemented.

I doubt we'll get broad agreement on a solution.