mavlink/mavlink #1305: Proposal: Prearm check handling

olliw42, 2020-02-17:
Prearm checks are obviously a vital part of a drone system, and I don't think I need to spend words on this. Unfortunately, MAVLink has no generic and satisfying mechanism defined. This has led to what I would call a sort of a mess.

For instance with ArduPilot, currently the prearm status is communicated via STATUSTEXT messages, send out at intervals of 30 secs. This implies that (i) one can find out about the current prearm state only by trying to arm and see if it fails or not, (ii) is informed only every 30 secs if a prearm check still failed, (iii) have to carefully follow and watch the messages in order to know whats up. 

The situation has somewhat improved very recently in that a global prearm flag `MAV_SYS_STATUS_PREARM_CHECK` was added to the `SYS_STATUS` message (https://github.com/mavlink/mavlink/commit/659f27a7598a2f0a146da7ead35ffd4c23486bdd) so that the operator can at least be informed continuously by a GCS. 

Yet I argue that this is only a minor step but by far not what a user wants to see. I argue that what a user wants to see is a list of the enabled prearm checks with those being passed being checked and those which are not yet passed being marked in red.

Furthermore, I argue, that prearm checks are not only relevant for the autopilot but also for other MAVLink components in the system too, specifically I do have a gimbal and a camera in mind. These too should be reported and here too I argue that a global flag is just a minor but not satisfying step in the direction. In fact, I've recently added more sophisticated pream checks to the STorM32 gimbal controller, upon a user request, which I have to say made totally sense to me - and which made me think about the issue resulting in this proposal here LOL.

Furthermore, I'd like to note that there can be cross correlations between the prearm states of the autopilot and those of a component. For instance, a gimbal may have to wait for a certain condition on the autopilot to be fullfilled, before it can go on with it's next initialization step. Examples I have in mind are e.g. that the AHRS is ok or a GPS is ok, so that the gimbal can configure itself accordingly. This sort of things will become much more relevant with the looming "new" gimbal messages. Here too, a more detailed prearm information is required than just a global flag.

In order to resolve these points, at least to a good extend, I'd like to make this proposal:

**A) Data fields**

One specifies 4 bitmasks, each of size uint32 or even better of size uint64 in order to have enough room for future. Two of them are for indicating which prearm checks are enabled, and two of them are for indicating which prearm checks have been passed. Furthemore, and I think that's a key idea, we have two pairs of flags, one pair for what I will call the **"MAVLink standard" prearm checks** and one pair for what I will call the **"custom defined" prearm checks**. 

The MAVLink standard checks are those defined as part of the MAVLink standard, i.e., checks which most likely are of use for any autopilot or component. However, this set of checks certainly will not cover all the checks offered by a specific implementation, and this is what the custom defined checks are for. So, we have these four bitmasks:

- prearm_enabled: bitmask indicating which of the MAVLink defined checks are enabled by the operator/implementation
- prearm_status: bitmask indicating which of the MAVLink defined checks have been passed
- custom_prearm_enabled: bitmask indicating which of the custom defined checks are enabled by the operator/implementation
- custom_prearm_status: bitmask indicating which of the custom defined checks have been passed

**B) Text/info displayed by a GCS**

Since the meaning of the MAVLink defined checks is defined, any GCS knows what text/info to display to the user. What is missing in order to display a fully satisfying prearm page to the user, is the text/info which the GCS should display for the custom defined prearm checks. This, I propose, can be resolved by the `COMPONENT_INFORMATION` mechanism, via which a xml definition file can be queried, and the proposal is to have a xml entry in this file which holds the text/info a GCS should display

With A) and B) a GCS would be able to provide the user with a prearm check page which gives the user all the info she wants in "real time", and the distinction into "MAVLink defined" and "custom defined" should be flexible enough and circumvent "finding consensus" issues like that with the flight modes, mission commands, and so on.

**C) MAVLink message**

One could add these bitmasks to already existing messages, via the extension mechanism, e.g. to the SYS_STATUS message. The advantages would be obvious, no new message. However, I'd like to mention that extending existing MAVLink1 messages i.e. in the 0-255 range apparently can cause dire issues with implementations which are not compiled with them (such an issue currently exists with the COMMAND_ACK extension, ArduPilot simply behaves ill here). Furthermore, SYS_STATUS is already "big" and mostly hosts data which are rarely used, especially not by a component. For that reasons I actually would argue in favor of a new `PREARM_STATUS` message. The message should be emitted at a reasonable interval like 1 sec while the prearm has not yet passed, but should also be emitted regularly but maybe down to e.g. 5 secs when the checks were passed.

Looking forward to hear your responses. I believe that this is the sort of issue which one would want to resolve rather sooner than later, since it brings immediate advantages to the user and because it's so obvious that one in fact may wonder why there hasn't been a spec before.

:)

hamishwillee, 2020-02-19:
@olliw42 Yes we need a much more refined mechanism to understand why something what we're waiting on if we can't arm. You can get quite a lot of information about common reasons not to ARM (e.g. via calibration status checks etc) but there are plenty of possible  blockers.

It has been discussed numerous times, but I'm very happy to take this to the devcall and see if we can get more concrete actions.

olliw42, 2020-02-19:
even though I have zero knowledge about what concepts were discussed before, I would think that a stopper kind of always was that it's difficult to standardize them and find a consensus on which ones to define. I.e., particular implementations just have too diverging views and needs. (I e.g. think that this is what hampers the efforts with the flight modes).

I think there is a specific novelty in this proposal here in that it tries to accommodate this from the beginning. 

It would be great if this would be discussed.

many thx.

julianoes, 2020-02-19:
@olliw42 this was discussed in today's call but I was too slow in following up here.

So, thanks for the proposal, we all agree that this is a weak point and needs fixing in protocol. 

@bkueng tackled this problem more than a year ago and came up with a proposal and some first prototype implementations, see: https://docs.google.com/document/d/18qdDgfML97lItom09MJhngYnFzAm1zFdmlCKG7TaBpg.
Unfortunately, there has not been any progress since, however, it is always on the todo list and has not fallen of the radar.

@olliw42 I would therefore suggest that you have a look at Beat's proposal and see if/how it solves the problem you are addressing and comment on how it could be changed or extended.

One thing I'm not sure about is how it would work for components other than the autopilot. That's certainly one thing to check, although I see that there might be some overlap with things like `COMPONENT_STATUS`, `GIMBAL_STATUS`, `CAMERA_STATUS` which more specifically cater these components and are not necessarily related to arming as much. Of course an autopilot might have an arming flag which checks if the gimbal is available but that would be handled by the autopilot and not the gimbal itself.

olliw42, 2020-02-19:
hey 

thx for having discussed it today, much appreciated

> I would therefore suggest that you have a look at Beat's proposal and see if/how it solves the problem you are addressing and comment on how it could be changed or extended.

I actually did knew about the events proposal, I have seen it before, but I also admit it fell a bit of my mind, so it is good that you did bring this up again.

it however doesn't really change my mind. I see two "issues" with it which make me argue that it probably would be better to have a mechanism for the prearming checks separately:

(1) As I indicated, I think a major problem with all this sort of proposals is that it is difficult to agree on the actually defined states. A key element of my proposal here therefore was to distinguish into "standard" and "custom", and explicitly provide support for "custom". Since one also wants to have proper information text associated with each state, a necessary result is the need for something like what is achieved with the xml file. I can't see any element in the events proposal which would go into that direction. I.e., all possible events need to be standardized. In contrast, in my proposal, each implementation is free to decide on its prearm checks, but yet they can be handled totally generically.

Of course, one could just do long enough enums to hold all desired prearm conditions. For autopilots alone this might be managable to some extend, but I think once also components come into play it's just growing. And it always needs interaction with MAVLink, to get it into the standard, needs interaction with the GCSes, to support and display the latest text, and so on forth. 

I think this was really the genius insight by those who came up with the camera definition file. All this also could have been done by huge enums in a standard. But it doesn't look very managable in practical terms.

(2) I think one should not only distinguish events as done in the "out of scope" and "in scope" parts, but also into "immediate" and "can wait a bit". For instance, a failsafe event is certainly something which should be reported immediately so that actions can be taken immediately. The immediacy also implies reliability, i.e. retransmission and so on. However, prearming states are IMHO of different nature, for them it really doesn't matter if they are coming in 1 or 2 seconds too late. It also would be totally ok to just report them every few seconds, and it would not compromise the system at all if one report would be missed. In short: The proposal mechanism is really a huge overdoing of what is needed for prearm check. For instance, it says "_a sender periodically broadcasts its current sequence number (with a suggested interval of 3 seconds_", so it just could report the prearmcheck state every 3 seconds, WITHOUT need of ANY of all the other clutter. 

> Of course an autopilot might have an arming flag which checks if the gimbal is available but that would be handled by the autopilot and not the gimbal itself.

I would have agreed with the "of course" one year back, but I don't do now anymore. The problem here really is that te configuration is spread over various locations, and on locations there it actually doesn't belong to. It should be a MAVLink gimbal who reports if it has passed it's check, and not the autopilot doing this on behalf of the gimbal. So, the autopilot would - and should - have that ONLY, and ONLY. if the autopilot really does something there it is critically dependent on the gimbal. What I think an autopilot rather would have is a check, which can be enabled or disabled, if ALL components are prearmed before it arms. But pl note the fundamental difference: This wouldn't be specific to any component, but generic for all components, and the autopilot actually wouldn't have to report that since the prearm checks are reported already by the components themselves. That is, the autopilot would only react on other's state, but not report or act on behalf of the other components. Likewise, the configuration would happen on the gimbal side, and not the autopilot side. I'm sure you see the fundamental difference in concept here.

julianoes, 2020-02-19:
The last point depends on how information flow about a gimbal is supposed to work. Is it up to the operator to decide if we should arm because autopilot and gimbal are ready (and both green on the ground station). Or do we prevent arming because the autopilot knows that it requires a gimbal and needs to wait for it? Or both? I don't think we should have to arm the gimbal/camera/etc., right?

olliw42, 2020-02-19:
> Is it up to the operator to decide if we should arm because autopilot and gimbal are ready (and both green on the ground station). 

I think this is a very viable option.

> Or do we prevent arming because the autopilot knows that it requires a gimbal and needs to wait for it? 

I think this is only in the most special cases a viable option, but not the default procedure.

If however you had written "Or do we prevent arming because the autopilot knows that it needs to wait for ALL components?" then I would have answered that this could be a very viable option too.

> Or both? 

or even both combined, if the user desires so

> I don't think we should have to arm the gimbal/camera/etc., right?

I wouldn't even know what for these components "to arm" possibly could mean, besides to power them up ...

bkueng, 2020-02-20:
> (1) As I indicated, I think a major problem with all this sort of proposals is that it is difficult to agree on the actually defined states. A key element of my proposal here therefore was to [...]

Apparently it is not be clear from the doc, but that is actually the case: only a small set of events need to be generally defined (for protocols like calibration), the rest is fully flexible, and I had an xml-file (or similar) in mind as well.

> For instance, it says "a sender periodically broadcasts its current sequence number (with a suggested interval of 3 seconds", so it just could report the prearmcheck state every 3 seconds, WITHOUT need of ANY of all the other clutter.

That is true and is one thing I'm considering.

Distributing reporting of prearm checks, I'm not convinced. It puts a lot of burden to a GCS (i.e. how does it know if a component just does not respond?), and in the end the autopilot needs to know if it can arm or not (and have the final word), and make an informed decision.
I see that it would simplify the set of messages and ease of extending them though. Do you have a potential list of prearm checks (that could fail) for a gimbal?

olliw42, 2020-02-20:
> only a small set of events need to be generally defined (for protocols like calibration), the rest is fully flexible, and I had an xml-file (or similar) in mind as well.

I will reread the document

> Distributing reporting of prearm checks, I'm not convinced.

what is the alternative? How would/should an autopilot get the info about the prearm status, if not via some MAVLink message, via SPI? If it's MAVLink messages which is send out by the component, what would make it then less "distributed" than what I've suggested ??? I'm sorry, I can't make sense of this comment

>  It puts a lot of burden to a GCS (i.e. how does it know if a component just does not respond?)

is this so? Would the burden of a generic method for the GCS to handle the prearms for all the different components so much higher than to get one prearm message form one component with a huge enum to disect and correlate with the compoinents? The only alternative with less burden than a generic approach is IMHO no approach (which is the current situation).

>  how does it know if a component just does not respond?

the same way like with all other messages?

> and in the end the autopilot needs to know if it can arm or not (and have the final word), and make an informed decision.

is this so? is this actually possible in any way? For instance, the user wants to take off in poshold and when switch to guided, could be handled by prearm checks "is_posholdable" and "is_guidable", but how would the autopilot decide? The message, only "simple" scenarios can be decided by the autopilot anyway.  Doesn't the autopilot ONLY need to know yes/no = one bit or does it really need to know all details?

Wouldn't it anyway need some interaction through the GCS to get all pieces to act as desired? What's the difference to a user to set a autopilot parameter, a component parameter, or a check box in a GCS? What would be actually most easy to implement? 

I for e.g. would be totally happy if I would get a prearm page presented by the GCS and then could decide MYSELF if I want to take off or not, and this would be the mode I most often would use. I would hate it if each time I would have to undergo plenty of reconfiguration steps (which mostly also means re-powering).

> Do you have a potential list of prearm checks (that could fail) for a gimbal?

This is what currently is implemented in the STorM32 gimbal controller:

[image]
(http://www.olliw.eu/storm32bgc-wiki/Prearm_Checks)

I wish to mention however that this is a relatively new feature and it is not clear yet if this is the most practical list of prearm checks, so, this may change somewhat based on experience, but I guess it gives a general flavor.

bkueng, 2020-02-24:
> what is the alternative? How would/should an autopilot get the info about the prearm status, if not via some MAVLink message, via SPI? If it's MAVLink messages which is send out by the component, [...]

- distributed: each component (like gimbals) send their status + messages directly to a GCS
- centralized: components send their status to the autopilot, which collects them and sends appropriate messages to a GCS

olliw42, 2020-04-25:
this is a very aggressive stale robot, isn't it

anyway, I still think that the made proposal is a very good, and the best seen so far approach and worth consideration

julianoes, 2020-04-27:
@olliw42 we have toned it down a bit: https://github.com/mavlink/mavlink/pull/1351

olliw42, 2020-04-27:
:)

thx for the feedback

I think that you may want also a mechanism to exclude some particular issues from the stale bot. I remember that there are a couple of issues which were specifically created so that a topic is not going to be forgotten about, even though it was clear at the point of their creation that there won't be action soon. So, these were specifically made to memorize!

It is good to have a bot reminding about issues. "Stale" and "pl don't forget" are however not exactly the same thing. A "Pl don't forget" would come across a bit less scary than a "Stale!". This comment is obviously from someone not very accustomed to stale bots :)

julled, 2022-09-09:
Any progress on this? 

I am in the need to send a prearm check results from a onboard vision system to the GCS and for this i would need a way to do  so. I would appreciate if there is a mavlink message for this.

olliw42, 2022-09-09:
sure, in operation since long
when I encounter a need and condense it into a proposal but it doesn't make its way, I have to go my own way, and the answer can often be found in storm32.xml: https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/storm32.xml#L560-L567
:)
the flags are obviously component specific, component info or however it is called today could be the resolution as suggested in the above