mavlink/mavlink #1306: Pl add VIDFPS and VIDFOV to Common Camera Parameters

olliw42, 2020-02-18:
currently, the common parameters for a camera are given per specification in the docs (https://mavlink.io/en/services/camera_def.html#common-parameters) as 

CAM_APERTURE | Aperture
CAM_EV | Exposure Compensation (usually only used for automatic exposure modes)
CAM_EXPMODE | Exposure Mode (Manual, Auto, Program Auto, Aperture Priority, etc.)
CAM_ISO | ISO
CAM_METERING | Metering Mode
CAM_SHUTTERSPD | Shutter speed
CAM_VIDRES | Video Resolution (for video capture)
CAM_WBMODE | White Balance Mode

However, most action cameras at least also allow setting the video frames per second (FPS) and the video field of view (FOV). Furthermore, for many of them correlations exist between the possible settings for video resolution (RES), FPS, and FOV, which means that displaying and allowing to set RES without letting the user know and adjust the effect on FPS and/or FOV is quite impractical and of little use.

I thus suggest to add

CAM_VIDFPS | Video frames per second (for video capture)
CAM_VIDFOV | Video field of view (for video capture)

to the set of common parameters. This will allow GCSes to also check for them in a generic manner and present the mentioned correlations to the user.

Thx and Cheers, Olli

hamishwillee, 2020-02-18:
@dogmaphobic This sounds reasonable, but I don't have enough experience to know how common they are. Thoughts?

olliw42, 2020-02-18:
as much as I understand it these are just recommendations, i.e., a GCS can but does not have to provide special GUI elements for them. I believe QGC actually doesn't handle any of them in a special way. 
So, I would argue, one can't do really much wrong.

I actually would argue that one could go even further and suggest common parameters separately for VID and PHO mode. E.g.,  the gopros and several other action cams hold the EV for video and photo mode in different parameters. That is, in order to handle them in a generic way one would need to define CAM_VIDEV and CAM_PHOEV. With just the CAM_EV one can't support that case. It's also so for the shutter speed, or ISO, which with the current common parameters couldn't be handled generically. A GCS could easily determine what to do from the fact what parameters, CAM_EV, or CAM_VIDEV and CAM_PHOEV it would find. 

But I didn't wanted to go that far and complicate the matter (even though it would be useful LOL) ... 

:)

dogmaphobic, 2020-02-19:
> So, I would argue, one can't do really much wrong.

That's true. It's better to have a common set of parameter names even if we don't have special UI handling for them because once you do, you're out of luck if each implementation uses different name for the same thing.

Here are a few examples:

```
CAM_VIDFMT          Video encoding format
CAM_VIDFPS          Video frame rate
CAM_PHOTORATIO      Photo Aspect Ratio
CAM_PHOTOFMT        Photo image (saving) format
CAM_PHOTOSIZE       Photo size
CAM_PHOTOQUAL       Photo image (saving) quality (for compressed modes)
```

Feel free to suggest others.

>However, most action cameras at least also allow setting the video frames per second (FPS)

"Action cameras" make up a minute fraction of the target audience for the camera API. I wouldn't bother with them unless a manufacturer were to provide a camera with MAVLink support. With that said, _all_ camera capable of recording video have these options (frame rate, resolution, encoding method, etc.) These are rarely handled from a "quick" UI though. You usually must go into a settings menu to change it. That is already supported and functional. Following the above logic however, the more standard these parameter names are, the better.

> I actually would argue that one could go even further and suggest common parameters separately for VID and PHO mode.

For the most part, that's already handled within the camera definition by the use of rules and exclusions. Having two parameter names for the same thing will only make things confusing or altogether unusable. When I look at the possible values for "EV", I already get a list of values that are appropriate for the current mode. When you switch modes, and the "other" mode has a different set of options, those are replaced automatically by the core camera API. The list is updated and the UI is updated accordingly. Further, if you switch to manual mode, where EV is not applicable, the whole EV control is hidden automatically.

olliw42, 2020-02-19:
> For the most part, that's already handled within the camera definition by the use of rules and exclusions.

one might think so but this is not correct. Note that each entry in the definition file corresponds to a Mavlink parameter, which is therefore so also in the rules and exclusions. Therefore, if a feature like say EV actually corresponds to two distinct parameters for VID and PHO modes one can't do it with only one CAM_EV and with what the definition file offers. One would to actually have to do an implementation nightmare  and somehow shadow this so that to the outside it looks as if there is just one parameter even though there are two internally.  (and this is actually the reason why I brought this up in the first place, I could find a way of doing it with the current tools).

but I think this discussion is largely academic and not so relevant, since it seems to me that you agree that it can't harm to rather have more common parameters than too less

so, I feel encouraged to suggest in addition

CAM_VIDEV
CAM_PHOEV
CAM_VIDISO
CAM_PHOISO
CAM_VIDSHUTSPD
CAM_PHOSHUTSPD

:)

olliw42, 2020-04-25:
this is a very aggressive stale robot, isn't it

anyway, I still think that the made proposal is a very good, and best I have seen so far, approach and worth consideration

hamishwillee, 2021-09-09:
@dogmaphobic @MatejFranceskin @julianoes I would appreciate your thoughts on @olliw42 response in https://github.com/mavlink/mavlink/issues/1306#issuecomment-588095502 that we really do need _separate_ [common parameters](https://mavlink.io/en/services/camera_def.html#common-parameters) for features that might be different in photo or video modes.

To summarize how it is now (using ISO as an example) we define `CAM_ISO` as the common parameter for camera ISO settings. The camera might support different ISO options in video and still modes, and the definition file allows us to specify the range to display in the UI in each. When you select an option, the value of the parameter `CAM_ISO` is sent to the camera.

This works for defining a UI to select the ISO values available for either video or photo modes. 

Ollie's assertion is that you can only send one parameter `CAM_ISO`, but the implementation might reasonably require separate parameters for CAM_PHOTO_ISO and CAM_VIDEO_ISO. These aren't defined by the UI so the camera implementation needs to be complicated to map the parameter we get to different modes. 

The first example that sprang to my mind was a use case where we want the camera to store and restore its video ISO setting on mode switch. That is trivially easy with two parameters. Annoying with one because you have to map parameter you get to the actual parameter used for the current state.

How has this been handled in current implementation? What do we think?

hamishwillee, 2021-09-16:
We discussed this in the [dev call](https://github.com/mavlink/mavlink/wiki/20210915-Dev-Meeting). We can see that there are cases where it would be useful to have separate parameters. I pushed against change unless there was a stakeholder that wants to verify that there are no unintended consequences by prototyping this end to end. 

Despite pushing for that I am not convinced I'm "right". We do want to reserve some of these parameters, and I want to convince myself that reserving the "wrong" params doesn't put us in a worse position.

hamishwillee, 2021-09-16:
Below is list of all the keys used in the definition files in mavlink, ordered by occurrences. 
- Bold indicates are already [common params](https://mavlink.io/en/services/camera_def.html#common-parameters).
- Italic indicates suggested by Gus and/or Ollie - I think we could just add these.

Any of these we think are no-brainers to usefully add now?

  | cgoet | ILCE7RM4 | e10t.xml | e30z.xml | e50.xml | e90.xml
-- | -- | -- | -- | -- | -- | --
CAM_MODE | Y | Y | Y | Y | Y | Y
**CAM_ISO** | Y | Y | Y | Y | Y | Y
**CAM_EXPMODE**  | Y | Y | Y | Y | Y | Y
**CAM_SHUTTERSPD** | Y |   | Y | Y | Y | Y
**CAM_EV** |   | Y | Y | Y | Y | Y
**CAM_WBMODE**  |   | Y | Y | Y | Y | Y
CAM_ASPECTRATIO | Y |   | Y |   | Y | Y
CAM_WIFIPASSWD | Y |   | Y | Y |   | Y
CAM_SYSTEMTIME | Y |   | Y | Y |   | Y
**CAM_METERING** |   |   | Y | Y | Y | Y
**CAM_VIDRES** |   |   | Y | Y | Y | Y
CAM_IRPALETTE | Y |   | Y |   |   |  
CAM_IRFFCMODE | Y |   | Y |   |   |  
CAM_IRATMO | Y |   | Y |   |   |  
CAM_IREMISS | Y |   | Y |   |   |  
CAM_CONVRATIO | Y |   | Y |   |   |  
CAM_IRATMOTEMP | Y |   | Y |   |   |  
CAM_TEMPSTATUS | Y |   | Y |   |   |  
CAM_CUSTOMWB |   |   |   | Y |   | Y
CAM_FLICKER |   |   |   | Y |   | Y
CAM_FWBUNDLEVER |   |   |   | Y |   | Y
_CAM_PHOTOFMT_ |   |   |   |   | Y | Y
CAM_COLORMODE |   |   |   |   | Y | Y
CAM_SPOTAREA |   |   |   |   | Y | Y
CAM_IRLOCKST | Y |   |   |   |   |  
CAM_IRTEMPRENA | Y |   |   |   |   |  
CAM_IRTEMPMAX | Y |   |   |   |   |  
CAM_IRTEMPMIN | Y |   |   |   |   |  
CAM_FOCUSMODE |   | Y |   |   |   |  
CAM_SHUTTER |   | Y |   |   |   |  
**CAM_APERTURE**  |   | Y |   |   |   |  
CAM_MOVFORM |   | Y |   |   |   |  
CAM_MOVQUAL |   | Y |   |   |   |  
CAM_IMGSIZE |   | Y |   |   |   |  
CAM_IMGDEST |   | Y |   |   |   |  
CAM_BITRATE   - video bitrate | Y |   |   |   |  
CAM_IRENV |   |   | Y |   |   |  
CAM_IRCALIBD |   |   | Y |   |   |  
CAM_IRCALIRUN |   |   | Y |   |   |  
CAM_IRSAVE |   |   | Y |   |   |  
CAM_IRGAINMODE |   |   | Y |   |   |  
CAM_IRACE |   |   | Y |   |   |  
CAM_IRDDE |   |   | Y |   |   |  
CAM_IRPNINFO |   |   | Y |   |   |  
CAM_IRRADPNT |   |   | Y |   |   |  
CAM_IRRADAREA |   |   | Y |   |   |  
CAM_IRROI |   |   | Y |   |   |  
CAM_OPTZOOM |   |   |   | Y |   |  
CAM_DZOOM |   |   |   | Y |   |  
CAM_DIGCTRL |   |   |   | Y |   |  
CAM_OPTCTRL |   |   |   | Y |   |  
CAM_FOCUS_MODE |   |   |   | Y |   |  
CAM_FOCUS_CTRL |   |   |   | Y |   |  
CAM_DEFOG_LEVEL |   |   |   | Y |   |  
CAM_IRIS |   |   |   | Y |   |  
_CAM_PHOTORATIO_ |   |   |   |   |   | Y
_CAM_VIDFMT_ |   |   |   |   |   | Y
_CAM_PHOTOQUAL_ |   |   |   |   |   | Y
CAM_IMAGEDEWARP |   |   |   |   |   | Y
CAM_COLORENCODE |   |   |   |   |   | Y
CAM_PHOTOMODE |   |   |   |   |   | Y
CAM_FILENUMOPT |   |   |   |   |   | Y
CAM_ZOOM |   |   |   |   |   | Y

Further, these are not above, but seem to be accepted by both Ollie and Gus. Are we OK with these to be added?
CAM_VIDFPS
CAM_PHOTOSIZE

The remaining question was whether we split these. I'd say we can defer that for now. I can understand shutterspeed, but would we actually ever need different ISO? 
- CAM_EV => CAM_VIDEV / CAM_PHOEV
- CAM_ISO => CAM_VIDISO / CAM_PHOISO
- CAM_SHUTTERSPD => CAM_VIDSHUTSPD / CAM_PHOSHUTSPD

olliw42, 2021-09-16:
I don't understand what's going on here

this is the list of params potentially suggested to standard? the large majority to me seems like very camera specific

the original idea was based on two simple view points:

1) if users have to name let's say 5 or 7 camera parameters they for sure would want to have access to in the GUI, what would these be?
e.g., I'm betting VIDFOV would be among everyone's list

2) concerning different params of the same kind for VID and PHOTO it is a technical fact that cameras exist which offer different values for either vid or photo mode
e.g. gopros 

I have a bit difficulties to understand that compression and storage related parameters should be of so much more common desire than video and photo framing parameters

also, sadly, the ones which are not included, i.e.
CAM_EV => CAM_VIDEV / CAM_PHOEV
CAM_ISO => CAM_VIDISO / CAM_PHOISO
CAM_SHUTTERSPD => CAM_VIDSHUTSPD / CAM_PHOSHUTSPD
would be of most interest to me (interested in gopro's)

I want to say that I do not see any reason why one could not have three params of one kind, one general, one VID specific, and one PHOTO specific. For instance
CAM_SHUTTERSPD & CAM_VIDSHUTSPD & CAM_PHOSHUTSPD
the GUI should have no really difficulty to figure out which one to use for which camera mode

(the rules could be: 
if mode is PHO and CAM_PHOxxxx exists use it, else use CAM_xxxx if it exist, else do nothing
if mode is VID and CAM_VIDxxxx exists use it, else use CAM_xxxx if it exist, else do nothing)

I'd like to add that I certainly do not have a very strong opinion here, since, as a matter of fact, it's all working fine thanks to this fantastic definition file concept.

julianoes, 2021-09-16:
> I want to say that I do not see any reason why one could not have three params of one kind, one general, one VID specific, and one PHOTO specific. For instance
CAM_SHUTTERSPD & CAM_VIDSHUTSPD & CAM_PHOSHUTSPD
the GUI should have no really difficulty to figure out which one to use for which camera mode

I would not be opposed to having both.

hamishwillee, 2021-09-16:
@olliw42 Relax, I can remove your confusion :-)

1. I'm more than happy to add everything you and Gus either agreed on or didn't explicitly agree on: CAM_PHOTOQUAL, CAM_VIDFMT, CAM_PHOTORATIO, CAM_PHOTOFMT, CAM_VIDFPS, CAM_VIDFOV
2. That includes CAM_VIDFOV which I accidentally omitted.
3. The list I generated is "complete". We don't have to add a single extra thing from it. It is presented so we can all see it and ask ourselves if there is any other parameter we would like to have in the common set. As you pointed out, better to have something fixed now if we might reasonably want it pulled out in the UI.
4. My involvement was kicked off by the discussion of CAM_STREAM - we should add that too. 

As I stated, my only concern about the duplicate parameters is that this discussion is completely abstract - there is no implementation so their might be unexpected consequences. 
- I can imagine someone using CAM_PHOSHUTSPD thinking that the GCS will render it nicely - but the GCS uses CAM_SHUTTERSPD only, so the camera does not get the benefit. Then what should the camera vendor do? Are we expecting GCS vendors to now support any combination of these settings in the UI?

@julianoes if you're happy to have duplicates for CAM_SHUTTERSPD & CAM_VIDSHUTSPD & CAM_PHOSHUTSPD (etc), are you happy to create the PR and approve it too? :-)

olliw42, 2021-09-16:
hihi ... I am quite relaxed on this :) sorry if I came across differently 

> As I stated, my only concern about the duplicate parameters is that this discussion is completely abstract - there is no implementation so their might be unexpected consequences.

from my perspective: 

no, it is not completely abstract but very real

It reflects how the gopros work, and there is also an available implementation and use case. I do have since more than 2 years now a MAVLink interface to the gopro hero5 & hero7 as part of the STorM32 gimbal project (http://www.olliw.eu/storm32bgc-wiki/NT_Camera#GoPro_Hero) and am currently working on an interface for the hero8 & 9. I also use that in my MAVLink for OpenTx project, which includes lua script which allows the user to set these settings for the video and photo mode with a press of a button on the Tx screen. Not all but most of this is os, and it is all for free (like free beer) and usable any time anyone wants to use it. Download, install, go. You may watch these videos to convince yourself that this is all real and not just paperwork :)
https://www.youtube.com/watch?v=kT4zbvChs2g
https://www.youtube.com/watch?v=_56VRxEAWkM

> I can imagine someone using CAM_PHOSHUTSPD thinking that the GCS will render it nicely - but the GCS uses CAM_SHUTTERSPD only, so the camera does not get the benefit. Then what should the camera vendor do? Are we expecting GCS vendors to now support any combination of these settings in the UI? Does it matter? I don't know.

this is academic, if not artificial

there is no problem. A GCS is NOT required to provide special nice ui for standardized parameters. If a GCS doesn't want to do that, no problem. Similarly, a camera is NOT required to use standardized parameters. However, the GCS or camera may choose to do so. It is all flexible, with no downsides but just upsides.

julianoes, 2021-09-17:
> there is no problem. A GCS is NOT required to provide special nice ui for standardized parameters. If a GCS doesn't want to do that, no problem. Similarly, a camera is NOT required to use [...]

I can see that more params make it more complext if you want to support both properly but I do agree with what you write here. Hence, both is fine with me.

> @julianoes if you're happy to have duplicates for CAM_SHUTTERSPD & CAM_VIDSHUTSPD & CAM_PHOSHUTSPD (etc), are you happy to create the PR and approve it too? :-)

I'll approve whatever comes along but I'm not driving this.

hamishwillee, 2021-09-18:
Given that @julianoes and @MatejFranceskin seem OK with this and I have no strong feelings either way, why not. I'll create a PR. 

@olliw42 Does the actual name matter for you? I am more likely to push for CAM_VID_SHUTSPD & CAM_PHO_SHUTSPD to make these easier to parse when I'm reading it.

julianoes, 2021-09-18:
My 2 cents is that an additional underscore makes it easier to read.

hamishwillee, 2021-09-23:
Fixed by https://github.com/mavlink/mavlink-devguide/pull/395

FYI stuck with common format, which is CAM_VIDWHATEVER and CAM_PHOTOWHATEVER

hamishwillee, 2021-10-06:
Thanks for your patience @olliw42

olliw42, 2021-10-09:
many thx, sir !!