mavlink/mavlink #1314: Clarification on TERRAIN_REPORT.pending/loaded

DonLakeFlyer, 2020-02-28:
Spec says:
* pending - Number of 4x4 terrain blocks waiting to be received or read from disk
* loaded - Number of 4x4 terrain blocks in memory

But from the work i'm doing to support in QGC it seems like this is really:
* pending - number of blocks the vehicle is still waiting on from the gcs
* loaded - number of blocks the vehicle currently has

Hopefully some ArduPilot folks could help out with how they implemented.

rmackay9, 2020-02-29:
@DonLakeFlyer, I'm not really the expert in this area but I think your interpretation is correct.

I think the two are pretty much the same, it's just that spec's version doesnt' mention that it's speaking from the vehicle/autopilot's point of view instead of from the GCSs point of view.

DonLakeFlyer, 2020-02-29:
Thanks @rmackay9.

DonLakeFlyer, 2020-02-29:
@hamishwillee Something to update in the spec

DonLakeFlyer, 2020-02-29:
@hamishwillee Another thing which wasn't clear to me from reading spec was on TERRAIN_REQUEST.mask. Bit 0 position (TERRAIN_REQUEST.mask = 1) in that mask refers to the last row first column in the 8x7 array of grids. I initially assumed it was first row/column.

hamishwillee, 2020-04-30:
Great. Feel free to PR this, and then I can review. Otherwise it is on the list, but not my highest priority.

hamishwillee, 2020-06-29:
@DonLakeFlyer Closing. This will be fixed by https://github.com/mavlink/mavlink-devguide/pull/259 - but I have open questions there for you.

closed 2020-06-29 by hamishwillee