mavlink/mavlink #1320: Message to substitute GNSS data in FC

wucke13, 2020-03-12:
Hi,
is there a message which allows one to overwrite the GNSS data (global current position of the vehicle) in the FC? If not, are there any plans to implement one?

amilcarlucas, 2020-04-02:
Try the HIL message or the GPS_INPUT message

hamishwillee, 2020-04-02:
Thanks @amilcarlucas. Sorry @wucke13  - I completely missed this message. 

Note that 
- [GPS_INPUT](https://mavlink.io/en/messages/common.html#GPS_INPUT) might not be implemented on your flight stack. If it is, it may be fused with other information - i.e. it should be fused with other sources, not replace them. 
- The HIL messages are only used for hardware simulation - it is unlikely they will work in a normal airframe.

So perhaps best to further explain why you need this message/use case?

wucke13, 2020-04-03:
Hi, thanks for the responses. I'm working on a MAVLink to MSP bridge which allows one to use a drone which supports MSP (namely iNAV based ones) in a MAVLink network. However, one of the remaining challenges is GPS and Waypoint navigation. AFAIK MSP only understands a global position. I would prefer the bridge to have only two communication ends, one for MAVLink and one for MSP. So, in order to push substituted location information to the FC (e.g. from a POZYX system), it would be necessary to push global position data to the Drone (as MSP doesn't understand local position frames), and in order to push a global position to the bridge there needs to be a MAVLink message for it. I hope this clears up my intentions a bit. 

Also, I think it is a nice thing to be able to substitute global position because it allows one to use MAVLink on the component level to transparently use a high quality position device without having it emit NMEA or similar things.

amilcarlucas, 2020-06-02:
AFAIK GPS_INPUT is the messag you need.

wucke13, 2020-06-02:
@amilcarlucas Ah, thank you!

closed 2020-06-02 by wucke13