mavlink/mavlink #1330: mavlink_helpers.h includes an undefined variable `mavlink_system`

Marxlp, 2020-03-31:
When tried to building the testsuite, I got
```bash
/mnt/d/github_/mavlink_cpp/common/../mavlink_helpers.h: In function ‘_mav_finalize_message_chan_send’:
/mnt/d/github_/mavlink_cpp/common/../mavlink_helpers.h:327:22: error: ‘mavlink_system’ undeclared (first use in this function); did you mean ‘mavlink_system_t’?
             buf[3] = mavlink_system.sysid;
                      ^~~~~~~~~~~~~~
                      mavlink_system_t
```
I checked the *mavlink_helpers.h*. It includes a variable `mavlink_system` which is not defined or declared in the context of function `_mav_finalize_message_chan_send`. Besides, I searched the code generated by *mavgen.py* using grep, but I can't find any `mavlink_system` in other files. 
Is it a **bug**?

hamishwillee, 2020-04-03:
Looks like it to me.

@peterbarker How can `mavlink_system` be reached here? https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L341

There is a mavlink_system_t that has this structure, but again, not defined anywhere. 

Should this move into ArduPilot/pymavlink issues?

peterbarker, 2020-04-03:
On Thu, 2 Apr 2020, Hamish Willee wrote:

> @peterbarker How can mavlink_system be reached here? https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L341

It's not made clear, but this symbol must exist in the global namespace.

Implementor's responsibility.

Here's ArduPilot implementing it:

https://github.com/ardupilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_MAVLink.cpp#L45

And here's PX4 Firmware implementing it:

https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink.c#L49

Marxlp, 2020-04-03:
I think it's not a good idea to not be self-contained. Is there any special consideration to be like this? And, is it possible to be one of function's parameter?

peterbarker, 2020-04-03:
On Fri, 3 Apr 2020, LP wrote:

> I think it's not a good idea to not be self-contained. Is there any special consideration to be like this? And, is it possible to be one of function's
> parameter?

Oh, it's a disaster on many fronts.  But it's unlikely to change in the 
near future.

Peter

hamishwillee, 2020-04-06:
Thanks @peterbarker - I created https://github.com/mavlink/mavlink-devguide/pull/223 to at least record the dependency:
1. Seem reasonable/sufficient
2. Are there any other global surprises like this?

It would be good if this were "fixed". If you could provide direction on what sort of fix might be considered "good" perhaps @Marxlp could do a PR?

peterbarker, 2020-04-06:
> Thanks @peterbarker - I created [mavlink/mavlink-devguide#223](https://github.com/mavlink/mavlink-devguide/pull/223) to at least record the dependency:
> 
>     1. Seem reasonable/sufficient

Added a review.  But yes, generally.

>     2. Are there any other global surprises like this?

Oh, probably.  Fun things around multi-threaded access which we don't have hooks sufficient to cope with spring to mind - but I don't have those straight in my head right now.  One of the more interesting is something like https://github.com/ArduPilot/ardupilot/pull/11524 - where we want to send from different component IDs.

> It would be good if this were "fixed". If you could provide direction on what sort of fix might be considered "good" perhaps @Marxlp could do a PR?

Notionally it could be added as a weak symbol - but that's something that would have to be tested on a wide variety of implementations (especially on random compilers/linkers people might be using).

hamishwillee, 2020-04-06:
Yuk. OK then I'd say this is as much as I can reasonably do for now. Thanks!

closed 2020-04-07 by hamishwillee