mavlink/mavlink #1336: Docs for DO_MOUNT_CONTROL don't match reality of ArduPilot

DonLakeFlyer, 2020-04-01:
The current wording matches PX4 only. I'm discovering that ArduPilot doesn't work that way. The ArduPIlot units for roll/pitch/yaw are centidegrees. Whereas for PX4 they are degrees.

DonLakeFlyer, 2020-04-01:
Would also be helpful to state that level for pitch is 0 and straight down is -90 (in appropriate unites).

LorenzMeier, 2020-04-01:
That's the MAVLink spec - I'm happy to review where the divergence occurred if someone has a pointer to if / when the spec changed (which would point at a spec process issue). If the spec didn't change then I'd say this is out of scope for MAVLink, as there is no way from a protocol perspective to enforce correct adoption.

There is a v2 protocol that has been decided on the dev call and that will put an end to a lot of the historic uncertainty of that part of the system: https://mavlink.io/en/services/gimbal_v2.html

It is currently being adopted.

DonLakeFlyer, 2020-04-01:
I thought there were other places in the messages spec where differences were documented? I thought so, but in looking now I can't find any. Oh well.

closed 2020-04-01 by DonLakeFlyer

LorenzMeier, 2020-04-01:
I just checked which commit changed this and it is since 2014 in this state and the commit refers to having checked this explicitly with the APM devs, so if there is a difference to the spec, I'd consider that an implementation problem: https://github.com/mavlink/mavlink/commit/8458b6ce66cb677c5c7fb55e4450f741b12fcc47

Note that the message is deprecated since 2019.

DonLakeFlyer, 2020-04-01:
DO_MOUNT_CONTROL is marked as deprecated yet the messages that replace it are marked as WIP/Do not use?

DonLakeFlyer, 2020-04-01:
> Would also be helpful to state that level for pitch is 0 and straight down is -90

@hamishwillee I think this would still be useful though

hamishwillee, 2020-04-02:
Hi @DonLakeFlyer 
1. The known issues with the (old) gimbal protocol are documented [here](https://mavlink.io/en/services/gimbal.html#known_issues). That there are so many problems with this is part of the reason why we moved to a [new protocol](https://mavlink.io/en/services/gimbal_v2.html). 
2. 
   > DO_MOUNT_CONTROL is marked as deprecated yet the messages that replace it are marked as WIP/Do not use?
   
   Yes. WIP is correct. I would have expected the old API to not yet be deprecated though. 
   - The new messages went in a couple of weeks ago. Docs this week. They stay WIP until there is an implementation that can validate them. 
   - There is currently work going on in MAVSDK to create a plugin for this - that has already discovered some issues with the API and docs (which we are fixing).
   - There is no PX4 implementation yet. I believe @julianoes is planning for this.

3. 
   > I think this would still be useful though

   Sure - if you can confirm that this is how it is implemented for both ArduPilot and PX4? How about you create a PR?

But probably I'd update the description for the whole message and related to a standard frame - that way it applies to all the inputs. I think what you are describing is FRU?

So something like: "Mission command to control a camera or antenna mount. Roll, pitch, yaw, values are in the FRU coordinate frame."

DonLakeFlyer, 2020-04-02:
> Would also be helpful to state that level for pitch is 0 and straight down is -90 (in appropriate unites).

I have verified that this is the case for both PX4 and ArduPilot.

> I think what you are describing is FRU?

I don't know what FRU is.

hamishwillee, 2020-04-02:
>> I think what you are describing is FRU?

> I don't know what FRU is.

I think I mean [FLU](https://mavlink.io/en/messages/common.html#MAV_FRAME_BODY_FLU), sorry. I say "think" because you will know more about the correct frame than me - I just know that it is better to refer to the frame to cover roll, pitch, yaw than attempt to deal with them invidually.

DonLakeFlyer, 2020-04-03:
FLU - I had never seen that either! I don't think that terminology really matches gimbal since it is for flight paths I think. 

As far as I know for gimbal say with a vehicle sitting on level groundl:
* Pitch: 0 - level with horizon, -90 pointing straight down
* Yaw: positive is clockwise, negative is counter-clockwise
* Roll: No idea

Whatever acronym that is, don't know.

julianoes, 2020-04-04:
FRD: Forward, Right, Down.

- Roll along x: positive is tilting to right
- Pitch along y: 0 - level with horizon, -90 pointing straight down
- Yaw along z: positive is clockwise, negative is counter-clockwise

hamishwillee, 2020-04-05:
@julianoes Can you confirm whether you mean [MAV_FRAME_LOCAL_FRD](https://mavlink.io/en/messages/common.html#MAV_FRAME_LOCAL_FRD) or [MAV_FRAME_BODY_FRD](https://mavlink.io/en/messages/common.html#MAV_FRAME_BODY_FRD).

I'm not entirely sure of the difference (always get confused by this, but I THINK that local is relative to a fixed world and that body is relative to the vehicle itself. I would have though probably that the axis would be relative to the vehicle ... but we should be clear.

julianoes, 2020-04-06:
@hamishwillee you're opening a whole can of worms with that. This would depend on the `yaw_absolute` discussion, and it's also one of the shortcomings of the old protocol. So often yaw would be relative to body but then roll and pitch are relative to the world because you want the gimbal to be stabilized with the horizon, at least for most use cases. Once you do a ROI or inspection you might want to have "world" yaw so look into a certain direction independent of the drone's yaw.

hamishwillee, 2020-04-07:
@DonLakeFlyer So I don't want to open a can of works ;-0. I agree in principle but as we're moving onto a new protocol I'm not sure it is worth the time to untangle.

DonLakeFlyer, 2020-04-07:
All seems fine to me then.