mavlink/mavlink #1344: Component Id for a companion computer

olliw42, 2020-04-14:
I'd like to make my companion computer (a RPi) to be conform with MAVLink, and hence want to give it a component ID. Unfortunately, it's not clear to me what it should be, and it in fact seems to me a suitable id is missing.

The issue I face is that a companion CAN be of some of the available ids, but that it also may not be, and more over might fit MORE THAN ONE of the available ids.

For instance:

MAV_COMP_ID_LOG: it may log or may not log
MAV_COMP_ID_MISSIONPLANNER: it may do a mission plan or may not
MAV_COMP_ID_PATHPLANNER: it may do path planning or may not
MAV_COMP_ID_OBSTACLE_AVOIDANCE: it may do avoidance or may not
MAV_COMP_ID_VISUAL_INERTIAL_ODOMETRY: it may do odometry or may not

so the closest seems to be
MAV_COMP_ID_PERIPHERAL
but here to: it may provide parameters or it may not.

I thus suggest to add a new component id

MAV_COMP_ID_ONBOARD_COMPANION

It's id might be e.g. 161, 176, 199, 194, ...

Thanks
Olli

hamishwillee, 2020-04-29:
Hi @olliw42 

Not sure, even after some discussion.

@baumanta You worked on obstacle avoidance. Can you confirm what MAV_TYPE and MAV_COMPONENT you use for a companion computer when it has multiple ROS nodes - ie it also supports VIO?

@auturgy Says that his understanding is that the component id belongs to the process. So if you have a companion that does VIO and Obstacle avoidance, those nodes should emit their own heartbeats with those ids. But would be great if you can confirm.

Not sure about what you should use if there are choices. A MAV_COMP_ID_OBSTACLE_AVOIDANCE **is a** path planner. It probably won't be a MAV_COMP_ID_MISSIONPLANNER though - that is intended for something that sends out complete plans like a GCS. 

A further inconsistency is that we don't expect an autopilot to necessarily emit a separate HEARTBEAT if it implements the behaviour of a component. Another complication is when to use MAV_TYPES vs components. 

Now I've added to the confusion, perhaps we can converge on an answer based on "what is done now".

baumanta, 2020-04-29:
There is no need to add a new COMPONENT_ID. Mavros will send heart beats automatically with component ID 195 (MAV_COMP_ID_PATHPLANNER) if I remember correctly. This is basically the same as the companion computer ID (because you will never have more than one mavros node running). And then all the additional nodes you might be running to do specific things will have to get their specific comp_id's (avoidance = 196, VIO = 197).

olliw42, 2020-04-29:
first, thanks for considering, and I hear you folks.

If there is a recommended ID which one could or should use for the purpose I would be happy to do so. 

Please consider that I also have a bit of a "User" perspective here. E.g., in Missionplanner for each ID I see in plain word what component it is, i.e., for a gimbal it displays "GIMBAL" and for a camera it displays "CAMERA", and I find that VERY helpful, and thus would like to see also something useful being displayed for the companion. And as I said, in contrast to seeing "GIMBAL" and "CAMERA" seeing "MISSIONPLANNER" or "PATHPLANNER" or else doesn't make much sense to me.

> A further inconsistency is that we don't expect an autopilot to necessarily emit a separate HEARTBEAT if it implements the behaviour of a component. 

ähm ... this adds to the confusion ... I was not even remotely trying to suggest that a autopilot should send out a separate heartbeat

I was however trying to suggest that a companion computer may well send out it's own heartbeat, i.e. present itself to a MAVLink network as a MAVLink component. I actually think that this is the proper way, i.e., I think that any thing on the MAVLink network which is not just listening but also sending messages needs it's own ID and also should send it's heartbeat.

There is talk of "processes". This is not what I'm talking about, I'm talking about "MAVLink component", assuming that this is a sufficiently well defined object. A process may be a MAVLink component, and a MAVLink component may be a process, but a process can also be not a MAVLink component, while a MAVLink component of course cannot be not a MAVLink component, so this are different things, and I talk about the latter. :)

> Another complication is when to use MAV_TYPES vs components.

I agree and believe to have pointed that out before and suggested possible approaches. However, I really didn't wanted to open that box (of p...) here, but just wanted to suggest introduction of one more ID.

hamishwillee, 2020-08-06:
This has come up [yet again](https://github.com/mavlink/mavlink/pull/1443), and I think we need to resolve it. 

Originally `MAV_TYPE` was the type of system and the component ID was the type of thing in the system using a preallocated instance id. By example, a gimbal in vehicle would have the MAV_TYPE of the vehicle (copter say) and the id (gimbal 1). This was a problem because said gimbal emitting a heartbeat can't know in advance the MAV_TYPE of the parent vehicle or its system ID. So either you have to remove the need to set that, or force gimbal supplier to provide an easy API for setting those params.

@olliw42 Is ^^^ correct summation. Don't you still need to set the system id to match your system. How do you get that?

We changed this so that the `MAV_TYPE` is not the type of the parent - it is the type of the thing - ie. a gimbal. The component id is the _instance_ of that component. It *may* use the instance ids for the type, but may not. Canonically, the receiving system should work out what sort of thing it is working with by checking the MAV_TYPE. If it gets multiple components of a type then it should treat the components as instance ids.
We also decided that the autopilot would use the MAV_TYPE of the vehicle so you could still identify it.

@olliw42 @julianoes Is that correct w.r.t. to what we discussed? I'm still not at all comfortable with all of that.

Doesn't this complicate message handling a lot? Previously I could say I'm only interested in component ids of a type - e.g. I check messages is from "mission planner component" so I will accept it. Now I have to have code that checks heartbeat, finds out the type is GCS, then in message handling code uses that mapped id. Similarly, I have to use my actual component ID from heartbeat to address the GCS, not the one that I could otherwise hardcode.

Another problem is consider proposal to add a companion computer component ID. IF I have a path planner as well then both components would currently have MAV_TYPE of MAV_TYPE_ONBOARD_CONTROLLER  - and be interpreted as 2 companion controllers.

So for this case to work wouldn't we need a new type for the path planner software that is running on the onboard controller?

olliw42, 2020-08-07:
I'm not surprised it did come up again 
(conceptual flaws tend to always come up - sooner or later)

I too would think that the matter should be resolved, the sooner the better.

I'd like to mention that we had related discussions ca 1 y ago, spawned over several threads in this repo, and I thought we kind of arrived at a conclusion. Ca 1/2 y later when I made some suggestions to QGC this seemed to have dissolved however. No complain, just trying to say that (1) there had been substantial discussion and it probably would be useful to recollect the arguments and not start over from scratch (I unfortunately can't remember the details anymore), that (2) I'm not totally sure to which point in time you are referring to, and that (3) the result should (must) be documented ASAP so that it can serve as reference to anyone. It shouldn't be dark matter.

IMHO the matter consists of

- identification
- discovery
- capability
 
These three points are IMHO inter-twinned. That is, any attempt to only solve one likely will fail, i.e., all three should (must) be looked at together. 

However, at the same time, the three should not be confused with each other !

> Don't you still need to set the system id to match your system. How do you get that?

Julian once made an excellent suggestion, which I'm using since then: If in the GUI the parameter SystemID is set to 0 (= auto discovery), then my gimbal and camera components wait for a heartbeat from a flight controller or a GCS, and uses their system id, only then it starts emitting a heartbeat and doing other MAVLink stuff itself. If the parameter SystemID > 0 then of course that is used and the wait is not done. The autodiscovery really works very well in practice.

> Is that correct w.r.t. to what we discussed? I'm still not at all comfortable with all of that.

I'm not sure about the words "instance" and such. In my recollection, ca 1 y ago we agreed that

- MAV_TYPE has higher priority than component_id. That is, a component must set it's MAV_TYPE correctly. It should use the standard component_id corresponding to it's type, but it is NOT enforced to do so! That is, one can rely on MAV_TYPE to identify, but one cannot rely on component_id. 

Corallary: Of course, a coponent should (must) not use a standard component_id which is in conflict with it's type. I.e., a gimbal should not use a servo id, but it could use any of the non-standard ids. 

- in order to have some backwards compatibility, we allowed old code to do the identification by only the component_id

- exceptions exist, like RADIO_STATUS/SiK radios.

What you are saying is probably exactly this, but as said I find the wording a bit confusing so wanted to sue my own words.

> Doesn't this complicate message handling a lot? Previously I could say I'm only interested in component ids of a type - e.g. I check messages is from "mission planner component" so I will accept it. [...]

here you seem to confuse somewhat identification with discovery. The MAV_TYPE/component_id thing was about identification. Now we seem to switch gears to discovery.

I strongly think that any component MUST send out a heartbeat, and that discovery of the presence of a component must be via detecting the components heartbeat (exception RADIO_STATUS).  A component getting unhealthy or to disappear would be detected by not getting a heartbeat for a certain time. Only this can ensure that e.g. a gcs knows the type (and implied capabilities) of the component.

The alternative would be to discover by any message. 

The 2nd might sound simpler at first. However, if you look at a code like QGC (at least as it was 1/2 y ago, don't know if that has changed), it did discovery in multiple places, even for one and the same component. One could however easily abstract this into one class which listens to the heartbeat and keeps a list of components accessible from everywhere ... this would IMHO substantially simplify the QGC code. 

So, as regards complicating things, my arguments in favor of the heartbeat scheme is that it ensures that the type is known and that it "forces" you to a coding style which effectively is simpler.

> Another problem is consider proposal to add a companion computer component ID. IF I have a path planner as well then both components would currently have MAV_TYPE of MAV_TYPE_ONBOARD_CONTROLLER - and be interpreted as 2 companion controllers.

here you seem to confuse somewhat with capabilities. 

The problem of the various "on-board-like" ids is that they have a lot of implied capability, and that there is no suitable more generic standard id. Instead of a generic id like "companion" the ids are very much targeted towards particular purposes (= capabilities). It is a bit like if there would be a standard component_id for a camera which does video only, and a camera which does photos only, and a camera which does video and photo. It is better to have more generic camera id for all cameras and other means to infer a particular camera's capabilities. 

> So for this case to work wouldn't we need a new type for the path planner software that is running on the onboard controller?

since it is how it is, I indeed would think: yes.

IMHO. :)

olliw42, 2020-08-29:
I'm not sure what to do with this issue. The original purpose seems to have been settled, from somewhere a MAV_COMP_ID_ONBOARD_COMPUTER has now sneaked in. So, one could close.

However, there are also the last posts discussing more general aspects which I think are of relevance, and closing may make them going lost.

pl decide.

hamishwillee, 2020-09-29:
Good point. Closing. Try to make sure we don't lose the discussion by creating https://github.com/mavlink/mavlink/issues/1497

closed 2020-09-29 by hamishwillee