mavlink/mavlink #1347: Proposal for ultimate an fully backwards compatible solution to protocol pain points

olliw42, 2020-04-20:
**Issues**

The MAVlink V2 specification was developed in order to overcome some pain points of the V1 protocol, but the V2 designers did not see and/or understand the negative impact of two aspects of the protocol, the etxra CRC and that target fields are part of the message/payload and not of the header.

This leads to a couple of pain points, and the serious issue that it becomes impossible to reliably detect the integrity of a packet if the message id is not known to the implementation, and the resulting issues for routing messages. The message id needs to be known in order to determine the extra CRC, but without the extra CRC the CRC check cannot be performed and hence MAVLink's main mechanism to check the integrity of a package is spoiled. The message id needs also to be known in order to determine the position of target fields in the payload, but without the targets the message not be properly routed/forwarded.

This issue is by now well recognized, and several "solutions" were proposed, but to the best of my knowledge none could be considered satisfying.

The MAVLink protocol has another serious issue, which - AFAIK - hasn't been yet realized, at least I haven't ever seen it been mentioned. While the MAVLink protocol is good at ensuring that a false packet is not inadvertedly considered as a valid packet, it has no mechanism whatsoever to ensure that a valid packet is not missed. In hypothesis tests these two cases would be called false-positive and false-negative. MAVLink significantly fails in the later.

A main cause for missing a valid packet is a corrupted len field. Note that even though this is just a specific byte in the whole packet, it's not unlikely to happen on typical losy links. If the len field is modified to be shorter than required AND the payload has no FD/FE bytes then there is no problem, However, in all other cases the wrong len and resulting wrong jumps into the data stream can lead to longer chains of completely VALID messages been missed or rejected.

One method to overcome this deficancy of the protocol would be to have sufficient time gaps in the packet stream and some timeout mechanism. For high data rates (which are especially likely on typical lossy links) this however becomes impractical. Another, and probably very effective, method would be a backstepping mechanism where each FE/FD in the stream is tested as a potential start of a message. However, in view of the coding complexity and performance penalty this to the best of my knowledge has never been realized.

**Proposed Solution**

The basic idea is to properly modify the header and payload, and to set a bit in the incompatibility field and to re-purpose a bit in the compatibility field in order to indicate and handle this modification. For the purpose of this proposal I call such a modified packet a V2B packet.

Since both fields are massively under-used, i.e., so far only one bit is been used in the specification, taking away a bit from each for the purpose should not incur any backwards compatibility issue.

In fact, this is totally within the specification, since any implementation is required to NOT consider a packet if it doesn't understand all set incompatibility flags. Thus, "old" implementations will not be confused by the presence af the new V2B packets. Furthermore, since the len and the position of the CRC is not modified, any router can fall back and use its previous mechanism also for routing the V2B packets. Finally, both the "old" and the V2B packets can coexist on the same MAVLink network, and any implementation can freely decide if to send some data as V2B or in the older formats. Of course, an implementation not aware of the V2B packets cannot extract their data. The situation is somewhat similar to V1 and V2, albeit a bit lesser since V2B is close to V2.

The proposed modifications are:

- set the 8th bit in the incompatibility flag, to indicate a V2B packet

- repurpose the 8th bit in the compatibility flag as follows 
  unset to indicate that the message is broadcast and has no target fields
  set to indicate that the message has target fields (which still can be 0 for broadcast)

- modify the header, by extending it with the following fields
  message has no target fields (8th bit of compatibility field not set)
  --  uint8_t extra_crc
  --  uint8_t header_crc
  message has target fields (8th bit of compatibility field is set)
  --  uint8_t target_source
  --  uint8_t target_component
  --  uint8_t extra_crc
  --  uint8_t header_crc
  
- modify the payload as follows
  message has no target fields (8th bit of compatibility field not set)
  ->  no modification, payload is as before
  message has target fields (8th bit of compatibility field is set)
   -> remove the target_fields form the payload

Thus, the incompatibility flag is used to indicate a V2B packet, the compatibility flag is used to indicate the presence of target fields, and the header is extended by the etxra_crc, by the target fields if present, and by a short crc over the header.

The short crc over the header allows to test the integrity of the header and whereby also of the len byte, and is a very common since simple yet effective method to reduce the possibility of losing valid packets.

**Implementation Considerations**

The proposed V2B packets are only two bytes longer than V2 packets, which in view of the verbosity of the MAVLink messages and relatively long average payload sizes is a very small disadvantage bandwidth-wise.  

The code additions in order to handle the modified header should be a small effort to implement, since it's basically extending what is already there for handling V1 and V2.

A more significant issue might be the modification to the payload, in case that the payload has target fields. Code-wise it just would need an if() and to not include it in the pack/unpack functions. However, since the proposal introduces a correlation between the header and the payload (technically the correlation comes through the xml  definition), the code needs to be carefully revisted in order to ensure that the header is always set/known when a payload is packed/unpacked. From scratch I wouldn't know if that is the case for all and every use case.

**Final Thoughts**

Considering previous proposed "solutions" to account for the extra CRC and target fields issue and resulting routing issue, and other thoughts, makes it clear that this issue only can be solved - "solved" in the sense of really solving and not just work around - on the protocol level, i.e., by modifications to the packet format. No other way.

However, V2 has already a mechanism for handling that through the incompatibility flag, which currently is exploited only for signing. Given that any additional bit to this flag potentially implies increasing the coding complexity, one should of course very carefully balance the advantages vs the disadvantages.

This topic is not new to me, which implies that I had plenty of time to go through scenarios, I however kind of proved to myself that there is no other way than to modify the packet structure, and this here is a compatible procedure. That is, I think the advantages outweigt the disadvantage of that additional incompatibility flag.

I actually believe that once implemented V2B packets will quickly become the standard, given their advantage, and also some code simplifications which come with them. I don't think this will be as slow as it was with going from V1 to V2.

The proposal as outlined in the above has some inherent flexibility, some of which - if one wanted to - could be exploited with benefit. For instance, currently the presence of target fields is intimately linked to the xml definition of the message, i.e., written in stone. In the proposal it is ultimately however ONLY linked to a bit being set or not, i.e., entirely in the header only. That is, ANY message could now be send as broadcast or as targeted, at the will of the user depending on the specific use case, independent on whether there are targets or not in the xml definition, that is even if the message didn't had target fields specified. 

I note that this property, that any arbitrary piece of structured data can be send as either broadcast or peer-to-peer, is provided by essentially ANY more reasonable protocol. This point alone, since it's about arithecture and concept, should be argument enough for all I said.

Some of the flexibility I have left open.

Cheers, Olli

PS: If the proposal, or an improved version of it of course ;), should be accepted I would be willing to volunteer working on the C (and possibly C++) code generators.

amilcarlucas, 2020-04-23:
This will later need a PR to pymavlink

hamishwillee, 2020-04-24:
Thanks for this. I do not have time to look at it right now. Maybe next week.

hamishwillee, 2020-04-29:
@tridge Can you please look at this, since you're the expert on original design for Mavlink 2

tridge, 2020-04-29:
I'm not convinced about this proposal. There are a couple of errors in the background. For a start, the message structure part of the CRC (the "extra" crc element) was also in MAVLink1, it is not a new feature in MAVLink2. Code that wants to route MAVLink packets while coping with new additions to the set of messages since the router code was built has had to deal with that issue ever since MAVLink1 was introduced. It is a problem that a lot of router code doesn't cope with that well, which is probably because the APIs don't make it easy to handle it, so the easy path when writing a MAVLink router is to drop messages that were not known when the router was built.
If I had a chance to do MAVLink1 or MAVLink2 again I would have made the extra CRC component only impact on one of the two CRC bytes, that way routers would have had a 8 bit component they could use for message IDs that they didn't have full information on, which would have been a considerable improvement.
The way routers should work is they should fwd packets without checking the crc for messages they don't know about. The burden is then on the final recipient of the message to check the crc (which it has to do anyway). This does means a small number of bad packets get forwarded unnecessarily, but it really isn't very common, so isn't a big issue as long as routers do go to the extra effort of not dropping packets that they don't understand.
The 2nd issue raised, that of losing valid packets when there is corruption in bytes preceding the valid packet, is also something that can be greatly reduced with careful coding of the state machine. I think there likely is scope to improve the state machine decoders in the most commonly used implementations (ie. python and C).
The protocol could be evolved to reduce the impact of both of these issues, but the cost to the user community of the proposed protocol change far outweighs the benefits in my opinion. The proposal is addressing two issues which rarely affect real use of MAVLink, but this will have a very real impact on a lot of users and developers and would require a lot of code churn. I just don't see that this stacks up in terms of the advantage for the cost.
Cheers, Tridge

olliw42, 2020-04-29:
many thx for the comments
however, claims are made which are not correct, and important points not addressed

> There are a couple of errors in the background. For a start, the message structure part of the CRC (the "extra" crc element) was also in MAVLink1, it is not a new feature in MAVLink2. 

this was nowhere said or claimed in the proposal, so unjustified claim of an error

also, this in NO way invalidates the major statement of the issues caused by this, so is totally irrelevant. The caused issue is in fact not disputed but confirmed, which by itself evidences the irrelevance

except of the false claim of this error, no further errors in the background are indicated

ergo, no justification for claiming "couple of errors in the background".

> The way routers should work

This is the way routers need to work BECAUSE of the issues of the protocol, but this is certainly NOT the way routers SHOULD work in the general sense of how routers should work. The protocol issues are taken here as argument for how routers should work, which is nothing else than turning logic upside down. The light goes on because the switch is flipped, and not the switch flips because the light goes on.

> The 2nd issue raised

It was explicitly mentioned in the proposal that alternatives exist, which seems to have been overlooked. However, fact is, besides many years, and it was explicitly pointed out that the crc issue exists even since MAVLink1, so it are really MANY years, none of this was done. It is stale to argue that things could be done despite they were never done for years besides they could have been done since years.

Also, careful inspection of the proposal should have made it clear that the 2nd issue was not the primary motivation of the proposal, but that this is a convenient occasion to also in addition make an improvement in this regards.

> The proposal is addressing two issues which rarely affect real use of MAVLink

this is probably a joke. The evidence as available through the various public inet communication channels clearly speaks to the contrary.

Finally, two important aspects, which are primary motivation for the proposal, were not commented on at all in the response, namely the targeting issue and the incapability of handling unknown msgids properly. Not a word. A statement like "The proposal is addressing two issues" is in fact is denial.

A router certainly should be able to properly forward messages, it's the whole idea of a router and of having data packets and of having recipients. Otherwise the "router" could just forward any byte to any link. It is highly unlogical to argue just half-ways; MAVLink currently is doing partially the later.

So, altogether, the response appears to be just a quick shoot out. It is not well thought out and not well based on arguments and misses key aspects. It is not surprising that it then may appear as not convincing. It also has a bit of play with words, which doesn't make it better.

All the best, 
Olli

Addendum:
> This does means a small number of bad packets get forwarded unnecessarily, but it really isn't very common, so isn't a big issue as long as routers do go to the extra effort of not dropping packets that they don't understand.

incorrect. On losy links, such as typical telemetry units, this produces a significant amount of noise routed to other components. Which is amplified in more complex MAVLink networks - which are well known to exist, just see some boats showcased in teh recent ArduPilot conference. It is fundamentally weak to argue just based on the simplest use cases.

olliw42, 2020-05-04:
I'd like to bring in a further fact.

ArduPilot is (also) a router. 

However: It completely fails in all of the mentioned pain points. 

It even fails in the one point which could be overcome to some extend in other ways than the proposed one, as it was mentioned in the proposal and also in tridge's response as how a router should work. Ardupilot doesn't route any message in any way which is not explicitly known to it.

The consequences are e.g.: One can't reasonably operate a MAVLink camera. One can't reasonably 
operate a MAVLink gimbal which implements the new gimbal v2 protocol, e.g. in the "Onboard Computer with Camera and Gimbal Connected to Autopilot" configuration. A GCS can't communicate with a component in pass-through mode. ArduPilot can't even route COMMAND_ACK's properly but spills the MAVLink network. And so on.

This is fact for ArduCopter 4.0.3, and while I have not re-tested it for them to the best of 
my knowledge it is also true for current ArduPilot master and ArduPlane stable. 

It is somewhat noteworthy that ArduPilot fails in even the one point which a main ArduPilot developer argued could be well done and is how a router should work. This underlines the notion that it is one thing to say that something would be possible, but a very different thing to have it in practice; there is a difference between talk and reality.

It was in particular argued that the proposal is addressing issues which rarely affect real use of MAVLink. I believe that there is evidence that not all ArduPilot community members agree to this assessment, but if ArduPilot thinks that the described restrictions and limitations are of no relevance to their community, so it be. There is no point to argue here. 

There is however the point to make that, given ArduPilot's complete failure in this matter, ArduPilot hardly can serve as a role model or guide for the larger MAVLink community!

For PX4 the situation is much better. It adopts the latest standard, and the latest firmware version always has knowledge of all current standard messages. However, this still impairs interoperability. Not everyone might jump immediately to the  latest firmware on all her/his models, and newer components may thus not be usable with these older albeit totally reliable and trusted setups. As a hobbyist in the matter I certainly cannot speak for the professional world, but it is my impression that relying on older since proven setups is not unheard of in professional environments.

If one wants to think about MAVLink as an industry standard, it is kind of difficult to imagine that a 
protocol with such elementary deficiencies could fill that role.

As it was mentioned now repeatedly, one of the several pain points could be overcome to some extend by alternative approaches than the proposed one. But these do not resolve the root cause, but work around. The proposal resolves the issue at the root. Moreover,  it resolves the other pain points for which there is no workaround, by again correcting the root cause. In addition, the proposal provides further benefits and flexibility. 

These points are facts too.

One would sincerely hope that MAVLink is not blocked by the egoisms of ArduPilot, and one would hope that the MAVLink standard bearers would find ways to not let this happen, but follow the arguments.

Clearly, that's a pleading only, but it's the best and all I can do. :)

tadam777, 2020-05-21:
> The MAVLink protocol has another serious issue, which - AFAIK - hasn't been yet realized, at least I haven't ever seen it been mentioned. While the MAVLink protocol is good at ensuring that a false [...]

Regarding the second point, it can partially be done with the Packet Sequence Number field. It's a field which is incremented with every message, and is used by GCS to determine the number of missed packets.

For the first point, maybe there should be something like we have with TCP / IP or rather UDP / IP : the IP protocol is completely agnostic of what's sent inside the TCP or UDP packets.

In the same way we could have a "MavTCP" protocol that would just contain :
0. 0xFE
1. payload length
2. packet sequence
3. system id
4. component id
5. CRC
6. Mavlink version
7. "payload"

And this would be guaranteed to stay the same accross all future mavlink versions. Notice that the magic byte would always stay the same, independently of the mavlink version. The mavlink version would be in a separate field (6).

And with this approach you can at the same time guarantee packet content through CRC, and know where to route it (system / component id) even if you don't understand the specific mavlink version it contains in the payload. That would be perfect for a mavlink router.

The payload would the be equivalent to the UDP protocol, and contain any data specific to the corresponding mavlink version.

That would be future proof. Because for the moment the header of the message is different between mavlink 1 and mavlink 2. If mavlink 3 is introduced, your point is valid, nobody will know for sure how to parse it, maybe the header will be different, as the headers of mavlink 1 and mavlink 2 are different.
Same goes for the mavlink xml definition files, the fields which are related to mavlink 2 only are after the <extensions/>. But how will it be for mavlink 3 ? Will there be an <extensions3/>?
Personnally I would have added the mavlink version as an attribute in every field :

For example

  <message id="152" name="MEMINFO">
      <description>State of APM memory.</description>
      <field type="uint16_t" name="brkval">Heap top.</field>
      <field type="uint16_t" name="freemem" units="bytes">Free memory.</field>
      <extensions/>
      <field type="uint32_t" name="freemem32" units="bytes">Free memory (32 bit).</field>
    </message>

Would become

   <message id="152" name="MEMINFO">
      <description>State of APM memory.</description>
      <field version = "1" type="uint16_t" name="brkval">Heap top.</field>
      <field version = "1" type="uint16_t" name="freemem" units="bytes">Free memory.</field>
      <field version = "2" type="uint32_t" name="freemem32" units="bytes">Free memory (32 bit).</field>
    </message>

In this way we can add many mavlink versions and be sure that current software will safely ignore the future fields.

hamishwillee, 2020-07-22:
> Regarding the second point, it can partially be done with the Packet Sequence Number field. It's a field which is incremented with every message, and is used by GCS to determine the number of missed packets.

Actually it **is** done with the sequence number when and if required, at the service layer. This allows us to add the cost of supporting message resending etc for just those messages where it matters/is required. As a bonus, if you run MAVLink over a lossless protocol then you aren't paying that cost twice - once in MAVLink and once in the new transport (e.g. TCP).

No one knows what MAVLink 3 will look like; at the moment it is still just a set of ideas. I can say that I don't see why we would need to have versions of extensions - the only reason this would be needed is if something changed in the protocol that meant a  new field could not be sent in mavlink 2. 

As for versioning fields, there is a temptation to do so but there are other ways of solving the underlying problem.