mavlink/mavlink #1367: Gimbal V2 Protocol Discussion II

olliw42, 2020-05-04:
I am in the course of implementing the new gimbal v2 protocol in my system, and do see some points which IMHO do need some discussion.

Some of the points might have been discussed in the previous long gimbal thread, i.e., relevant info might be there, but I thought it's not worthwhile to look back, and this old thread became discouragingly long anyway, so I thought I should better open a new issue.

Since I would actually want to touch a couple of points, and each may induce longer discussions, I also thought that it would be more productive to put the points which could be discussed separately into separate threads (the long thread suffered from so many topics being touched in one thread, which was not always helpful).

So, you will see a couple of these "Gimbal V2 Protocol Discussion II: XXXX" issues appear in the next time.

I hope you find this approach acceptable, even though it spills the issue list. If not, please tell, and I will be happy to condense all into this one thread here.

This thread here I would suggest to use as a kind of contents list, and summary, if that is acceptable.

Content List
-------------

- [ ] Discovery, see 

Setup
-------------
For reference, and as background, so you know what I am talking about, I'd like to describe my setup, since it's not exactly as in the docs:

The setup consists of a GCS, a flight controller, a companion computer, and a gimbal controller. These are arranged into a MAVLink network as follows (each <-> represents a physical link):

GCS <-> flight controller <-> companion computer (gimbal manager) <-> gimbal controller (gimbal device)

This configuration is thus somewhat different to those described in the documentation. The main reason is that I don't want to have the companion computer inbetween the GCS and the flight controller, which are linked by a telemetry unit, for which I'm using SiKs.

The GCS in my case is a OpenTx transmitter running a MAVLink enabled OpenTx firmware fork.

The flight controller runs a fork of ArduCopter 4.0.3 which I call BetaCopter.

The companion computer is a RaspberryPi (on the copter it's a Zero and on the bench a 3A+) running mavlink-routerd and some home-brewed python code. The gimbal manager is located on this component.

The gimbal controller is a STorM32. This is the gimbal device.

So, I do have all the pieces under my full control in a software sense, i.e., can tweak their behavior at any time as needed and desired.

julianoes, 2020-06-02:
@olliw42 sorry, I had missed your issue posts and only found them last week.

It makes sense to go through the various aspects one by one, thanks for posting these.

olliw42, 2020-06-06:
no problem :)

many thx for considering and answering to all of them. much appreciated.

For the moment I would like to answer to only "Multiple Clients and Deconfliction" #1380, as it's seems to be the most "difficult" one for the moment and to keep focus, if that is ok.

dlwalter, 2020-11-10:
I don't see a separate thread for the GIMBAL_DEVICE_STATUS message but I would like to mention the following need - 

It's great being able to report the gimbal attitude back to the GCS for display, but another user need I see is also to display where the sensor is looking on the GCS map.  

One option would be to add the global position of where the device is looking if available.

The other option is to have the GCS do the geodesy and convert the reported angles to a location on the ground for display.

Are there any thoughts on how this framework and how a gimbal manager  would communicate the global position of where the gimbal is looking or are others not seeing this as a need?

julianoes, 2020-11-11:
That would be handy. I would argue that it's up to the gimbal manager or camera to provide that information, e.g. with:
https://mavlink.io/en/messages/common.html#CAMERA_TRACKING_IMAGE_STATUS
https://mavlink.io/en/messages/common.html#CAMERA_TRACKING_GEO_STATUS

Would that work?

olliw42, 2020-11-11:
it would be handy but I don't think it should be the gimbal, gimbal manager, nor camera which is emitting the info. Lot's of info is needed, such as position, altitude over ground, and potentially also info from terrain data if non-flat earths are considered ...

dlwalter, 2020-11-11:
@julianoes Those seem to have the necessary information, thanks for pointing those out.

@olliw42 If the gimbal manager can support an DO_SET_ROI_LOCATION based on lat/lon/amsl or local coordinates then I believe it would be reasonable to expect it to emit a lat/lon/amsl message as to where it is looking. In order to handle the DO_SET_ROI_LOCATION it will need aircraft global position and geodesy libraries.

olliw42, 2020-11-11:
I don't think that it should be the gimbal manager which should support something like DO_SET_ROI_LOCATION, but I know that this appears to not be the current fabric of the gimbal manager.
:)

dlwalter, 2020-11-11:
Interesting.  I would recommend having Gimbal Manager handling attitude stabilization as well as translational stabilization and not put this load on the flight controller.   Gimbal Manager can indicate this support with a capability flag.  

I personally would rather not use DO_SET_ROI_LOCATION and have a specific GIMBAL_MANAGER_SET_GLOBAL_TGT (or something similar) message or command.

olliw42, 2020-12-05:
has kind of become irrelevant, so I close it. :)

closed 2020-12-05 by olliw42