mavlink/mavlink #1368: Gimbal V2 Protocol Discussion II: Discovery

olliw42, 2020-05-04:
The discovery process as described in the documentation so far has IMHO a logical flaw (or two), and could need some generalisation. The general problem is that there are some time lines involved for when the various parts become online, available, known, and so on. This easily, and usually will, lead to situations which are not in the text book. 

1.

Let's discuss the logical flaw by means of the discovery sequence https://mavlink.io/en/services/gimbal_v2.html#discovery.

The point is that the COMMAND_ACK to a GIMBAL_MANAGER_INFORMATION request is send before the GIMBAL_DEVICE_INFORMATION is requested. At this point in time the gimbal manager may however NOT yet know about the existance of the gimbal device or the gimbal device may not yet be able to respond, e.g. because it has not yet booted up or what ever else may prevent it from sending MAVLink messages.

Now, what should happen in that case? The GCS somehow needs to know that it is supposed to resend the request at some later time in order to see if there is now a positive response. It could note this by realizing that there is no GIMBAL_MANAGER_INFORMATION coming. However, the canonical approach, along the general MAVLink guide lines, would be that the gimbal manager would return a COMMAND_ACK with result MAV_RESULT_TEMPORARILY_REJECTED. However, in the current sequence, at the given time the gimbal manager cannot decide if it should send a COMMAND_ACK with ok or rejected, since it only afterwards checks if the gimbal_device responds to the GIMBAL_DEVICE_INFORMATION request.

It's easy to see that this sort of potential conflict is present in the other sequences too.

Furthermore, there is actually no reason why these sequences have to be as they are. The communication between gimbal manager and gimbal device and the communication between the gimbal manager and the GCS could equally well be totally asynchronous!

That is, the gimbal manager could do a request for GIMBAL_DEVICE_INFORMATION whenever it likes, e.g. early on at startup. The GCS on the other hand also could send its request for GIMBAL_MANAGER_INFORMATION at any time it likes. These two communications don't need to be linked time-wise in any way. In fact, I would argue, that it would make programming the gimbal manager substantially easier if they were done asynchronous! It also would naturally resolve the mentioned issue. For as long as the gimbal manager is still in the process of discovering the gimbal devices, it can send COMMAND_ACK with MAV_RESULT_TEMPORARILY_REJECTED to the GCS' request. Once it either knows that there are is no gimbal it cand respond with MAV_RESULT_DENIED or it has finished gimbal discovery and can respond with MAV_ACCEPTED, and the GCS would understand.

Asynchronous communication is maybe even more evident for the other sequences, which are about control of the gimbal. Situations certainly exist there the gimbal manager may want to send out GIMBAL_DEVICE_SET_ATTITUDE messages at a potentially much higher rate than it receives control info from a GCS.

Thus, my proposal: 
The communication between gimbal manager and gimbal device should be organized to be asynchronous to the communication of the gimbal manager with the outside world.

2.

There is a second time twist in that a GCS does not know if a gimbal is already operational. The gimbal device first needs to have discovered its gimbal manager before it can respond to e.g. GIMBAL_DEVICE_SET_ATTITUDE and react. A GCS doesn't know when this has happened - and this may be actually a long time if one also considers smart gimbals which need the autopilot assistance in addition - and thus might start to control a gimbal even though the gimbal is not yet ready.

This could be handled by adding a GIMBAL_DEVICE_ERROR_FLAGS, but since it's a general problem of the current communication sequences one might look for more general solution.

3.

One inconvenience in the current scheme is that it needs a bit too much configuration work, e.g. the gimbal manager needs to be configured to know which gimbal device it is assigned to, because otherwise it could not reasonably respond to requests or even send out GIMBAL_MANAGER_STATUS messages (for which it needs to know the gimbal_device_id).

The sequences can however very easily be adjusted to be more plug and play.

In the big old gimbal thread Julian made the suggestion that a gimbal could look for the autopilot heartbeat, and use it to determine the system_id. This was a great suggestion and I immediately implemented it in STorM32, and it works just great. The advantage is obvious: The gimbal needs to be configured only for its component_id, usually GIMBAL1. But once done, it can be attached to any flight controller, and the user can even change the system_id of the flight controller at any time without having to worry about the gimbal. Just great.

Note that this scheme is actually suggested in the documentation's FAQ.

However, the problem, this can leads to some unresolvable regressions, since the gimbal can't start before it knows the system_id and the gimbal manager can't start because there is no gimbal heartbeat. 

An alternative sequence could be this gimbal discovery:

- gimbal waits for autopilot heartbeat to get system_id, only now the gimbal can start to send out heartbeats and do other MAVlink things

- gimbal manager is mostly silent, it cannot yet send out e.g. GIMBAL_MANAGER_STATUS messages since gimbal_device_id is yet unknown. The gimbal manager waits for the gimbal heartbeat to arrive. Once received, it assigns its gimbal_device_id to the gimbal's component_id, and only now it can start to send out GIMBAL_MANAGER_STATUS messages, as well as to request for GIMBAL_DEVICE_INFORMATION.

- gimbal waits for GIMBAL_MANAGER_STATUS messages, to discover its assigned manager, only now it can digest and react to e.g. GIMBAL_DEVICE_SET_ATTITUDE messages

The proposal is: 
A GCS or other outside component should wait for at least one GIMBAL_MANAGER_STATUS message, before it starts sending GIMBAL_MANAGER_INFORMATION requests or other gimbal things.

Assuming that the gimbal device has not missed the first GIMBAL_MANAGER_STATUS message, which should be a very good assumption given this is a reliable link, this ensures that the gimbal device is all set up and the gimbal manager too. If the gimbal device should have missed that first status message, it is no harm as the gimbal manager would delay the GCS as discussed in 1.

Thus, in combination with the asynchronous communication suggested before this makes a fully working scheme, where only the gimbal needs to be configured for it's component_id.

I''d like to add that it just feels like very natural that any outside component should first watch out for GIMBAL_MANAGER_STATUS messages before it attempts to start any communication with a gimbal manager.

Note that the scheme extends also to several gimbals. There could be an instance which creates a gimbal manager for each gimbal heartbeat it has seen, and the logic continues as before. 

 4. Summary:

To sum up that long text, I suggest two things:
- The communication between gimbal manager and gimbal device should be organized to be asynchronous to the communication of the gimbal manager with the outside world.
- A GCS or other outside component should wait for at least one GIMBAL_MANAGER_STATUS message, before it starts sending GIMBAL_MANAGER_INFORMATION requests or other gimbal things.

That's it on this. :)

julianoes, 2020-06-02:
1. 

> That is, the gimbal manager could do a request for GIMBAL_DEVICE_INFORMATION whenever it likes, e.g. early on at startup. The GCS on the other hand also could send its request for [...]

I completely agree. And that's how I thought about this. The diagram is only meant to visualize how it "could" look like. It was just an example, and mostly served as illustration of what message is used between which agent.

> For as long as the gimbal manager is still in the process of discovering the gimbal devices, it can send COMMAND_ACK with MAV_RESULT_TEMPORARILY_REJECTED to the GCS' request. Once it either knows that there are is no gimbal it cand respond with MAV_RESULT_DENIED

One problem here is that existing autopilots (at least PX4) will answer the REQUEST_MESSAGE command with MAV_RESULT_DENIED if it doesn't know about the new protocol and thus GIMBAL_MANAGER_INFORMATION yet. This means we might want to send out a GIMBAL_MANAGER_INFORMATION with `gimbal_device_id` set to `0` to signal that we have not found anything.

Instead of returning MAV_RESULT_TEMPORARILY_REJECTED I would suggest to return MAV_RESULT_IN_PROGRESS and try to do the gimbal device discovery for e.g. 2 seconds, and then answer with MAV_RESULT_ACCEPTED and send an empty GIMBAL_MANAGER_INFORMATION.

2. 

> A GCS doesn't know when this has happened - and this may be actually a long time if one also considers smart gimbals which need the autopilot assistance in addition - and thus might start to control a gimbal even though the gimbal is not yet ready.

My suggestion is that the ground station should keep requesting GIMBAL_MANAGER_INFORMATION at a slow rate, e.g. every 10 seconds. If any new gimbal is discovered, or attached it can be found this way.

3. 

> - gimbal waits for autopilot heartbeat to get system_id, only now the gimbal can start to send out heartbeats and do other MAVlink things

> - gimbal manager is mostly silent, it cannot yet send out e.g. GIMBAL_MANAGER_STATUS messages since gimbal_device_id is yet unknown. The gimbal manager waits for the gimbal heartbeat to arrive. Once [...]

> - gimbal waits for GIMBAL_MANAGER_STATUS messages, to discover its assigned manager, only now it can digest and react to e.g. GIMBAL_DEVICE_SET_ATTITUDE messages

I agree with the first two steps. That makes sense. However, the third one is odd. I don't think the gimbal needs to wait for GIMBAL_MANAGER_STATUS because it doesn't need to know its manager. All it needs to do is to react to GIMBAL_DEVICE_SET_ATTITUDE messages, and broadcast GIMBAL_DEVICE_ATTITUDE_STATUS. It doesn't need to filter from where GIMBAL_DEVICE_SET_ATTITUDE comes, it only needs to check if target_system and target_component match itself.

Following that I don't think your suggestion is needed, unless I'm missing something...

4.

> The communication between gimbal manager and gimbal device should be organized to be asynchronous to the communication of the gimbal manager with the outside world.

Makes sense.

> A GCS or other outside component should wait for at least one GIMBAL_MANAGER_STATUS message, before it starts sending GIMBAL_MANAGER_INFORMATION requests or other gimbal things.

As said, I'm not convinced that's necessary.

olliw42, 2020-12-05:
discussion has kind of settled, and the issue went stale anyway, so I close it. :)

closed 2020-12-05 by olliw42