mavlink/mavlink #1369: Gimbal V2 Protocol Discussion II: Absolute/relative yaw

olliw42, 2020-05-04:
In the long gimbal thread the topic of absolute and relative yaw, and the related issues such as with smart gimbals which need autopilot assistance, had been discussed at quite some length, since it's not an easy topic. I had the impression that some conclusion was reached in that thread. It is even addressed in the google document.

However, I do not see any appearance of this in the current mavlink gimbal v2 messages. It is in fact totally unclear to me how that should be dealt with.

The issue becomes more involved when in addition autopilot assistance through the AUTOPILOT_STATE_FOR_GIMBAL_DEVICE message is at play. This introduces some additional time lines in the whole flow, and I can't see how one could resolve that with what is existing.

So, in short, no idea how that all is supposed to work, and I would appreciate some clarifying comments.

:)

julianoes, 2020-06-02:
You're right that this topic is not addressed enough in the documentation. Let's improve it, and by the way, feel free to make pull requests to the docs and suggest changes and additions as you would like to see them. The current iteration is merely a first try of mine.

Now, here are the notes coming from the message definitions that do already exist:

1.

Quaternion in [GIMBAL_MANAGER_SET_ATTITUDE](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_SET_ATTITUDE)
> the frame is depends on whether the flag GIMBAL_MANAGER_FLAGS_YAW_LOCK is set

Quaternion in [GIMBAL_DEVICE_SET_ATTITUDE](https://mavlink.io/en/messages/common.html#GIMBAL_DEVICE_SET_ATTITUDE))
> the frame is depends on whether the flag GIMBAL_DEVICE_FLAGS_YAW_LOCK is set

This is not very clear, maybe something like this would be better:

The frame is the world frame if ..._FLAGS_YAW_LOCK is set (e.g. yaw 0 points North).
The frame is the world frame, however, aligned with yaw of the drone (e.g. yaw 0 points forward).

2. 

The meaning of FOLLOW and LOCK is explained in [GIMBAL_MANAGER_CAP_FLAGS](https://mavlink.io/en/messages/common.html#GIMBAL_MANAGER_CAP_FLAGS)/[GIMBAL_DEVICE_CAP_FLAGS](https://mavlink.io/en/messages/common.html#GIMBAL_DEVICE_CAP_FLAGS), however, it would make sense to add a point to the docs to explain the meaning.

3.

Regarding: [AUTOPILOT_STATE_FOR_GIMBAL_DEVICE](https://mavlink.io/en/messages/common.html#AUTOPILOT_STATE_FOR_GIMBAL_DEVICE):

This message is based on what gimbal companies that I have been working with usually asked for which is the drone's attitude and speed. Additionally, I was told that delay/timing is critical and knowing the approximate delay can further help the estimate. Hence the fields `q_estimated_delay_us` and `_v_estimated_delay_us`.

I don't have much understanding of this and no strong opinion, so I'm all ears for suggestions or enhancements as you see fit.

olliw42, 2020-12-05:
discussion has kind of settled, and the issue went stale anyway, so I close it. :)

closed 2020-12-05 by olliw42