mavlink/mavlink #1371: Gimbal V2 Protocol Discussion II: Gimbal device attitude status

olliw42, 2020-05-05:
I wonder why the GIMBAL_DEVICE_ATTITUDE_STATUS message has no target fields. I actually think it should have some.

This point we might have actually been overlooked in the old long gimbal thread, and I actually might have expressed a different opinion there. If so, I'm happy to accept the blame of not having realized it. :)

The point is this: 

It is easy to anticipate situations there a particular component would want to receive GIMBAL_DEVICE_ATTITUDE_STATUS messages at high rate, or where this actually would be required for proper operation. This would be e.g. situations like smart shoots or quick shots or whatever one wants to call them, there e.g. a companion computer takes control of the gimbal, but requires a fast feedback in order to position the gimbal properly. Update rates of 10 Hz can be too slow for such cases. (the 3DR Solo e.g. I think uses 20 Hz, and in my own experiments I found 10 Hz also a bit at the edge)

On the other hand, there are certainly situations there one can easily get away with low update rates. Most typical this would be the GCS. ArduPilot by default uses 2 Hz, I am using 4 Hz, and I really find 4 Hz plenty for the purpose. You certainly realize that it is especially the link to the GCS which usually is quite bandwidth limited (telemetry), and that it thus makes lots of sense to really not overdo the rates here.

In the current scenario where GIMBAL_DEVICE_ATTITUDE_STATUS is broadcast the conflict of interests as regards the rates cannot be resolves, but only a (usually bad) compromise be set.

With target fields, a gimbal device could e.g. do this: 
Per default it could send out GIMBAL_DEVICE_ATTITUDE_STATUS at e.g. a rate of 4 Hz, with target set to broadcast. Another component such as a companion could request a targeted stream at e.g. a rate of 20 Hz. This would make the gimbal to send at 20 Hz, but every 4 out of 20 messages the target would be broadcast and not the requester. If the GCS is happy with lets say 2 Hz, it could also get this by requesting a stream at 2 Hz but with the requested target being broadcast.

This would address the issue, and allow a companion to get the gimbal position at high rates without blocking up the telemetry channel.

(btw, do you realize how handy it now comes in that the requester messages have become fields to determine the target of the response, which was somewhat disputed? That's the beauty and benefit of a conceptually designed specification, it opens up options ;)).

My suggestion:
Add targets to this message. And maybe some hints in the docs as regards the suggested request behavior.

:)

julianoes, 2020-06-02:
> This would address the issue, and allow a companion to get the gimbal position at high rates without blocking up the telemetry channel.
> Add targets to this message. And maybe some hints in the docs as regards the suggested request behavior.

Sounds good, makes sense to me. It should have a comment saying "normally this can just be broadcast".

> btw, do you realize how handy it now comes in that the requester messages have become fields to determine the target of the response, which was somewhat disputed?

Yes, fair enough. However, for the PX4 case this part is not implemented because it technically creates some headaches :smirk:.

olliw42, 2020-06-23:
resolved per https://github.com/mavlink/mavlink/commit/8bfccff9fe2eda10927eba4ee18bef7d47812f4f

hence closing

MANY THX!

closed 2020-06-23 by olliw42