mavlink/mavlink #1372: Gimbal V2 Protocol Discussion II: Autopilot state for gimbal device

olliw42, 2020-05-06:
The new message AUTOPILOT_STATE_FOR_GIMBAL_DEVICE is IMHO quite good as is, but could be furtehr improved IMHO by a flags fields.

Before jumping into the case I really want to say that it is really great that the message AUTOPILOT_STATE_FOR_GIMBAL_DEVICE has been added. It was soo much missing (at least I was desperately hoping for something like it since 4 years or so, and it's absence invoked a repeated number of partially awkward workarounds). So, this is really great, and the community should applaud to MAVLink.

Overall I'm also very happy with the current design. I only would wish that it would have a flags field which would allow it to communicate some information on various autopilot conditions.

The most crucial conditions, such as if the attitude and/or the velocity is good to use, can be decoded via setting these fields to NAN or not, and this works fine.

There can be however additional conditions which could be helpful and make a difference.

The flags I am missing are all, more or less, related to the startup of the system, and gimbal in particular, and the fact that the gimbal somehow needs to process the data received with AUTOPILOT_STATE_FOR_GIMBAL_DEVICE, and the implemented alogorithms can be simple or more advanced. 

For instance, at least that's so with ArduPilot, both the attitude and velocity estimate have different accuracy when the system starts up. Velocity estimates become available after few 10 seconds, but they become much better once a position lock has been acquired, which happens much later. For the processing it now can be helpful to know the "accuracy" of the velocities. E.g., when the better values arrive the STorM32 gimbal controller uses somewhat different tuning parameters in its estimators, improving the overall behavior. Now, one could argue, lets just always wait for the better values, and the NAN disabling/enabling mechanism would be sufficient. This however would prevent a user to be able to start the copter earlier in a non-gps mode. For such reasons the STorM32 actually knows about four flags, AHRSHEALTHY, AHRSINITIALIZED, NAVHORIZVEL, GPS3DFIX (there the first is admittedly superflous however, and GPS3DFIX I should replace by something which is better indicating the position lock)

For similar reasons one may want to exploit if the vehicle is flying or not, since on the ground, when it is sure that the vehicle is at rest, filter parameters can be choosen tighter or even reset or a gyro be recalibrated, which in flight would not be possible. Since this are three states, "one the ground", "flying", "transitional", there the latter would be all somewhat undertermined situations such as from arming to actually hovering in the air, this would suggest two flags "ONTHEGROUND" and "INTHEAIR".

I even see use of an armed flag. Of course, the arming state could also be inferred from the autopilot heartbeat. However, this introduces a lag of up to 1 sec. When armed by the user, ArduPilot for instance first waits 2 secs before it starts the motors. This is very usefull since it's sufficient time to recalibrate some gimbal estimators, or to actually startup motors, so that the gimbal is in best condition before actual take off. The 1 sec lag however reduced the useable time to 1 sec, which can be too short (and for the STorM32 indeed is). (the purpose of the arming flag could be also achieved with the "ontheground" flag, depending on the definition, the point here is just to add to the point that flags can be usefull, and even better one would have a generic MAVLink way to have components introduce a moment of silence afetr arming before the motors start, but this would be a total different story)

I could provide two more examples, but I guess the point has been made. And the suggested flags, better info on the attitude and velocity qualities, and on the ground vs in the air, are those which I miss most.

My suggestion:
Let's add a flags field, e.g. of size unit16. The flags could improve over time. The above could give some hints towards initial candidates.

Cheers,
Olli

julianoes, 2020-06-02:
Ok, I can see how that would be handy. I would suggest to avoid re-inventing flags and re-use as much of the enums as possible though.

E.g. we can use [MAV_LANDED_STATE](https://mavlink.io/en/messages/common.html#MAV_LANDED_STATE) though, and for estimator accuracy I would either separately subscribe to [https://mavlink.io/en/messages/common.html#ESTIMATOR_STATUS](https://mavlink.io/en/messages/common.html#ESTIMATOR_STATUS) or add some sort of `std_dev_attitude` and `std_dev_velocity` flag.

olliw42, 2020-06-10:
this is interesting. I indeed have not had these kind of enums in my mind.

I guess what this would imply however is that there would be several flag fields and not just one (tailor-made) uint16_t, right? 

Following your links there currently would be
uint8 landed_state
uint16 estimator_status

Correct?

julianoes, 2020-06-10:
Yes, I'd say so.

olliw42, 2020-06-10:
ok. I will do a corresponding PR, if you don't mind.

julianoes, 2020-06-10:
Great, thanks.

olliw42, 2020-06-10:
I am sooo happy that this message is existing now in this form, that it is really me who has to thank you guys :)

olliw42, 2020-06-23:
resolved per https://github.com/mavlink/mavlink/commit/e1386e79761916dc1aacccb38c848def6fbbbee0

hence closing

MANY THX!

closed 2020-06-23 by olliw42